MarlinFirmware/Marlin #15437: [BUG] Can't assign chamber pin on 32-bit board (MKS_SGEN_L)

projectoblio, 2019-10-01:
### Description

Anytime I assign a value to `HEATER_CHAMBER_PIN`, my board freezes up. I followed the guide here for activating a chamber:  https://3dprinting.stackexchange.com/questions/10907/how-to-set-up-marlin-to-use-a-heated-chamber

### Steps to Reproduce

Please follow the steps above. They worked on my 8-bit MKS_GEN board, but not my 32-bit MKS_SGEN_L board. 

**Expected behavior:** My board to turn on and chamber to work

**Actual behavior:** Board appears "bricked" and won't even connect to the computer.

biocrypto730, 2019-10-01:
How do I find the pin mappings for LPC1768? I thought they would be in the comments in src/HAL/HAL_LPC1768/fastio.h but they aren't there

biocrypto730, 2019-10-01:
I updated the title to reflect further testing

biocrypto730, 2019-10-01:
Any suggestions -good or bad- for debugging this would be welcome. I have followed all instances of `HEATER_CHAMBER_PIN` throughout the codebasw and I still have no idea why this would happen. It works with any pin even pins that work with other variables

p3p, 2019-10-01:
I responded yesterday on discord but I'l say it again here, unless you supply your configs (and any other files you change from default) as requested in the template you are unlikely to get a response.

As a guess you are more than likely trying to use the digital pin number rather than the analog channel when setting the thermistor pin number, the mapping can be found [here](https://github.com/p3p/pio-framework-arduino-lpc176x).

biocrypto730, 2019-10-01:
@p3p thank you for the pin mappings, Google wouldn't procure that repo for me... Can we add a link to that in the lpc1768 fastio.h file? 

I am actually using an analog channel for the thermistor. It's just the same one from TEMP_1_PIN. But your sentiment is correct - The problem comes when trying to use certain channel for the HEATER_CHAMBER_PIN. However this should just be a digital pin since it's a bang-bang for the heated chamber.

What I would really like is for my heater_chamber_pin to be controlled by X_MAX (P1_28), and my heated bed to be controlled by Y_MAX (P1_26). The heated bed works fine on P1_26 (AnalogWrite) but the X_MAX endstop (P1_28, just a digital pin... I can't figure out its mapping from the link you sent) is giving me trouble. 

Here is the pin file for my board:
`
/**
 * MKS SGEN-L pin assignments
 */

#define TEMP_CHAMBER_PIN 2 // works fine always
#define HEATER_CHAMBER_PIN P1_28 // bricks!!, but works with just the number "4" thanks to your suggestion
...
//
// Limit Switches
//
#define X_MIN_PIN          P1_29
#define X_MAX_PIN          P1_30// default: P1_28
#define Y_MIN_PIN          P1_27
#define Y_MAX_PIN          P1_31// default: P1_26
#define Z_MIN_PIN          P1_25
#define Z_MAX_PIN          P1_24
//
// Temperature Sensors
// 3.3V max when defined as an analog input
//
#define TEMP_0_PIN         0   // Analog Input A0 (TH1)
#define TEMP_BED_PIN       1   // Analog Input A1 (TB)
// used by chamber //#define TEMP_1_PIN         2   // Analog Input A2 (TH2)

//
// Heaters / Fans
//
#define HEATER_BED_PIN     P1_26// default: P2_05 [but i just learned these use the same 'hardware' channel? What might the consequences of this be?]
#define HEATER_0_PIN       P2_07
#define HEATER_1_PIN       P2_06
#ifndef FAN_PIN
  #define FAN_PIN          P2_04
#endif
...

`
Basically I just want to use the X_MAX pin for the heated chamber and the Y_MAX pin for the heated bed. The reason I'd really like to use these ports in particular is because they come with a built in JST connector and won't dislodge if something pulls on the board. It also matches my setup on my MKS BASE. Lastly AnalogRead pins are inherently valuable as thermistors. When I eventually use this board for a tool-switching printer, I'm effectively limited only by the number of thermistor input pins on my board. Using AnalogRead pins for this takes away from the number of tools I can use.

biocrypto730, 2019-10-01:
Here is a zip including my Configuration.h and Configuration_adv.h  and the full pin file. When I try to upload my entire codebase it breaks, but these are the only files i changed anyway. Thank you sir!
[zipper.zip](https://github.com/MarlinFirmware/Marlin/files/3677499/zipper.zip)

p3p, 2019-10-01:
As you say `HEATER_CHAMBER_PIN` is just a standard output so any `PX_XX` pin should work, the `TEMP_CHAMBER_PIN` is the one that needs the Analog channel number and you have that correct.

If there isn't a thermistor connected to the pin you assign to `TEMP_CHAMBER_PIN` Marlin will output an error and stop running for safety reasons, you can see which adc channel is connected to which PX_XX pin number at the link above to make sure its connected correctly.

For heaters Marlin doesn't use the hardware PWM so It shouldn't matter that you have pins connected that use the same hardware.

biocrypto730, 2019-10-01:
P1_28 is not listed at the link above.

biocrypto730, 2019-10-01:
This issue was caused because at startup the CHAMBER_TEMP pins are polled, and are returning a negative value. This negative value is causing the firmware to go into "safe mode" and not startup.

By changing the `MIN_CHAMBER_TEMP` and `MAX_CHAMBER_TEMP` values to lower and higher, respectively, the board boots.

Because this hasn't been reported anywhere else, I'm going to assume this was a hardware issue with the MKS SBASE L, and not a problem with marlin firmware. If someone else gets this issue then I'll reopen it.

closed 2019-10-01 by biocrypto730

iwasinnamuknow, 2019-10-13:
I'm also experiencing the same issue on LPC1768 board SKR 1.3. `HEATER_CHAMBER_PIN` is set to `P1_24` which is the Z+ endstop. I haven't managed to get it to boot yet with expanded max and min limits as mentioned previously. If I drop the min temp to -25 then I get a blank LCD and no activity. Without it I get an instant thermal error.