MarlinFirmware/Marlin #15438: New Repo: MarlinFirmware/Configurations

thinkyhead, 2019-10-01:
Using scripting, I constructed a new repository with all the current Marlin Configurations.

### [MarlinFirmware/Configurations](//github.com/MarlinFirmware/Configurations)

Each configuration is stored as a separate branch so that we can use `git merge` and `git rebase` to bring all the configurations up to date at once whenever new options are added.

In the current schema, the machine-specific configuration is stored in a `contrib` folder, and the current Marlin reference configuration is kept in the `marlin` folder. The `marlin` copy provides a historical copy of the unaltered Marlin configurations at the time of download and these are never altered in any of the other branches.

Putting each machine configuration into its own branch also makes it more straightforward to experiment with splitting up configurations by category … Starting with LCD options as a first test. It begins by propagating the new config file in its default form to all the children of `master`, and then the new file is updated in each branch via scripting.

Following the one-branch-per configuration approach permits each individual machine can have its own `README.md` file, `_Bootscreen.h`, `_Statusscreen.h`, and other custom additions. This approach also allows the curation of Marlin configurations as a project separate from the curation of the codebase.

Some minor inconveniences…. For Travis testing (and the upcoming Github CI) this means fixing the scripts so they pull down the configurations from the other repository during testing. The zip files are very small, and unzip simply replaces the copy. Branch names are identical to the old folder names, except spaces and dashes were replaced by underscores.

Once Travis CI is updated and working with the other repo, we can delete all example configs from Marlin and update the documentation to explain where examples can be found going forward.

RudyBenner, 2019-10-01:
I see how this works, I think. Very interesting.

reloxx13, 2019-10-02:
Heya,

Nice2Have: Backlink to Marlin repo (2.0.x branch?) in the readme files of the config repo for better navigation.

brianredbeard, 2019-10-17:
@thinkyhead Thinking through the overall guidance on how to use the repository, this leads me to two questions:

  1. Should changes which would be needed for a file like `platformio.ini` ever end up in this repository?
  1. What are the criteria for generating a new branch?

In regards to the second question, I see that some units have branches for both standard and BLtouch variants. I would assume (based on the LCD comments) that ideally this would be structured more as a "hardware feature" sub-directory within a branch.  Is that correct?

With a number of units there are variations which are also pseudo printer specific (for example, the new BigtreeTech STM32 based E3 DIP which is designed to be a drop in replacement for the original Melzi board) for which templatized configurations wouldn't be unreasonable.

Put differently if there aren't any concrete plans, are there any strictly opinionated areas to which adherence now could optimize contributions later?

boelle, 2019-11-24:
this one has servered its purpose for now so closing it
we can always reopen it when the new repro comes in full effect and more active

closed 2019-11-24 by boelle