MarlinFirmware/Marlin #15439: [BUG] (mks Robin compiling error cs_pin not declared)

fizga, 2019-10-01:
i try to build marlin in Visual studio en have this error : 

In file included from Marlin\src\feature\touch\../../inc/../HAL/./HAL_STM32F1/HAL.h:36:0,
                 from Marlin\src\feature\touch\../../inc/../HAL/HAL.h:26,
                 from Marlin\src\feature\touch\../../inc/MarlinConfig.h:30,
                 from Marlin\src\feature\touch\xpt2046.cpp:25:
Marlin\src\feature\touch\xpt2046.cpp: In static member function 'static void XPT2046::init()':
Marlin\src\feature\touch\xpt2046.cpp:40:26: error: 'CS_PIN' was not declared in this scope
   #define TOUCH_CS_PIN   CS_PIN
                          ^
Marlin\src\feature\touch\../../inc/../HAL/./HAL_STM32F1/fastio.h:37:53: note: in definition of macro '_SET_MODE'
 #define _SET_MODE(IO,M)       gpio_set_mode(PIN_MAP[IO].gpio_device, PIN_MAP[IO].gpio_bit, M)
                                                     ^~
Marlin\src\feature\touch\../../inc/../HAL/./HAL_STM32F1/fastio.h:41:35: note: in expansion of macro '_SET_OUTPUT'
 #define OUT_WRITE(IO,V)       do{ _SET_OUTPUT(IO); WRITE(IO,V); }while(0)
                                   ^~~~~~~~~~~
Marlin\src\feature\touch\xpt2046.cpp:50:3: note: in expansion of macro 'OUT_WRITE'
   OUT_WRITE(TOUCH_CS_PIN, HIGH);
   ^~~~~~~~~
Marlin\src\feature\touch\xpt2046.cpp:50:13: note: in expansion of macro 'TOUCH_CS_PIN'
   OUT_WRITE(TOUCH_CS_PIN, HIGH);
             ^~~~~~~~~~~~
Marlin\src\feature\touch\xpt2046.cpp:40:26: note: suggested alternative: 'SS_PIN'
   #define TOUCH_CS_PIN   CS_PIN
                          ^
Marlin\src\feature\touch\../../inc/../HAL/./HAL_STM32F1/fastio.h:37:53: note: in definition of macro '_SET_MODE'
 #define _SET_MODE(IO,M)       gpio_set_mode(PIN_MAP[IO].gpio_device, PIN_MAP[IO].gpio_bit, M)
                                                     ^~
Marlin\src\feature\touch\../../inc/../HAL/./HAL_STM32F1/fastio.h:41:35: note: in expansion of macro '_SET_OUTPUT'
 #define OUT_WRITE(IO,V)       do{ _SET_OUTPUT(IO); WRITE(IO,V); }while(0)
                                   ^~~~~~~~~~~
Marlin\src\feature\touch\xpt2046.cpp:50:3: note: in expansion of macro 'OUT_WRITE'
   OUT_WRITE(TOUCH_CS_PIN, HIGH);
   ^~~~~~~~~
Marlin\src\feature\touch\xpt2046.cpp:50:13: note: in expansion of macro 'TOUCH_CS_PIN'
   OUT_WRITE(TOUCH_CS_PIN, HIGH);
             ^~~~~~~~~~~~
Marlin\src\feature\touch\xpt2046.cpp: In static member function 'static uint16_t XPT2046::getInTouch(XPTCoordinate)':
Marlin\src\feature\touch\xpt2046.cpp:40:26: error: 'CS_PIN' was not declared in this scope
   #define TOUCH_CS_PIN   CS_PIN
                          ^
Marlin\src\feature\touch\../../inc/../HAL/./HAL_STM32F1/fastio.h:37:53: note: in definition of macro '_SET_MODE'
 #define _SET_MODE(IO,M)       gpio_set_mode(PIN_MAP[IO].gpio_device, PIN_MAP[IO].gpio_bit, M)
                                                     ^~
Marlin\src\feature\touch\../../inc/../HAL/./HAL_STM32F1/fastio.h:41:35: note: in expansion of macro '_SET_OUTPUT'
 #define OUT_WRITE(IO,V)       do{ _SET_OUTPUT(IO); WRITE(IO,V); }while(0)
                                   ^~~~~~~~~~~
Marlin\src\feature\touch\xpt2046.cpp:103:3: note: in expansion of macro 'OUT_WRITE'
   OUT_WRITE(TOUCH_CS_PIN, LOW);
   ^~~~~~~~~
Marlin\src\feature\touch\xpt2046.cpp:103:13: note: in expansion of macro 'TOUCH_CS_PIN'
   OUT_WRITE(TOUCH_CS_PIN, LOW);
             ^~~~~~~~~~~~
Marlin\src\feature\touch\xpt2046.cpp:40:26: note: suggested alternative: 'SS_PIN'
   #define TOUCH_CS_PIN   CS_PIN
                          ^
Marlin\src\feature\touch\../../inc/../HAL/./HAL_STM32F1/fastio.h:37:53: note: in definition of macro '_SET_MODE'
 #define _SET_MODE(IO,M)       gpio_set_mode(PIN_MAP[IO].gpio_device, PIN_MAP[IO].gpio_bit, M)
                                                     ^~
Marlin\src\feature\touch\../../inc/../HAL/./HAL_STM32F1/fastio.h:41:35: note: in expansion of macro '_SET_OUTPUT'
 #define OUT_WRITE(IO,V)       do{ _SET_OUTPUT(IO); WRITE(IO,V); }while(0)
                                   ^~~~~~~~~~~
Marlin\src\feature\touch\xpt2046.cpp:103:3: note: in expansion of macro 'OUT_WRITE'
   OUT_WRITE(TOUCH_CS_PIN, LOW);
   ^~~~~~~~~
Marlin\src\feature\touch\xpt2046.cpp:103:13: note: in expansion of macro 'TOUCH_CS_PIN'
   OUT_WRITE(TOUCH_CS_PIN, LOW);
             ^~~~~~~~~~~~
In file included from Marlin\src\feature\touch\../../inc/../HAL/./HAL_STM32F1/HAL.h:36:0,
                 from Marlin\src\feature\touch\../../inc/../HAL/HAL.h:26,
                 from Marlin\src\feature\touch\../../inc/MarlinConfig.h:30,
                 from Marlin\src\feature\touch\xpt2046.cpp:25:
Marlin\src\feature\touch\xpt2046.cpp:40:26: error: 'CS_PIN' was not declared in this scope
   #define TOUCH_CS_PIN   CS_PIN
                          ^
Marlin\src\feature\touch\../../inc/../HAL/./HAL_STM32F1/fastio.h:33:40: note: in definition of macro 'WRITE'
 #define WRITE(IO,V)           (PIN_MAP[IO].gpio_device->regs->BSRR = (1U << PIN_MAP[IO].gpio_bit) << ((V) ? 0 : 16))
                                        ^~
Marlin\src\feature\touch\xpt2046.cpp:123:9: note: in expansion of macro 'TOUCH_CS_PIN'
   WRITE(TOUCH_CS_PIN, HIGH);
         ^~~~~~~~~~~~
Marlin\src\feature\touch\xpt2046.cpp:40:26: note: suggested alternative: 'SS_PIN'
   #define TOUCH_CS_PIN   CS_PIN
                          ^
Marlin\src\feature\touch\../../inc/../HAL/./HAL_STM32F1/fastio.h:33:40: note: in definition of macro 'WRITE'
 #define WRITE(IO,V)           (PIN_MAP[IO].gpio_device->regs->BSRR = (1U << PIN_MAP[IO].gpio_bit) << ((V) ? 0 : 16))
                                        ^~
Marlin\src\feature\touch\xpt2046.cpp:123:9: note: in expansion of macro 'TOUCH_CS_PIN'
   WRITE(TOUCH_CS_PIN, HIGH);
         ^~~~~~~~~~~~
*** [.pio\build\mks_robin\src\src\feature\touch\xpt2046.cpp.o] Error 1

I can not find the solution !
here the file i use
[Marlin-2.0.x.zip](https://github.com/MarlinFirmware/Marlin/files/3676176/Marlin-2.0.x.zip)

tpruvot, 2019-10-02:
right, maybe my fault...  need #define TOUCH_CS_PIN           PB1

tpruvot, 2019-10-02:
made the PR #15445

fizga, 2019-10-03:
Thanks for the answer ! 
I made the change! Now build success but when i launch the robin motherboard the tft indicate complete And he stays like that

tpruvot, 2019-10-04:
According to the schematic https://github.com/makerbase-mks/MKS-Robin/blob/master/MKS%20Robin/Hardware/MKS%20Robin%20V2.3_001%20SCH.pdf

It might be required to manually set software SPI2 pins too, in the touchscreen section... 
```
#define TOUCH_SCK_PIN PB13
#define TOUCH_MISO_PIN PB14
#define TOUCH_MOSI_PIN PB15
```
If you could test and confirm.... 

what do you mean by the "tft indicates complete" ? could you do a picture ?

fizga, 2019-10-04:
I made the changes in the file xpt2046.cpp but unfortunately for the same result.
here is an image of the screen
[image]

I tried another version of MARLIN on this link
[https://github.com/makerbase-mks/MKS-Robin/tree/master/MKS%20Robin/firmware](url)
this works well I can activate almost all the options.
but when I turn on the BLtouch line or the line #define NUM_SERVOS 1
the screen remains black

tpruvot, 2019-10-04:
right, look like this "Complete" is displayed by the bootloader, doesnt look like Marlin code. Maybe after the SD upgrade is done... Did you try to reboot the board then, with and without the SD inserted ?

fizga, 2019-10-04:
I just tried but it's the same thing.
can be a problem on the TFT library ??
I think it comes from the code because when I test the code mentioned above it works well except for the BLTOUCH.

tpruvot, 2019-10-04:
Well the code you linked is not current marlin source code... can be anything... hard to compare both. But I seen weird pin definitions, like the "Custom SPI" section, affected to some pins required for the SD (SDIO)

fizga, 2019-10-04:
OK, but I'm not a coding expert!  I would like the original MARLIN to work but it is not!  the display shows nothing!  I browse the different files, I can not find the solution!

megamih, 2019-10-05:
> the code you linked is not current marlin source code
Hi @tpruvot,

I believe this code came from https://github.com/jmz52/Marlin/tree/MKS-Robin-Touch-POC
And I can confirm, that MKS Robin screen stopped to work for me June - July this year. I believe it is related to generalizing MKS Robin 320x240 color display into FSMC display (MKS Robin, Alfawise U20, JGAurora A5S, A1, etc.) (somewhere nearby commit bc58e678bde3399b6852ce5cd19e0031bb76eccb from 2019-07-30 10:00).
btw. If there is a way to gather some useful logs etc. - could you please point me to proper procedure?

Best regards,
Alex

fizga, 2019-10-06:
I also think that it comes from that
do you have the latest version that worked for you.  I could try it with the bltouch

tpruvot, 2019-10-06:
made a PR #15474 to fix the wrong pin definitions, could be related...

tpruvot, 2019-10-06:
I ordered the board, but will take some weeks to get it... Also i seen the Robin2 wiring seems very different on schematics... and not yet in Marlin sources

fizga, 2019-10-06:
Of course robin 2.4 is different from robin2
The processor is a 32f4xx instead of 32f1xx
Hope you find the solution
Thanks

tpruvot, 2019-10-06:
yep, noticed that in their sources, after... for the F4 (robin2) had some confusion with "v2.4" first...

megamih, 2019-10-06:
Hi @tpruvot,
First of all, thank you very much for your time and effort.
I pulled latest Marlin bugfix-2.0.x (fc6a0937b86f501b39263d7f4dfc7a4fa70b2636) and locally merged [PR #15474 ](https://github.com/MarlinFirmware/Marlin/pull/15474) - but still same story :( 'Complete' is shown on the screen after re-flash.

tpruvot, 2019-10-07:
ok, no other info over usb/serial during the boot ? Which size/model is your TFT ?

fizga, 2019-10-07:
My tft is mks robin tft_V2.0 3.2 inch
Never test upload firmware by usb! Will try this!

fizga, 2019-10-08:
ok when i try to upload have this
[image]
maybe I did something wrong !??

tpruvot, 2019-10-09:
You are using the file Robin.bin for the SD update, right?

boelle, 2019-10-12:
@fizga issue still there?

fizga, 2019-10-12:
yes still not solved

tpruvot, 2019-10-12:
Title and problem is no more related to your issue... maybe close this one and reopen a new one.. with current status

boelle, 2019-10-12:
suggestions for a better title would be? i can edit the title so it better shows about the problem

but yes i agree to hit close an create a new issue with correct title and status

fizga, 2019-10-12:
Ok i will close this and open a new one !

closed 2019-10-12 by fizga