MarlinFirmware/Marlin #15440: [FR] LULZBOT TOUCH UI for FSMC display

silentst0rm, 2019-10-01:
Is it possible to run LULZBOT UI on MKS Robin FSMC display?

tpruvot, 2019-10-02:
No, these external touch displays are standalone devices connected by a serial port to a Marlin board... So they have their own firmware...

thisiskeithb, 2019-10-02:
> No, these external touch displays are standalone devices connected by a serial port to a Marlin board... So they have their own firmware...

The TFTs on the MKS Robin boards are "dumb" TFTs that don't use their own processor: https://www.aliexpress.com/item/33013776323.html

marcio-ao, 2019-10-09:
The `LULZBOT_TOUCH_UI` can work on regular TFTs as long as you have a [FTDI FT800 or FT810 based EVE controller](https://www.ftdichip.com/EVE.htm). The LULZBOT UI is not serial based.

I would have to see the schematics for a MKS Robin to see if the motherboard uses a FTDI EVE chip.

TechsavvyCNC, 2019-12-05:
Yes that LCD and board would work with the lulzbot touch ui which is now called the TOUCH_UI_FTDI_EVE.

closed 2020-04-29 by thinkyhead