MarlinFirmware/Marlin #15442: [FR] Move text strings for Host Action Prompts

thordarsen, 2019-10-01:
### Description

Right now the host action prompt text strings (that get sent to Octoprint and the like) are hard coded into multiple .cpp files.
I believe the strings should be moved to the language header files to allow for translation, and for general code quality issues.

### Steps to Implement

I have a fork with an implementation here: https://github.com/thordarsen/Marlin/tree/HostPromptLanguage
To see the detailed file changes the pull request to my base fork can be found here: https://github.com/thordarsen/Marlin/pull/2
It works for pausing/stopping a print from the LCD. 

To do:
I haven't tested every changed string yet, and I know I missed replacing a few.
I should probably compare the size of the AVR binaries with/without the change.

InsanityAutomation, 2019-10-01:
Typically serial output is not translated as it may be parsed by scripts. In this case its intended for UI display, so maybe it warrants breaking convention... @thinkyhead Any issue with making these translatable and in language_en?

thordarsen, 2019-10-01:
I believe that even if these are considered serial output not to be translated, the strings should be placed in core/language.h with the other console messages. That way anyone writing a parser would have a single list to work from rather than having to hunt through multiple files of code.

closed 2020-06-30 by github-actions[bot]