MarlinFirmware/Marlin #15448: SKR1.3 TMC2130

Lino77, 2019-10-03:
Hello, following problem, I have a SKR1.3 with TMC2130 from Watterott on XYZ and E0. I think the error is not all axes to E0 run, attached my config.
In the spread cycle, the engine is screeching and in the STEALTHCHOP, it does not even do. Marlin's booth from last night.

SENDING:M122
		X	Y	Z	E
Enabled		false	false	false	true
Set current	400	500	600	500
RMS current	397	489	581	489
MAX current	560	689	819	689
Run current	12/31	15/31	18/31	15/31
Hold current	6/31	7/31	9/31	7/31
CS actual	6/31	7/31	9/31	7/31
PWM scale	24	32	22	66
vsense		1=.18	1=.18	1=.18	1=.18
stealthChop	true	true	true	true
msteps		16	16	16	16
tstep		max	max	max	max
pwm
threshold	121	121	658	188
[mm/s]		65	65	3	30
OT prewarn	false	false	false	false
OT prewarn has
been triggered	false	false	false	false
off time	3	3	3	3
blank time	24	24	24	24
hysteresis
-end		-1	-1	-1	-1
-start		1	1	1	1
Stallguard thrs	3	3	0	0
DRVSTATUS	X	Y	Z	E
stallguard				*
sg_result	0	0	0	242
fsactive
stst		*	*	*	*
olb
ola				*
s2gb
s2ga
otpw
ot
Driver registers:
		X	0x80:06:00:00
		Y	0x80:07:00:00
		Z	0xA1:09:00:00
		E	0x80:07:00:F2
Testing X connection... OK
Testing Y connection... OK
Testing Z connection... OK
Testing E connection... OK
ok P15 B4
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3685426/Marlin.zip)

komandrik, 2019-10-04:
@Lino77 try to register this parameter
`#define MINIMUM_STEPPER_PULSE 2`

Lino77, 2019-10-04:
okay thanks did work, but why then top 0 for tmc is a bit misleading

closed 2019-10-04 by Lino77