MarlinFirmware/Marlin #15454: [BUG] AZSMZ 12864 LCD doesn't initialize properly on first power on

alexyu132, 2019-10-04:
### Description

I'm currently running an AZSMZ board with its included AZSMZ 12864 LCD. The system is running off of a 24V PSU. When I power on my machine via the mains switch, the LCD sometimes stays blank and sometimes displays a very faint and uneven image. The printer still acts as if the display is working properly and I can navigate the menus and print just fine. Adjusting the contrast upward helps, but parts of the display are still very faint even at max contrast. Rebooting the board via the reset button always fixes the issue and never produces a blank display, but powering on with the mains switch always seems to result in a faint or blank display.

Can this issue be fixed in software, or is it exclusively a hardware issue? It seems possible that the initial voltage on mains power on might not be fully stable yet and is causing issues with the LCD's initialization.

#### Additional Information
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3688537/Configuration.zip)

Normal LCD:
[image]

Faint LCD:
[image]

alexyu132, 2019-10-06:
Updated title and description after tinkering around with it some more. I'd initially thought the problem was solely related to power cycling via the mains switch, but after further investigation the issue never happens when an SD card is not present in the LCD's reader. Could this be caused by some SPI conflict between the LCD and SD reader?

tvixen, 2019-10-07:
Try a 22uf over your 5v supply on the LCD.

boelle, 2019-10-12:
@alexyu132 still have an issue? is it a hardware problem and not a firmware problem?

alexyu132, 2019-10-12:
I haven't had a chance to test adding a capacitor to the LCD, but after doing some digging it appears that the 3.3v regulator is probably the culprit. Closing as it seems to be more likely a hardware issue.

closed 2019-10-12 by alexyu132