MarlinFirmware/Marlin #15455: [BUG] Megatronics V3.0 cannot read Ext. SD and onboard SD with Marlin bugfix 2.0

swatkxs, 2019-10-04:
Megatronics V3.0 cannot read Ext. SD and onboard SD with Marlin bugfix 2.0.
Hey guys, I have a Megatronics V3.0 and I have tried Marlin 1.1.7, 1.1.9 and 2.0.0, the #SDSUPPORT only work at 1.1.7.
here are the configration and configrationadv.

[config.zip](https://github.com/MarlinFirmware/Marlin/files/3688939/config.zip)

boelle, 2019-10-12:
@swatkxs is this still an issue?

boelle, 2019-10-24:
@swatkxs still having an issue? have you checked that our config is correct?

boelle, 2019-10-30:
closing this one as it seems @swatkxs has solved his issue

we can always reopen

closed 2019-10-30 by boelle

swatkxs, 2019-11-12:
sorry，i have waited for too long, I forgot to check this comment, but I still have this issue. And I have uploaded my config, could someone help me have a check on it?