MarlinFirmware/Marlin #15456: [BUG] Marlin-bugfix-2.0.x compile err

hn0550, 2019-10-04:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3689109/Configuration.zip)

RudyBenner, 2019-10-04:
looks like you included just the default configuration.h with your zip file?

hn0550, 2019-10-04:
> 看起来您的zip文件中仅包含默认的configuration.h？

Sorry, this is my configuration file
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3691368/Configuration.zip)

RudyBenner, 2019-10-04:
Linking .pio\build\DUE\firmware.elf
Checking size .pio\build\DUE\firmware.elf
Building .pio\build\DUE\firmware.bin
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [=         ]   9.2% (used 9032 bytes from 98304 bytes)
PROGRAM: [===       ]  25.5% (used 133780 bytes from 524288 bytes)
==================================== [SUCCESS] 
I did not change anything.

hn0550, 2019-10-04:
> Linking .pio\build\DUE\firmware.elf
> Checking size .pio\build\DUE\firmware.elf
> Building .pio\build\DUE\firmware.bin
> Memory Usage -> http://bit.ly/pio-memory-usage
> DATA: [= ] 9.2% (used 9032 bytes [...]
Thank you. What is your IDE version? OS version?
My is win10pro

RudyBenner, 2019-10-04:
I will try to compile on Arduino IDE 1.8.9 now.

hn0550, 2019-10-04:
> I will try to compile on Arduino IDE 1.8.9 now.
-------------------------------------------------------------
Thank you. This document is my error message.
[err.txt](https://github.com/MarlinFirmware/Marlin/files/3691484/err.txt)

RudyBenner, 2019-10-04:
No luck with Arduino IDE

RudyBenner, 2019-10-04:
My suggestion, provided you don't mind a challenge, learn to use VSCode. Bit of a steep learning curve. Instructions on marlinfw.org

RudyBenner, 2019-10-04:
Update your libraries too.

hn0550, 2019-10-04:
> VSCode
----------------------
MS Visual Studio Code?

sl1pkn07, 2019-10-04:
yes

hn0550, 2019-10-05:
> 链接.pio \ build \ DUE \ firmware.elf
> 检查大小.pio \ build \ DUE \ firmware.elf
> 构建.pio \ build \ DUE \ firmware.bin
> 内存使用情况-> [http://bit.ly/pio-memory-用法](http://bit.ly/pio-memory-usage)
> 数据：[=] 9.2％（使用98304字节中的9032字节）
> 程序：[===] 25.5％（使用524288字节中的133780字节）
> ================= ================== [成功]
> 我没有做任何更改。

-------------------------------------------
Thank you very much. I can compile it with VSCODE.  In addition, could you please tell me how the PlatformIO IDE USES the Arduino IDE download libraries? My English is poor, thank you

RudyBenner, 2019-10-05:
Can you close this?

closed 2019-10-05 by hn0550