MarlinFirmware/Marlin #15458: [BUG] Z-Axis raising 80 mm after homing

tijsrx, 2019-10-04:
Hi,

I downloaded the 1.1.9 Marlin for my dual extrusion printer.

I enabled Bltouch and Z-safe homing but changed nothing else concerning homing.
Before homing the Z-axis will raise 40mm, and after finishing homing it will raise itself 80mm. Can I adjust this height?

blueeagle69, 2019-10-04:
Try these (configuration.h)
#define Z_CLEARANCE_DEPLOY_PROBE   5 // Z Clearance for Deploy/Stow
#define Z_CLEARANCE_BETWEEN_PROBES  2 // Z Clearance between probe points
#define Z_CLEARANCE_MULTI_PROBE     2 // Z Clearance between multiple probes
#define Z_AFTER_PROBING           10 // Z position after probing is done

tijsrx, 2019-10-04:
thanks, will try tonight and report back

blueeagle69, 2019-10-04:
Just to add.
You shouldn't need to adjust;
#define Z_CLEARANCE_BETWEEN_PROBES 2 // Z Clearance between probe points
#define Z_CLEARANCE_MULTI_PROBE 2 // Z Clearance between multiple probes
These are my settings, as I wanted my Bltouch pin to stay closer to the bed during levelling to speed up the process. The main one for you should be the
#define Z_AFTER_PROBING
Hope it sorts it for you.

reloxx13, 2019-10-04:
> Do you want to ask a question? Are you looking for support? Please don't post here. Instead please use the Marlin Firmware forum at http://forums.reprap.org/list.php?415 or the Marlin Facebook Group https://www.facebook.com/groups/1049718498464482/.

blueeagle69, 2019-10-06:
Apologies @reloxx13 I saw the OP, knew the reason for his issues and jumped in.

boelle, 2019-10-12:
@tijsrx so this is fixed?

tijsrx, 2019-10-14:
yes, turned out to be badly calibrated Z-axis. E-steps were about 9 times too high

closed 2019-10-14 by boelle