MarlinFirmware/Marlin #15464: [FR] Relayed ("Outsourced") CaseLighting - Help/MoreInfo Wanted

p1ngb4ck, 2019-10-05:
### Description
For specific reasons I want to relay/outsource case-lighting to a 2nd Arduino (Nano/R3).

My specific usecase (just as example, so people could more understand the advantage of a relayed case-lighting system) :
I use an Arduino-Nano to control three parallel led-stripes (ws2812b). These stripes are glued into the extrusions of a delta-printer - it does not make sense to have additional cables from two of the stripe-ends back to the arduino, just to be able to make 1 stripe out of them. Thats why I need 3 data-pins in my custom parallel-stripes setup.
Which is absolutely no problem, if you use a second device (arduino) for that. I do not think it would make sense to sacrifice that many pins of the printer mainboard.
In short :
Having the option to "relay" lighting to a second device would be awesome - then, you`d need some kind of integration to marlin though, to still be able to control the lighting from marlin lcd / use event based lighting.

I thought about using a simple led-control-protocol using serial or pwm to minimize the necessary pin-count for the data-connection to the second device.

A PWM-controlled way would be extremely easy to achieve and would allow for a "preset" - system. I personally tested pwm reading of a pin controlled by Marlin using a real simple Nano-Firmware - it would be possible to control about 25 presets using different pwm states.

A serial connection to the Nano/R3 would require more pins, but would allow for a simple serial protocol to control the nano - set brightness, countless mode presets, colors.

I have a complete testing environment, and I am able and willing to participate and develop sth myself - I just want to ask here first, for :

Your opinions towards this in general.
Hints to policies/information regarding what to specifically obey when trying to contribute to Marlin
Info that would help me, not to re-invent the wheel and use available code as much as possible

Thanks for reading !
Best regards, Oliver

AnHardt, 2019-10-05:
The light-freaks will not be satisfied until Marlin is a [DMX](https://en.wikipedia.org/wiki/DMX512)-controller.

p1ngb4ck, 2019-10-05:
Dude it's indeed about NOT making Marlin the Controller. That's what it's all about. Not annoy the average Marlin user with bloated lighting functions, but instead just provide a way to "outsource" the lighting control.
But (maybe you agree).. for those who want to have lights it would be awesome no to have to use a 2nd additional screen for the light controller, and let the lights react on marlin events.
That's just why I asked how to realize a marlin => lights controller communication.
If done correctly, such a communication could be done with just 1(pwm) or 2 (rx/TX) pins, without bloating marlin with a lot of more light code for each "direct" solution.
And a nano clone doesn't cost much actually.

closed 2020-06-30 by github-actions[bot]