MarlinFirmware/Marlin #15466: [BUG] Anycubic kossel Linear Plus is much to slow

CorFlotscj, 2019-10-05:
Hello,

i update my Anycubic kossel Linear plus with the Marvin firmware and used some different Config.....h from thinkervers as setting.
the general problem is that the printer seam now to be much slower. 
I'M usind cura for slicing, which with the standard firmware was alway close to printing time, but now prints take 1.5 or even twice the time that cura expects.

(xyz cube with 3 shell 0.2 layer 20% infill, Cura ~22min, real 37min)

in increase acceleration and jerks already with no effect. 
(auto leveling and al the other suff works very well, just the speed is very bad, 
can someone give me a hint what i can still tweak to get it inline? 
best regards
Martin 

i asked this question in a lot of other places and dind got anything usefull so i thoght i got to the roots...

AnHardt, 2019-10-05:
Try less segments/second and/or a larger planner buffer.

hapklaar, 2019-10-05:
Is the print really taking longer than before, or just longer than what Cura predicts? You can enhance this by enabling acceleration control and jerk control, so Cura can use these values in its prediction.

CorFlotscj, 2019-10-06:
Hi again,
i tryed the tipps
segments/second : 

- 160 (wish were from the profile) to
-  200 (standard)
-  to 300

larger planner buffer.
 set was 16/16
test with 32/64 (more war not possbile, even with those i was warned that is could be instable

best results were 36,5 min from 38min -> no real improvment

i also flashed back to standard Anycubic  26,3 min with the same gcode.
with same setting Cura 3.6 and 4.2 are close to gether 25 to 26 minutes.
but there fo i got some ghosting :\

is there may be a mismatch of units for accelration and speed?

CorFlotscj, 2019-10-06:
in the meantime i've gone thru the conficuration.h and found a solution for me

it with the new way /calulation of the DEFAULT_AXIS_STEPS_PER_UNIT
i've modifined with below setting, back to 80/80/80/96 as hard setting and not calulations. 
and got the predicted 26min...

but i would have expecet if there are problems with this also the sizing should be effect but this was very good with Marling compared to before

maybe someone with more inside then me has an idea...

----

/**
 * Default Axis Steps Per Unit (steps/mm)
 * Override with M92
 *                                      X, Y, Z, E0 [, E1[, E2[, E3[, E4]]]]
 */
#define DEFAULT_AXIS_STEPS_PER_UNIT   { 80, 80, 80, 98 } // DaHai: This is no longer used in Marlin 1.1.9 - it is now calculated below (But extruder steps must be defined)
/**
 * Default Axis Steps Per Unit (steps/mm)
 * Override with M92
 *                                      X, Y, Z, E0 [, E1[, E2[, E3[, E4]]]]
 */
// variables to calculate steps
#define XYZ_FULL_STEPS_PER_ROTATION 200 // DaHai: These are the setting for the stock Stepper Motors, pulleys and Belts on the AnyCubic Kossel
#define XYZ_MICROSTEPS 16
#define XYZ_BELT_PITCH 2
#define XYZ_PULLEY_TEETH 20
// delta speeds must be the same on xyz
//#define DEFAULT_XYZ_STEPS_PER_UNIT ((XYZ_FULL_STEPS_PER_ROTATION) * (XYZ_MICROSTEPS) / double(XYZ_BELT_PITCH) / double(XYZ_PULLEY_TEETH))
#define DEFAULT_XYZ_STEPS_PER_UNIT (80)
//#define DEFAULT_AXIS_STEPS_PER_UNIT   { DEFAULT_XYZ_STEPS_PER_UNIT, DEFAULT_XYZ_STEPS_PER_UNIT, DEFAULT_XYZ_STEPS_PER_UNIT, 98.0 }  // (extruder steps defined) default steps per unit for Kossel (GT2, 20 tooth)
 ----

DkOuvE, 2019-10-06:
I've noticed this too. I put 120mm/s speed for all values in Cura but doesn't exactly seem like it's going at that speed.

AnHardt, 2019-10-06:
> i tryed the tipps
segments/second :

160 (wish were from the profile) to
200 (standard)
to 300

No. Wrong direction. Less not more!

There is no Marling-Firmware it's "Marlin"

If you don't see a change in size of the part after flashing new steps/mm-values you do have EPROM active and did not reset that to the new numbers.
Probably all your measured numbers are meaningless.

han9988, 2019-10-08:
I had some unpredicted results when I changed segments number on my delta so I thought the segements is still a work-in-progress. What do you think @thinkyhead ?

boelle, 2019-10-12:
@CorFlotscj still an issue? did you remember to overwrite eeprom? ie send M502 and then M500

sjasonsmith, 2019-10-15:
I recommend that you slice a file, then use the exact file with both the old and new firmware. Time the print with both firmwares, to see if there is _really_ a performance change. It is hard to act on "it feels slower" without having some solid numbers to compare.

There are so many variables that could impact performance, that without printing the same file with both firmwares, it is hard to know whether something in Marlin changed, or if it is something about the file/slicer.

I have a Kossel LP, but haven't used it for a while. With the Aug 24th state of bugfix-2.0.x I could get it moving much faster than the machine was rigid enough to handle.

CorFlotscj, 2019-10-15:
Hello,
i do have it still/again.
For some time the print timing was back to that cura expected

But since yesterday its back to too long. I did have this with my setting from above, and another try with 80 segments per secound, but the "new" step calculation. (basical what  AnHardt suggested). both worked fast.

But i had some issuses with the extruder setting. so if change the speps /mm for Extruder and did auto leveling, as i had also not the best  with a nother test print with bad adhesion and to much material extruded.

after this adjsutements timing  of the same file took longer. ~1.5 times londer

today i reflashed marvin fully with als levelung (except the Autoleveling, only those inside Delta calibation) and beeing still only at ~50% real speed.

flashed the AC standard "beta" firmware and the print is as fast as exepcted....

can board issued cause this? that there are some physical damages?
or that some calculations with the leveling take to long?
also during of the "test belt" the display "fizzeled" and showed only "nonsence" 

I'm very confused.

BR
Martin

AnHardt, 2019-10-15:
> today i reflashed marvin

Whatever marvin is - go there and ask for help.

Miss typing "Marlin" two consecutive times is looking careless to me.
If you are that careless with your Configs and observations i don't wonder about anything.

End of communication.

DkOuvE, 2019-10-17:
I'd test it too if it weren't for bug #15495.

boelle, 2019-10-26:
https://github.com/MarlinFirmware/Marlin/issues/15495 was fixed 4 days ago

so maybe there is news here too?

DkOuvE, 2019-10-27:
Can confirm. 120mm/s looks so much faster now.

boelle, 2019-10-27:
oki, will close this one as "fixed" then

we can always reopen

closed 2019-10-27 by boelle