MarlinFirmware/Marlin #15472: [BUG] Linear advance: actual extruder acceleration slower than retraction acceleration

BarsMonster, 2019-10-06:
### Description

I set Extruder acceleration and retract acceleration to very low values (10 mm/s2) to see how it works in slow motion. Retract/prime speed  is now expected, but linear advance is causing  veeeeery slow acceleration/deceleration when printing lines. I.e. extruder accelerate much slower during print moves, compared to retraction/prime moves. 

Retracts are generated by Cura 4.3 (7mm)

### Steps to Reproduce

1. M201 E10 ;Extruder acceleration
2. M204 R10 ;Retract acceleration
3. Print retraction test spikes

**Expected behavior:** 
Almost all moves are limited by extruder acceleration. I.e it almost always accelerates/decelerates at maximum speed of 10mm/s2.

**Actual behavior:** [What actually happens]

Print moves accelerate extruder ~10x slower compared to retraction, very slow acceleration/deceleration during print moves. XY speeds and accelerations are very high, and not limiting the process. 

#### Additional Information
Latest Marlin 2.0.x
Bowden extruder BMG
M900 K2

boelle, 2019-10-12:
please attach your 2 config files just as the issue template tells you to

boelle, 2019-10-22:
given the lack of activity and no config i will close this one

we can reopen when the configs are attached, without them there is no way for anyone to see what setup you have

closed 2019-10-22 by boelle