MarlinFirmware/Marlin #15479: Execute command while moving?

CaptainNico, 2019-10-07:
I have a unique use case for Marlin and I am wondering if it is possible to execute a command (in this case the servo M280 command) during a Z move? 

I want a servo to move to position while the Z axis is moving. Currently, I can only execute the M280 command before the Z starts to move or after its done moving and this is not ideal. The best case would be both starts to move at the same time, or z starts to move then while its moving the M280 command gets run.

Is this possible? Any workarounds to achieve this?

thinkyhead, 2019-10-08:
You must do a partial Z move up to where you want the servo to start its thing. Then issue the servo command. Then fulfill the remainder of the Z move. The `M280` command will not interrupt the move.

boelle, 2019-10-24:
will close this one as it was more a question and that has been answered

we can always reopen

closed 2019-10-24 by boelle