MarlinFirmware/Marlin #15480: [FR] Move 2 extruders independently

FNeo31, 2019-10-07:
### Description

Make possible to move 2 or more Extruders in different directions synchronized with XYZ and with the possibilities of homing and endstop on this extruders.

Objective:

Make possible to add 5 axis CNC features to work with Marlin in CNC Machines with 5 axis

Actually all variables arrays just have 4 stores for coordinates  [X,Y,Z,E] the idea is be possible to control like [X,Y,Z,E0,E1]

Anyone knows something about this possibility? or want to work in this kind of project?

thinkyhead, 2019-10-08:
Marlin will have support for up to 9 linear axes before the end of November.

FNeo31, 2019-11-26:
@thinkyhead do you have some news about this subject?

Roxy-3D, 2019-11-26:
@FNeo31  

Have you checked out the IDEX Duplication, Mirror Duplication and Independent control?

FNeo31, 2019-11-26:
No I'll check

FNeo31, 2019-11-28:
@Roxy-3D sorry but I didn't find the way to control and synchronize 5 independent axis. Maybe I'm looking at wrong spot, can you please say me how can I make that?

Roxy-3D, 2019-11-28:
It isn't everything you want...  But IDEX (Independent Dual Extruder) machines have two separate X-Axis.   Each extruder is on its own X-Axis.    So that gets you part way to where you want to be.

Those machines provide a lot of features.   Besides having multiple extruders so you can do color changes or print support material with dissoluble filament...   You can print duplicates of a part or a mirror image of a part at the same time the original part is printing.

FNeo31, 2019-11-28:
All the ways that I test I can't sync the 5axis.
The maximum number of independent axis that I could move synchronized was 4axis making different things at same time.

I started to expand the array to 5 positions has optional and the axis store the value  but just update after I change  from T0 to T1. I have some error on planner but I don't know where

FNeo31, 2019-11-30:
@Roxy-3D do you know how can I sync the 5 independent steppers, I tried a lot of things but I didn't get the solution all main code are based to read just 4 steppers at a time and with all the mirrors and duplications of each. Can you help please?

@thinkyhead  do you will really implement the control of 9 linear independent axis?

Roxy-3D, 2019-12-23:
> @Roxy-3D do you know how can I sync the 5 independent steppers, I tried a lot of things but I didn't get the solution all main code are based to read just 4 steppers at a time and with all the mirrors and duplications of each. Can you help please?

I will help you as much as I can...    But I'm away from my machines and am limited on what I can code and test.     Where are you now with this problem?

thinkyhead, 2019-12-24:
> do you will really implement the control of 9 linear independent axis?

Eventually. There are a lot of layers to the onion, as I am sure you can imagine.

FNeo31, 2019-12-26:
@Roxy-3D from beginning I tried 3 different ways.

At first I tried to use mixing feature, and work at some point. the trouble in this situation is that I can't move one axis to negative and other to positive at same time.

After see this trouble, I tried to create a 5 elements array just to use this feature. in this point of view to don't mess up with all features in Marlin. the trouble that I can't understand where is made the sync to the steppers (I'm not expert in code) so to update the 5th element position I need to change from tool 0 to tool 1. so isn't possible to sync the 4th and 5th axis at same time.

Now I'm trying on offline branch to change the main array of movements of Marlin to put 5 elements on it and solve all the mess that it makes to check if I can sync all.

I'll attach my CNC basic machine and the  5x Bed prototype pictures

@thinkyhead  Yes, I know that. This isn't easy to make this is why I ask for help. I'm not an expert on coding, I know the basics so I need a real programmers help.

** Merry Christmas **

FNeo31, 2020-02-23:
Hello @Roxy-3D and @thinkyhead 

I make some changes on marlin to be possible to control 2 extruders like rotary axis on my CNC.

I'm using one workaround of mixing extrusion.

At the moment I can put the axis moving in both ways but for some reason that I can't solve at the moment the board crash and shutdown with no connection to any screen or host. Normally I have troubles with moving just one of the axis 2 times  like A90; A45; without move C axis.

In simple mode I put current position of extruder at 0, make the mix and the calcs in abs values
and I think the trouble is the way that I'm change the DIR PIN State.

I tried a lot of ways (in indirection files, stepper files... and now I put directly on the function)
in stepper files I defined:

>**count_direction.e = 1;**

The code of the function is this (have a lot of dprint's for debug propuse):

```
#include "../../inc/MarlinConfig.h"

#if ENABLED(CNC_5X)

#include "../../gcode/gcode.h"
#include "../../feature/bedlevel/bedlevel.h"

#include "../../MarlinCore.h"
#include "../../module/planner.h"
#include "../../module/stepper.h"
#include "../../module/motion.h"
#include "../../module/tool_change.h"
#include "../../module/temperature.h"
#include "../../lcd/ultralcd.h"
#include "../../lcd/menu/menu.h"

extern bool fast_move;/*=false*/

extern int FLG5X;

/**
 *   M168 - 5 Axis Simple Convertion
 *
 *  X Y Z - position of cartesians axis
 *  A B C - position of rotary axis
 *  
 * 
 */

// External references

// Private functions

float X_5,
      Y_5,
      Z_5,
      IX_5,
      IY_5,
      IZ_5,
      OX_5,
      OY_5,
      OZ_5,
      A_5,
      B_5,
      C_5,
      IA_5,
      IB_5,
      IC_5,
      OA_5,
      OB_5,
      OC_5,
      E_5,
      I_5,
      J_5 = 0;

      

      

/**
 * M168
 */
void GcodeSuite::M168()
{
  
    if (parser.seen('X')) X_5 = parser.value_float();

      SERIAL_ECHO("X");
      SERIAL_ECHOLN(X_5);

    if (parser.seen('Y')) Y_5 = parser.value_float();

      SERIAL_ECHO("Y");
      SERIAL_ECHOLN(Y_5);
    
    if (parser.seen('Z')) Z_5 = parser.value_float();

      SERIAL_ECHO("Z");
      SERIAL_ECHOLN(Z_5);

    if (parser.seen('A')) A_5 = parser.value_float();
    
      SERIAL_ECHO("A");
      SERIAL_ECHOLN(A_5);

    if (parser.seen('B')) B_5 = parser.value_float();
   
      SERIAL_ECHO("B");
      SERIAL_ECHOLN(B_5);

    if (parser.seen('C')) C_5 = parser.value_float();
   
      SERIAL_ECHO("C");
      SERIAL_ECHOLN(C_5);

    IA_5=A_5-OA_5;
    IB_5=B_5-OB_5;
    IC_5=C_5-OC_5;

    FLG5X = 0;

    SERIAL_ECHO("IA");
    SERIAL_ECHOLN(IA_5);

    SERIAL_ECHO("IB");
    SERIAL_ECHOLN(IB_5);

    SERIAL_ECHO("IC");
    SERIAL_ECHOLN(IC_5);

    #if ENABLED (A_5x) && ENABLED(B_5x)

      if (IA_5>=0) {FLG5X += 0; I_5=IA_5;} else {FLG5X += 1; I_5=-IA_5;}

      if (IB_5>=0) {FLG5X += 0; J_5=IB_5;} else {FLG5X += 2; J_5=-IB_5;} 

    #endif

    #if ENABLED (A_5x) && ENABLED(C_5x)

      if (IA_5>=0) {FLG5X += 0; I_5=IA_5;} else {FLG5X += 1; I_5=ABS(IA_5);}

      if (IC_5>=0) {FLG5X += 0; J_5=IC_5;} else {FLG5X += 2; J_5=ABS(IC_5);}

    #endif

    #if ENABLED (B_5x) && ENABLED(C_5x)

      if (IB_5>=0) {FLG5X += 0; I_5=IB_5;} else {FLG5X += 1; I_5=-IB_5;}

      if (IC_5>=0) {FLG5X += 0; J_5=IC_5;} else {FLG5X += 2; J_5=-IC_5;} 

    #endif

    if (FLG5X==0 || FLG5X==2) {
      E0_DIR_WRITE( !INVERT_E0_DIR);
    }
    else
    {
      E0_DIR_WRITE( INVERT_E0_DIR);
    }
    
    if (FLG5X==0 || FLG5X==1) {
      E1_DIR_WRITE( !INVERT_E1_DIR);
    }
    else
    {
      E1_DIR_WRITE( INVERT_E1_DIR);
    }

    OA_5=A_5;
    OB_5=B_5;
    OC_5=C_5;

    E_5=(I_5+J_5);
       
    mixer.set_collector(0, I_5);
    mixer.set_collector(1, J_5);
    mixer.normalize(0);

    current_position.e = 0;

    SERIAL_ECHO("E_5");
    SERIAL_ECHOLN(E_5);

    SERIAL_ECHO("current.e");
    SERIAL_ECHOLN(current_position.e);
    
    destination =  {X_5,Y_5,Z_5,E_5};

    SERIAL_ECHO("destination.e");
    SERIAL_ECHOLN(destination.e);
    
      prepare_move_to_destination();
      
      planner.synchronize();   

}

#endif

```

Can you help?
I'm working on a SKR Pro v1.1 and test on SKR v1.3 and have the same crashes

Roxy-3D, 2020-03-13:
I apologize...  I didn't see your posts...    Where are you at?    Can you give us an update of your progress and what is currently a problem.   I'll try to coach you.

FNeo31, 2020-03-13:
@Roxy-3D like you can see on the post above I'm trying to workaround with the mixing extruders.

Reason: I don't know where Marlin are synchronizing the vector xyze to put each calculated point on buffer or if it's working like that. (this is my main trouble)

I make a workaround by mixing extruder making the direction control changing this on stepper.cpp (like you can see on code below) the machine moves on correct way but have some crashes and shuts down.

So I tried to change the direction directly on M168 and the result is the same.

If I know where the synchronization is made I can create the vector xyzee (for example) and change all chain.

This is the change made on stepper.cpp (first try commmented):
`    
    #if DISABLED(LIN_ADVANCE)
    
    #if ENABLED(MIXING_EXTRUDER) && DISABLED (CNC_5X)
       // Because this is valid for the whole block we don't know
       // what e-steppers will step. Likely all. Set all.
      if (motor_direction(E_AXIS)) {
        MIXER_STEPPER_LOOP(j) REV_E_DIR(j);
        count_direction.e = -1;
      }
      else {
        MIXER_STEPPER_LOOP(j) NORM_E_DIR(j);
        count_direction.e = 1;
      }
      
    #elif ENABLED(MIXING_EXTRUDER) && ENABLED (CNC_5X)
       // M168 Send always positive values      

      count_direction.e = 1;
      
      /*
      if (FLG5X==0){
      E0_DIR_WRITE(!INVERT_E0_DIR);
      E1_DIR_WRITE(!INVERT_E1_DIR);
      }
      else if (FLG5X==1) {
      E0_DIR_WRITE(INVERT_E0_DIR);
      E1_DIR_WRITE(!INVERT_E1_DIR);
      }
      else if (FLG5X==2) {
      E0_DIR_WRITE(!INVERT_E0_DIR);
      E1_DIR_WRITE(INVERT_E1_DIR);
      }
      else if (FLG5X==3) {
      E0_DIR_WRITE(INVERT_E0_DIR);
      E1_DIR_WRITE(INVERT_E1_DIR);
      }
      */
      
    #else
      if (motor_direction(E_AXIS)) {
        REV_E_DIR(stepper_extruder);
        count_direction.e = -1;
      }
      else {
        NORM_E_DIR(stepper_extruder);
        count_direction.e = 1;
      }
    #endif
  #endif // !LIN_ADVANCE `

This is the actual M168 that I have:

`~
#include "../../inc/MarlinConfig.h"

#if ENABLED(CNC_5X)

#include "../../gcode/gcode.h"
#include "../../feature/bedlevel/bedlevel.h"

#include "../../MarlinCore.h"
#include "../../module/planner.h"
#include "../../module/stepper.h"
#include "../../module/motion.h"
#include "../../module/tool_change.h"
#include "../../module/temperature.h"
#include "../../lcd/ultralcd.h"
#include "../../lcd/menu/menu.h"

extern bool fast_move;/*=false*/

extern int FLG5X;

/**
 *   M168 - 5 Axis Simple Convertion
 *
 *  X Y Z - position of cartesians axis
 *  A B C - position of rotary axis
 *  
 * 
 */

// External references

// Private functions

float X_5,
      Y_5,
      Z_5,
      IX_5,
      IY_5,
      IZ_5,
      OX_5,
      OY_5,
      OZ_5,
      A_5,
      B_5,
      C_5,
      IA_5,
      IB_5,
      IC_5,
      OA_5,
      OB_5,
      OC_5,
      E_5,
      I_5,
      J_5 = 0;

      

      

/**
 * M168
 */
void GcodeSuite::M168()
{
  
    if (parser.seen('X')) X_5 = parser.value_float();

      SERIAL_ECHO("X");
      SERIAL_ECHOLN(X_5);

    if (parser.seen('Y')) Y_5 = parser.value_float();

      SERIAL_ECHO("Y");
      SERIAL_ECHOLN(Y_5);
    
    if (parser.seen('Z')) Z_5 = parser.value_float();

      SERIAL_ECHO("Z");
      SERIAL_ECHOLN(Z_5);

    if (parser.seen('A')) A_5 = parser.value_float();
    
      SERIAL_ECHO("A");
      SERIAL_ECHOLN(A_5);

    if (parser.seen('B')) B_5 = parser.value_float();
   
      SERIAL_ECHO("B");
      SERIAL_ECHOLN(B_5);

    if (parser.seen('C')) C_5 = parser.value_float();
   
      SERIAL_ECHO("C");
      SERIAL_ECHOLN(C_5);

    IA_5=A_5-OA_5;
    IB_5=B_5-OB_5;
    IC_5=C_5-OC_5;

    FLG5X = 0;

    SERIAL_ECHO("IA");
    SERIAL_ECHOLN(IA_5);

    SERIAL_ECHO("IB");
    SERIAL_ECHOLN(IB_5);

    SERIAL_ECHO("IC");
    SERIAL_ECHOLN(IC_5);

    #if ENABLED (A_5x) && ENABLED(B_5x)

      if (IA_5>=0) {FLG5X += 0; I_5=IA_5;} else {FLG5X += 1; I_5=-IA_5;}

      if (IB_5>=0) {FLG5X += 0; J_5=IB_5;} else {FLG5X += 2; J_5=-IB_5;} 

    #endif

    #if ENABLED (A_5x) && ENABLED(C_5x)

      if (IA_5>=0) {FLG5X += 0; I_5=IA_5;} else {FLG5X += 1; I_5=ABS(IA_5);}

      if (IC_5>=0) {FLG5X += 0; J_5=IC_5;} else {FLG5X += 2; J_5=ABS(IC_5);}

    #endif

    #if ENABLED (B_5x) && ENABLED(C_5x)

      if (IB_5>=0) {FLG5X += 0; I_5=IB_5;} else {FLG5X += 1; I_5=-IB_5;}

      if (IC_5>=0) {FLG5X += 0; J_5=IC_5;} else {FLG5X += 2; J_5=-IC_5;} 

    #endif

//try to change E steppers directions from M168

    if (FLG5X==0 || FLG5X==2) {
      E0_DIR_WRITE( !INVERT_E0_DIR);
    }
    else
    {
      E0_DIR_WRITE( INVERT_E0_DIR);
    }
    
    if (FLG5X==0 || FLG5X==1) {
      E1_DIR_WRITE( !INVERT_E1_DIR);
    }
    else
    {
      E1_DIR_WRITE( INVERT_E1_DIR);
    }

    OA_5=A_5;
    OB_5=B_5;
    OC_5=C_5;

    E_5=(I_5+J_5);
       
    mixer.set_collector(0, I_5);
    mixer.set_collector(1, J_5);
    mixer.normalize(0);

    current_position.e = 0;

    SERIAL_ECHO("E_5");
    SERIAL_ECHOLN(E_5);

    SERIAL_ECHO("current.e");
    SERIAL_ECHOLN(current_position.e);
    
    destination =  {X_5,Y_5,Z_5,E_5};

    SERIAL_ECHO("destination.e");
    SERIAL_ECHOLN(destination.e);
    
      prepare_move_to_destination();
      
      planner.synchronize();   

}

#endif `

Roxy-3D, 2020-03-14:
There are a few of us that want to evolve the code base to handle mixing extruders on IDEX machines.   What we initially envision is a Dual Carriage machine that has a 3 color mixing extruder on each carriage.   One of our hardware developers has a board that he designed to help us with that effort.   It has support for 12 stepper motors.   (3 for each mixing extruder and then some more to move the build plate around and to position the Z-Axis.)

All of Marlin's motion is based on Gerbil.    A given move potentially changes (X,Y,Z) and also moves the extruder's stepper motor.    Each move has an acceleration ramp, a constant speed period, and then a de-acceleration ramp.    And then these moves are processed to connect them so where possible, the carriage is not slowing down and speeding up.   Instead, the speed of each axis is connected and smoothed out.

The only difference with the mixing extruders is the extruder is not a single stepper motor.   Instead, the movement is divided across how ever many stepper motors is contributing to the 'mix'.

If you search the code base for MIXING_EXTRUDER you will see all the conditional code to handle mixing extruders.  (Maybe use AstroGrep...   That works well and lets you edit any location you click on)   What specifically do you want to 'synchronize'?    Any given move that is scheduled already has each components X,Y,Z and E movement synchronized for all phases of the movement.   The ramp up in speed...  the constant speed...   and the de-acceleration ramp down.    All phases are already synchronized to each other.

FNeo31, 2020-03-19:
Hello @Roxy-3D,

The idea is the jump to 5 axis CNC and 3D Printing.

I want to add 2 rotary axis synchronized with all other axis.

To the CNC Mode I can make one workaround instead of develop all the code to create more 3 independent axis with endstops.

I'm not a Programmer, this is the reason that I'm asking for help, I can read code and make some basic programs. At the moment Marlin is working with all parametric variables and this is more complicated for me to track where the functions are running or not.

I followed you advice and search for MIXING_EXTRUDER and isn't suppose to have the crash that I have when I'm using the 5 axis mode. I checked, and in my point of view I have 2 or 3 things that can make the crash.

- overflow on mixing (don't make too much sense because in I use the base feature of mixing I don't have the crash)

-overflow on mixing calcs (when I test with the base feature I send M163 and after M164 and in the end the line with E steps <like: G1 X10 Y20 E360> on M168 I'm changing all this at each entry of M168 <like: M168 X10 Y20 A90 C270 - that represents a mix of 90/360 to E0 and 270/360 to E1>)

-dubious information on steppers directions (because I'm forcing it on M168 and I don't know if it's in use for some calcs)

At the moment exist one 5 axis version of Super Gerbil, is it easy to compare and implement?

Can you help?

Roxy-3D, 2020-03-23:
I'm confused...   Are you talking about adding a rotary tool?   Or do you mean you want to point the 3D-Printer's nozzle in a GCODE specified direction?    The existing code base pretty much already lets you do either of those things.

FNeo31, 2020-03-23:
Rotary Axis not rotary tool, for a simple explanation I want to move the machine like this fully synchronized:

G0X0Y0Z0A0C0
G0X0Y0Z0A-90C90
G0X0Y0Z0A0C0
G0X0Y0Z0A90C-90
G0X0Y0Z0A0C0

Axis A can be E0 (for example)
Axis C can be E1 (for example)

Can you understand like that?
And after in next step put endstops on them

Roxy-3D, 2020-03-23:
We have had people do rotary axis's before...   And Marlin works just fine.
Here is an example where somebody decided to 3D-Print on an Easter Egg:

https://github.com/MarlinFirmware/Marlin/issues/10218#issuecomment-377063855

FNeo31, 2020-03-23:
@Roxy-3D like this: 

https://youtu.be/w8Fl8L4yk8M

thinkyhead, 2020-03-24:
I have a contract to add more axes to Marlin. I'm starting by first porting the `HANGPRINTER` feature —which adds a single extra linear axis— from Marlin 1.1 to Marlin 2.0. For your purposes, and if you're in a hurry, you may want to backtrack to Marlin 1.1 and try extending `HANGPRINTER` to add more linear axes as the starting point. I won't be done with this port of `HANGPRINTER` to Marlin 2.x anytime soon.

FNeo31, 2020-03-24:
@thinkyhead Have I the possibility of compile Marlin 1.1 for SKR Pro V1.1?

FNeo31, 2020-03-24:
@thinkyhead and @Roxy-3D I have good news, not so good for the for development because it's a workaround and at the moment just work for CNC Machine, but can also be available for 3D printers for only one extruder.

It's working moving in 5 axis without crashes at the moment.

Can I make a pull request with CNC drilling cycles, CNC Menu and CNC_5axis together?

I'm waiting for your decision to make it

Roxy-3D, 2020-03-24:
Yes!!!   Very cool!!!    

Really...   Most of the support is already in place to do that.   However, that print was using arcs and not straight lines.   We would have to warm over the arc code to handle a few more axis.

Roxy-3D, 2020-03-24:
> Can I make a pull request with CNC drilling cycles, CNC Menu and CNC_5axis together?

It would be best to break out the various components just because it makes looking for problems in the revision history much easier.

FNeo31, 2020-03-24:
@Roxy-3D até the moment I make the changes of drilling cycles and 5 axis to put on the same pull request. It's easy to understand the changes because just add variables on main config files, add the codes on gcode reader, PIN change conditions on SKR board and don't change nothing on main Marlin so if the feature are disabled on config files we can isolate each feature. I think that isn't a problem

About the arcs, like normally 5axis programing is in little lines and don't contemplate arcs (on industrial machines) and this 5axis feature at the moment isn't modal so need to put the M168 before the position instead of G1, G0, G2 or G3 so in my opinion doesn't need to change arc feature.

I will make the pull request for this 2 now and after another for CNC Menu

Roxy-3D, 2020-03-24:
> this 5axis feature at the moment isn't modal so need to put the M168 before the position instead of G1, G0, G2 or G3 so in my opinion doesn't need to change arc feature.

From your comments I think we would want to put in a sanity check that disallows the Arc support if the extra rotary axis's are enabled.   But if this is working with small line segmentation, we can add Arc support latter.

FNeo31, 2020-03-24:
> But it would seem we should figure out how to make this work with the standard G0's and G1's.

Yes, I want this to make it modal.
At the moment I need to put endstops working on them. 
And after the RTCP mode.
Because the feed used on G1 and G0 are used by M168 and you can make a move on G0 G1 G2 G3 between M168 codes and don't have any worry. 
The machine knows the position.
And don't have any trouble with arcs, I have the arcs enabled on my config

Roxy-3D, 2020-03-24:
> At the moment I need to put endstops working on them.

G28 won't take much to home the extra axis's.    But we are probably going to need a number of Configuration.h options to control the homing.  Besides the direction and whether it is home to min or max,  it is very possible the generic machine with this will need to home the rotary axis's at a specific point in time.   Like after the X home, but before the Y home.   I don't know, but we should give some thought on what options are likely to come up as needed.

> And don't have any trouble with arcs, I have the arcs enabled on my config

Yes, but you probably are not sending any G2's or G3's in your GCode.   It won't take much, but we should probably not allow the firmware to be built with Arc support if rotary axis's are turned on.

What are you using to slice your object?   I'm wondering what tool chains exist to generate the GCode you are using.

FNeo31, 2020-03-25:
> Yes, but you probably are not sending any G2's or G3's in your GCode. It won't take much, but we should probably not allow the firmware to be built with Arc support if rotary axis's are turned on.

I'm making parts with M168 and G2 and G3 on same file 

Like I post, this isn't like I want to 3D printer. Because I make a workaround instead of change all main Marlin to add axis.

But for CNC Milling I'm using CAM software and not a slicer.

And is possible to make 3D print from same way

Without 5x I'm using Simplify 3D

And yes I know that I need to put some conditions for home the rotary axis this is why I said that need to put endstops working and after that the remaining things

It's a hard work for a person without too much time -.-

But with "baby steps" I think that is possible sooner

FNeo31, 2020-03-28:
This is the expected motion. And the first target: continuous 5axis motion

Vimeo: https://vimeo.com/401730557?ref

@thinkyhead  and @Roxy-3D can see if you want :)

FNeo31, 2020-03-31:
Hello @Roxy-3D and @thinkyhead,

I have a issue on the step number 2 of this project (implementation of endstops).

I don't know why but I create the variables for include the E_AXIS (because I'm using mixing extruder and didn't move any axis if I try directly on E0 and E1)

So at the moment I have the machine move the axis for the right side but don't trigger the on endstop. 
So it's possible that I forgot something. 

The strange thing for me is the M119 (Endstops report) give me the right information.

Try to explain:
I create one function where I make a directions workaround and ask for: 

homeaxis(E_AXIS);

So the machine move to right side but don't trigger and get halted because didn't get the endstop when it's really triggered and when check with M119.

The function comes this report:

X:0.00 Y:0.00 Z:280.00 A:0.00 C:0.00
E:270.00 Count X:0 Y:0 Z:224000
Entrou no M167  **get in M167**
Calculou HomeDir5X: 3 **make good calc of directions**
Fez o mix A ** make the mix E0 1 and E1 0 to move just E0**
**(here is the call of homeaxis(E_AXIS);**
echo:busy: processing
echo:busy: processing
Error:Printer halted. kill() called!
//action:poweroff

Restart Machine and test endstops:
M119
Reporting endstop status
x_min: TRIGGERED
y_min: TRIGGERED
z_min: open
z_max: TRIGGERED
e_min: TRIGGERED
e_max: TRIGGERED
z_probe: open

M119
Reporting endstop status
x_min: TRIGGERED
y_min: TRIGGERED
z_min: open
z_max: TRIGGERED
e_min: open
e_max: open
z_probe: open

Can you help?

Roxy-3D, 2020-04-05:
End stops should be very easy to add for the extra axis's.    The big problem I have thinking about this is we need a tool chain that can actually slice the object and generate motion commands for the extra axis's.    

This is important for two reasons.   First, if there is not support for this that the average user can use on their parts, this is a lot of complexity that will never get used.     The second reason it is important to have a working tool chain to generate the motion is we have no way to test the code that gets implemented if we can't run various objects through the process and see how the machine behaves.

How hard was it to get that GCode that was used in the video with the globe being printed?   Can you publish the starting globe .STL file, and step by step directions to get the 5-axis GCode generated?    We need the ability for an average user to generate meaningful 5-Axis GCode.   If we have that, I would be willing to spend time and organize a larger effort to get the whole code base warmed over for full support.

FNeo31, 2020-04-05:
Dear @Roxy-3D ,
At first, the endstops I add and the axis but didn't stop the move when the endstop are triggered. If you or someone else could help it's a good thing.

At Second, 5 axis code at the moment isn't just put the model and slice. In CNC machining world we can automatically program one part in 3 axis like 3D printing but in 5axis we have a lot of variables so you need to say to the the software what you want to make and in each zone and what's the kind of path that you want to make.

I developed this feature thinking about CNC world because it's a awesome feature to a controller firmware (at the moment in my case works well, isn't perfect but all new features aren't perfect on the begging) and it's a first step for future implementation of 5 axis 3D printing on Marlin. 
I don't know if the Marlin's target people isn't the CNC and if don't want to make able this path.
If Marlin development don't want to add new features of CNC to the firmware I understand and I'm sorry for taking your time with CNC pull requests.
I'm not ask for develop things I'm just ask to say how can I develop things for you.

What I'm asking now about the endstops is because get easy to make the zero's of this 2 new axis and I don't need to make the alignment manually all the time that I restart the machine and like I add similar way to the X axis don't make too much sense this didn't work, so I just wanted to know the rules of implementation of endstops and where I need to change.

FNeo31, 2020-04-06:
@DerAndere1 can you help me to know where I need to change to get the endstop pin working on a extra pin?

If I have that I can complete the endstop trouble.

I add the variables and checks on:

[image]

and report the trigger in M119 but don't stop the moves

DerAndere1, 2020-04-06:
@FNeo31 You can have a look at my branch https://github.com/DerAndere1/Marlin/tree/Marlin2ForPipetBot_v1 to see how I added endstops and G28 auto-homing to the E axis in pre-Marlin-2.0.0 code. You can enable an issue tracker in your fork: see https://help.github.com/en/github/managing-your-work-on-github/disabling-issues. In your the issue tracker of your fork I could discuss with you without cluttering this MarlinFirmware bug tracker.

FNeo31, 2020-04-06:
Check all my changes here:
[https://github.com/FNeo31/Marlin](url)
I have one work around solution but I don't want this solution because needs external hardware

DerAndere1, 2020-04-07:
In stepper.h and stepper.cpp make sure the type of endstop_trigsteps is big enough to support the additional E endstops. Also, add support for software endstops on the E axis: search for all occurances of MIN_SOFTWARE_ENDSTOP_X and MAX_SOFTWARE_ENDSTOP_X and add similar code for MIN_SOFTWARE_ENDSTOP_E and MAX_SOFTWARE_ENDSTOP_E (which has to be defined in Configuration.h)

FNeo31, 2020-04-07:
@DerAndere1 Thank you a lot,
I checked in your branch and my biggest trouble was the length of the variables instead of uint8_t I need at least uint16_t.
The changes of xyz to xyze I will check tomorrow detailed your changes and will change what is in fault.
About your branch to add 3 more axis, you just want to make it asynchronous or you want to have synchronized 6 axis too?

DerAndere1, 2020-04-07:
@FNeo31 With my 6 axis branches, you have different options. Tested default behaviour is synced movement in the sense that G1 X5 Y10 Y15 I20 J25 starts and stops the movement of all axes simultaneously. I would love if someone could add tool center point type I (G43.4) and type II (G43.5).

FNeo31, 2020-04-07:
This is called RTCP this is the thing that I want to add on next step

FNeo31, 2020-04-07:
@DerAndere1 I checked your Branch and need to congrats you for the good work. 

I Checked one thing that is not so good for RTCP implementation but it's easy to make the changes.

The standard axis isn't IJK like maybe you know but ABC.
Normally A rotate around X, B around Y and C about Z with right hand rule.

This is the standards and for a easy future use I think is a good way to work and it's simple to edit. This for RTCP if we come with the standards we can make the rules like standard rules. (Easier to implementation and for CAM Postprocessors)

So for 5axis machine we just need 5drivers so XYZ + AB or AC or BC depending of the kinematics so if we allow A and C (I and K) like you have we need to activate the B (J) so my opinion is to make a condicional ti just work like 'A' and 'C' jumping do internal axis 'I' and 'J'.

What's the errors that you detect on your branch? Can I help you in this errors? How can I help 
**{I have a machine made by me working with SKR Pro and TMC2209 in UART mode and with dual Z driver's I can easily test the firmware, my kinematic is table-table with A(X+)C(Z+) }**

DerAndere1, 2020-04-07:
If you are good at debuging code, you can compile my branch bf2_6axis_dev13, test it on your machine. and contribute to https://github.com/DerAndere1/Marlin/issues/7. If you rather want to solve the naming issue (https://github.com/DerAndere1/Marlin/issues/3), it is probably better to compile my working branch bf2_6axis_dev with configs adjusted to your needs and send some G1 commands for testing. If not relevant for this thread, please continue the discussion at https://github.com/DerAndere1/Marlin/issues. Thanks for your interest

Roxy-3D, 2020-04-07:
> So for 5axis machine we just need 5drivers so XYZ + AB or AC or BC depending of the kinematics so if we allow A and C (I and K) like you have we need to activate the B (J) so my opinion is to make a condicional ti just work like 'A' and 'C' jumping do internal axis 'I' and 'J'.

I think we would be best off keeping the 6 axis support.   That seems a lot more general and more people would be able to use it.   People can always leave an axis unpopulated if they don't have the extra hardware on their machine.

I am still concerned about how we get the tool chain support in place so people can actually use it.   And so that developers can test what they have done.

DerAndere1, 2020-04-08:
I made the number of NON_E_AXES and the axis names configurable, see my branch [bf2_6Xdev_axis_rename](https://github.com/DerAndere1/Marlin/tree/bf2_6Xdev_axis_rename) (untested). In Appendix A of [Øyvind K. Grutle 2015, master thesis, University of Oslo](http://www.robotikk.com/student/projects/OyvindKallevikGrutle3Dprint_5ax_ELDAT/) you can find source code for a program that can create 5axis G-code scripts for different shapes which can be used to test/develop 5 axis 3D-printers. 
*******
*******
**[unpaid advertisement]** 
I read that Autodesk Fusion 360 has free license options for personal use or startups. It is supposed to integrate a HSMworks-based [Multi-axis CAM workspace](https://forums.autodesk.com/t5/fusion-360-support/accessing-millturn-and-5-axis-machining/td-p/8486543) called ["Manufacture"](https://knowledge.autodesk.com/support/fusion-360/troubleshooting/caas/sfdcarticles/sfdcarticles/Manufacturing-Workspace-is-missed-in-Fusion-360.html). In addition, access  to "Advanced Manufacture Extensions", including an "Additive Manufacturing" extension can be [unlocked for a fee](https://www.autodesk.com/products/fusion-360/blog/fusion-360-extensions/). The  Autodesk post processor library contains many 5 axis posts written in Javascript. They can be downloaded as [editible text files (with file extension .cps) for free](https://knowledge.autodesk.com/support/autodesk-hsm/learn-explore/caas/sfdcarticles/sfdcarticles/How-to-make-a-3-axis-post-processor-from-the-HSM-post-processor-library-support-4-or-5-axis-motion.html).

DerAndere1, 2020-04-09:
@FNeo31 Regarding the E endstop issue, I got updated code from user @rnj1 and rebased it onto bugfix-2.0x. See the diff below for my proposal how you might get endstops working (I did not test). Just pick the changes you like. https://github.com/FNeo31/Marlin/compare/bugfix-2.0.x...DerAndere1:PR_fix_e_homing.

astorun, 2020-05-09:
Hi all,

Any news on supporting linear axes control on Marlin? I am trying to build a rotary filament dryer with 5 filament spool holder controlled via Marlin. I am repurposing the extruder motors to turn the filament spools simultaneosly and independently and using hotend controls to heat the custom 5 seperate pcb heatbeds I built.

FNeo31, 2020-05-09:
Hello @astorun,
For your question the answer is yes and no.

Yes it's possible to move another 2 or 3 motors independently. 

I just test on my branch and it's working. But I used on CNC machine not 3D printer. So the workaround that I made can't be a good thing for you because affect the main extruders.
Otherwise you can use @DerAndere1 branch that have really independent 3 axis.
This have some limitations, but this was what all interested people get at the moment with the low support that we have.

I don't have too much time, so I didn't make a full test of @DerAndere1 branch because I need to put the machine producing some parts so needed to stop with the tests.

We moved all things to the people with interest to here:
https://github.com/DerAndere1/Marlin/issues
To be easier to check all the codes and make implementations

ellensp, 2021-09-10:
Marlin now has @DerAndere1 6 axies +tools.  This should for fill this FR

closed 2021-09-10 by ellensp