MarlinFirmware/Marlin #15484: [BUG] (Marlin 2.0 HEATER_0_MAXTEMP and BED_MAXTEMP are no longer observed)

NickMilton, 2019-10-07:
### Description HEATER_0_MAXTEMP and BED_MAXTEMP are no longer observed

Board BIGTREE_SKR_PRO V1.1
Marlin 2.0 (30th Sept 2019)
Cartesian printer

Maximum hot end and bed temperatures appears to be limited to those defined as Preheat Constants (eg PREHEAT_2_TEMP_HOTEND, PREHEAT_2_TEMP_BED) rather than those defined  as maximums (HEATER_0_MAXTEMP  and  BED_MAXTEMP respectively).

eg In LCD advance menu, when attempting to adjust max hot end temp, the value initially shows 250 (which is my HEATER_0_MAXTEMP) but when I click on the item it reverts immediately to 235 (which is my PREHEAT_2_TEMP_HOTEND) and I cannot increase past 235.

### Steps to Reproduce

1. Load Marlin and do normal G28, G29 etc ready to print. Start print...
2. Try to adjust hot end or bed temperature above preheat constants (in any of  Temperature, Tuning or Config menus or via USB host)
3. Observe that they can only  be raised to  PREHEAT_2_TEMP_HOTEND and PREHEAT_2_TEMP_BED respectively, not HEATER_0_MAXTEMP or BED_MAXTEMP as expected

**Expected behavior:**
Able to increase hot end or bed temps up to HEATER_0_MAXTEMP  and  BED_MAXTEMP  respectively

**Actual behavior:** 
Can only raise temps to PREHEAT_2_TEMP_HOTEND, PREHEAT_2_TEMP_BED  respectively

#### Additional Information

I made sure to load f/w values into EEPROM to ensure no 'old' values remain.
Using VSC to compile. All libraries up to date (8th Oct 2019)

In the attached ALL edits I have made are denoted with //NDM. Nothing else altered
[Config Files.zip](https://github.com/MarlinFirmware/Marlin/files/3700045/Config.Files.zip)

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

shitcreek, 2019-10-08:
Have you tried changing `PREHEAT_2_TEMP_HOTEND` to a different value to see if the 235 is not just a coincident? Are you certain you're using the right temp sensor?

NickMilton, 2019-10-08:
Good suggestion. Oddly I set the below to be sure (all values are different and on 2 different printers).

I find that BOTH printers refuse to go above 235C on hot end and 70C on bed. Ghost in the machine?

// Preheat Constants
#if ENABLED(Printer1Selected) //NDM code
    #define PREHEAT_1_LABEL       "PLA"  //*NDM for Nicks RepRap#2 printer
    #define PREHEAT_1_TEMP_HOTEND 201
    #define PREHEAT_1_TEMP_BED     51
    #define PREHEAT_1_FAN_SPEED     0 // Value from 0 to 255
    #define PREHEAT_2_LABEL       "PETG"  //*NDM for Nicks RepRap#2 printer
    #define PREHEAT_2_TEMP_HOTEND 231
    #define PREHEAT_2_TEMP_BED     71
    #define PREHEAT_2_FAN_SPEED     0 // Value from 0 to 255
#endif //NDM code

#if ENABLED(Printer2Selected) //NDM code
    #define PREHEAT_1_LABEL       "PLA"  //*NDM for Nicks RepRap#2 printer
    #define PREHEAT_1_TEMP_HOTEND 202
    #define PREHEAT_1_TEMP_BED     52
    #define PREHEAT_1_FAN_SPEED     0 // Value from 0 to 255
    #define PREHEAT_2_LABEL       "PETG"  //*NDM for Nicks RepRap#2 printer
    #define PREHEAT_2_TEMP_HOTEND 232
    #define PREHEAT_2_TEMP_BED     72
    #define PREHEAT_2_FAN_SPEED     0 // Value from 0 to 255
#endif //NDM code

Appreciated

NickMilton, 2019-10-08:
What I expected is for the below to enable me to go to 250C and 80C..

#define HEATER_0_MAXTEMP 250 //NDM hot end Printer 1 and Printer 2 
#define HEATER_1_MAXTEMP 275
#define HEATER_2_MAXTEMP 275
#define HEATER_3_MAXTEMP 275
#define HEATER_4_MAXTEMP 275
#define HEATER_5_MAXTEMP 275
#define BED_MAXTEMP      80 //NDM bed Printer 1 and Printer 2

shitcreek, 2019-10-08:
Is this a recent development? Were you ever able to get to 250c on the hotend before? If not, then it could just be the limit of your machine.

ManuelMcLure, 2019-10-08:
MAXTEMP is defined as “if a heater reaches this temperature there’s something seriously wrong and we need to do an emergency stop”. By default you can only set a heater temperature to MAXTEMP minus 15 degrees - the extra is so the error condition doesn’t trigger when the temperature is converging to the set value. If you need to set the temperature to 250C you need to set MAXTEMP to 265.

NickMilton, 2019-10-08:
ManuelMcLure & shitcreek..

Thank you! I had previously been able to  exceed 235 on the hot end but that was because I had previously left  HEATER_0_MAXTEMP as default 275..

Makes sense now. I did not know about 'Maxtemp-15 degrees' safety margin. So not a bug - just a knowledge limitation. Will test now

shitcreek, 2019-10-08:
excellent. let us know if there is any issue, and please close the thread if it's all good.

NickMilton, 2019-10-08:
All good!

Much thanks.

closed 2019-10-08 by NickMilton