MarlinFirmware/Marlin #15485: [BUG] (short description)

kaijrock23, 2019-10-08:
Hi in the move axis menu I have x.y,z,e, x is x, y is y, but if I want to move z I click z but is y, why?

kaijrock23, 2019-10-08:
[image]
[image]

shitcreek, 2019-10-08:
That's strange. Sometimes, on some machines, when you click down on the knob, it turns it a bit which then causes it to select either the option above or below the one you choose. Maybe this is the case, how many times have you tried it?

We can't help you much without more information. Is this a recent issue? What version of Marlin are you using? Configuration files?

kaijrock23, 2019-10-08:
> That's strange. Sometimes, on some machines, when you click down on the knob, it turns it a bit which then causes it to select either the option above or below the one you chose. Maybe this is the case, how many times have you tried it?
> 
> We can't help you much without more information. Is this a recent issue? What version of Marlin are you using? Configuration files?

I tried many times, it's the latest build, downloaded and compiled today, is there an option to look for to see if the y is instead of the Z in that variable?

shitcreek, 2019-10-08:
Looks like it was a bug afterall and @thinkyhead  has addressed it: https://github.com/MarlinFirmware/Marlin/commit/cb1cb72c76c32e0fb7644874f326d3dd7d726543
Try again with the latest build.

closed 2019-10-08 by shitcreek