MarlinFirmware/Marlin #15487: _Statusscreen.h examples for Anet A6, A8 and A8 Plus

Lord-Quake, 2019-10-08:
I have prepared 3 examples for the above mentioned printers that may be included under printer examples if so desired.

Here an example for the Anet A8
[image]

File for the:
Anet A8: [_Statusscreen.zip](https://github.com/MarlinFirmware/Marlin/files/3703653/_Statusscreen.zip)
Anet A8 Plus: [_Statusscreen.zip](https://github.com/MarlinFirmware/Marlin/files/3703658/_Statusscreen.zip)
Anet A6: [_Statusscreen.zip](https://github.com/MarlinFirmware/Marlin/files/3703663/_Statusscreen.zip)

Patag, 2019-10-08:
My A8 thanks you very much :-)

shitcreek, 2019-10-08:
I've created a PR for this. Thank you for your contribution.

closed 2019-10-08 by shitcreek

Patag, 2019-10-13:
For Anet A8, just for fun :-)
[_Statusscreen.zip](https://github.com/MarlinFirmware/Marlin/files/3721728/_Statusscreen.zip)