MarlinFirmware/Marlin #15492: [BUG] (Error compiling)

407s, 2019-10-09:
Downloaded latest Marlin on 09/10/2019.
Trying compile via Platformio in VisualStudio and always got error:

> > Executing task: C:\.platformio\penv\Scripts\platformio.exe run <
> 
> Processing STM32F103RE (platform: ststm32; framework: arduino; board: genericSTM32F103RE)
> --------------------------------------------------------------------------------------------------------------------------------------------------------------------
> Verbose mode can be enabled via `-v, --verbose` option
> CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RE.html
> PLATFORM: ST STM32 5.6.0 > STM32F103RE (64k RAM. 512k Flash)
> HARDWARE: STM32F103RET6 72MHz, 64KB RAM, 512KB Flash
> DEBUG: Current (blackmagic) External (blackmagic, jlink, stlink)
> PACKAGES: toolchain-gccarmnoneeabi 1.70201.0 (7.2.1), framework-arduinoststm32-maple 1.10000.190819 (1.0.0)
> Converting Marlin.ino
> Error: **stm32f1 family is not supported by maple core**

I just need compile firmware for SKE E3 DIP with STM32F103RE.

default_envs =  STM32F103RC_bigtree
No different bettwen Configuration.h/Configuration_adv.h my or default.
#define MOTHERBOARD BOARD_BIGTREE_SKR_E3_DIP

VStudio stable or insider - no difference

RudyBenner, 2019-10-09:
https://github.com/bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0

reloxx13, 2019-10-09:
try to delete anything in 

`C:\Users\YOURNAME\.platformio\platforms`
and
`C:\Users\YOURNAME\.platformio\packages`

and try again.

407s, 2019-10-09:
https://github.com/MarlinFirmware/Marlin/issues/15492#issuecomment-539845894

Hello. In my path \.platformio i have only folder **.cache** and **penv**

407s, 2019-10-09:
> https://github.com/bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0

Same problem.

407s, 2019-10-09:
Oh... It was my mistate.
1. Full deleting VStudio
2. Manual deleting .platfromio
3. Clean install Vstudio (system)

closed 2019-10-09 by 407s