MarlinFirmware/Marlin #15494: When using UBL G29 J the new G29 T shows all points at the same level

hidara2000, 2019-10-09:
#Description
When running 3-point probe after UBL mesh levelling the mesh new results are all in the same horizontal plane

### Steps to Reproduce

Run the following commands
G28           ; Home XYZ.
G29 P1        ; Do automated probing of the bed.
G29 T         ; View the Z compensation values.
G29 S1        ; Save UBL mesh points to EEPROM.
G29 F 10.0    ; Set Fade Height for correction at 10.0 mm.
G29 A         ; Activate the UBL System.
M500          ; Save current setup. WARNING - UBL will be active at power up, before any G28.

G29 L1        ; Load the mesh stored in slot 1 (from G29 S1)
G29 J 

G29 T

**Expected behavior:** [What you expect to happen]
G29T to output original values with slight adjustments to compensate for plane

**Actual behavior:** [What actually happens]
New points are all in the horizontal plane (ie bed is perfectly flat??)

terminal output 
[Marlin-1.zip](https://github.com/MarlinFirmware/Marlin/files/3705872/Marlin-1.zip)

Recv: Probing mesh point 4/16.
Recv: Probing mesh point 5/16.
Recv: Probing mesh point 6/16.
Recv: Probing mesh point 7/16.
Recv: Probing mesh point 8/16.
Recv: Probing mesh point 9/16.
Recv: Probing mesh point 10/16.
Recv: Probing mesh point 11/16.
Recv: Probing mesh point 12/16.
Recv: Probing mesh point 13/16.
Recv: Probing mesh point 14/16.
Recv: Probing mesh point 15/16.
Recv: Probing mesh point 16/16.
Send: G29 T
Recv: X:200.00 Y:150.00 Z:16.35 E:0.00 Count X:16146 Y:12028 Z:6574
Recv: ok
Recv: Bed Topography Report:
Recv:     ( 50,250)              (250,250)
Recv:         0       1       2       3
Recv:  3 | +0.341  +0.200  +0.141  +0.239
Recv:    |
Recv:  2 | +0.193  +0.073 [+0.038] +0.120
Recv:    |
Recv:  1 | +0.150  -0.001  -0.044  +0.045
Recv:    |
Recv:  0 | +0.101  -0.050  -0.064  +0.023
Recv:         0       1       2       3
Recv:     ( 50, 50)              (250, 50)
Recv: ok
Send: G29 S1
Recv: Done.
Recv: ok
Send: G29 A
Recv: Unified Bed Leveling System v1.01 active
Recv: ok
Send: M500
Recv: ok
Send: G29 L1
Recv: Done.
Recv: ok
Send: G29 J
Recv: Tilting mesh (1/3)
Recv: Tilting mesh (2/3)
Recv: Tilting mesh (3/3)
Recv: X:100.00 Y:100.00 Z:16.35 E:0.00 Count X:8073 Y:8019 Z:6574
Recv: ok
Send: G29 T
Recv: Bed Topography Report:
Recv:     ( 50,250)              (250,250)
Recv:         0       1       2       3
Recv:  3 | +1.157  +1.157  +1.157  +1.157
Recv:    |
Recv:  2 | +1.157  +1.157  +1.157  +1.157
Recv:    |
Recv:  1 |[+1.157] +1.157  +1.157  +1.157
Recv:    |
Recv:  0 | +1.157  +1.157  +1.157  +1.157
Recv:         0       1       2       3
Recv:     ( 50, 50)              (250, 50)
Recv: ok

#### Additional Information

tvixen, 2019-10-09:
Will try to do a G29 T after G29 J, tonight. As I too have issues with UBL. 
But haven't thought about making this sequence.

Arti4ever, 2019-10-09:
I have exactly the same behavior. 
Since commit df1e512 ABL and UBL seems completely broken.
Also, custom 3 points probing with G29 J does not work either.

sdp8483, 2019-10-09:
I have been having the same problem and just replicated the issue.
Performed G29 P1, G29 T topography report shows bed variability.
Then performed G29 J, G29 T shows all points are +1.086

tvixen, 2019-10-09:
I can NOT confirm the behavior of the G29 J you see. But then again, I'm running version 2.1.x-dev.

This is my test data:
Unified Bed Leveling System v1.01 active
Active Mesh Slot: 1
EEPROM can hold 22 meshes.
echo:Material heatup parameters:
echo:  M145 S0 H205 B50 F0
echo:  M145 S1 H240 B60 F0
echo:PID settings:
echo:  M301 P21.73 I1.54 D76.55
echo:Z-Probe Offset (mm):
echo:  M851 Z-1.70
echo:Stepper driver current:
echo:  M906 X760 Y760 Z760
echo:  M906 T0 E900
echo:Driver stepping mode:
echo:  M569 S1 X Y Z
  M569 S1 T0 E
>>> G29 L1 T
SENDING:G29 L1 T
Done.
Bed Topography Report:
    (  5,220)                              (220,220)
        0       1       2       3       4       5
 5 |   .     +0.093  -0.141  -0.286  -0.405  -0.565
   |
 4 |   .     +0.196  -0.041  -0.178  -0.290  -0.426
   |
 3 |   .     +0.264  +0.028  -0.099  -0.214  -0.325
   |
 2 |   .     +0.351  +0.114 [-0.022] -0.123  -0.222
   |
 1 |   .     +0.349  +0.100  -0.028  -0.120  -0.223
   |
 0 |   .     +0.304  +0.069  -0.054  -0.156  -0.389
        0       1       2       3       4       5
    (  5,  5)                              (220,  5)
>>> G29 L1
SENDING:G29 L1
Done.
>>> G29 J
SENDING:G29 J
Tilting mesh (1/3)
echo:busy: processing
echo:busy: processing
echo:busy: processing
Tilting mesh (2/3)
echo:busy: processing
echo:busy: processing
echo:busy: processing
Tilting mesh (3/3)
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
>>> G29 T
SENDING:G29 T
Bed Topography Report:
    (  5,220)                              (220,220)
        0       1       2       3       4       5
 5 |   .     +0.292  -0.027  -0.256  -0.460  -0.704
   |
 4 |   .     +0.374  +0.052  -0.169  -0.366  -0.587
   |
 3 |   .     +0.420  +0.099  -0.112  -0.312  -0.507
   |
 2 |   .     +0.485  +0.164 [-0.056] -0.242  -0.425
   |
 1 |   .     +0.462  +0.129  -0.084  -0.260  -0.448
   |
 0 |   .     +0.395  +0.076  -0.132  -0.318  -0.636
        0       1       2       3       4       5
    (  5,  5)                              (220,  5)

SENDING:G29 L1 T
Done.
Bed Topography Report:
    (  5,220)                              (220,220)
        0       1       2       3       4       5
 5 |   .     +0.093  -0.141  -0.286  -0.405  -0.565
   |
 4 |   .     +0.196  -0.041  -0.178  -0.290  -0.426
   |
 3 |   .     +0.264  +0.028  -0.099  -0.214  -0.325
   |
 2 |   .     +0.351  +0.114 [-0.022] -0.123  -0.222
   |
 1 |   .     +0.349  +0.100  -0.028  -0.120  -0.223
   |
 0 |   .     +0.304  +0.069  -0.054  -0.156  -0.389
        0       1       2       3       4       5
    (  5,  5)                              (220,  5)

tvixen, 2019-10-09:
I can confirm that version 2.0.x is having this problem.
Here is my test data:

Unified Bed Leveling System v1.01 inactive
Active Mesh Slot: -1
EEPROM can hold 89 meshes.
echo:Material heatup parameters:
echo:  M145 S0 H205 B50 F0
echo:  M145 S1 H240 B60 F0
echo:PID settings:
echo:  M301 P21.73 I1.54 D76.55
echo:Z-Probe Offset (mm):
echo:  M851 X48.00 Y-2.00 Z0.00
echo:Stepper driver current:
echo:  M906 X760 Y760 Z760
echo:  M906 T0 E900
echo:Driver stepping mode:
echo:  M569 S1 X Y Z
echo:  M569 S1 T0 E
>>> G28
SENDING:G28
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
>>> G29 P1
SENDING:G29 P1
Default storage slot 0 selected.
Mesh invalidated. Probing mesh.
Probing mesh point 1/9.
echo:busy: processing
echo:busy: processing
Probing mesh point 2/9.
echo:busy: processing
echo:busy: processing
echo:busy: processing
Probing mesh point 3/9.
echo:busy: processing
echo:busy: processing
echo:busy: processing
Probing mesh point 4/9.
echo:busy: processing
echo:busy: processing
echo:busy: processing
Probing mesh point 5/9.
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
Probing mesh point 6/9.
echo:busy: processing
echo:busy: processing
echo:busy: processing
Probing mesh point 7/9.
echo:busy: processing
>>> G29 T
SENDING:G29 T
Bed Topography Report:
    (  5,220)      (220,220)
        0       1       2
 2 |   .     -0.154  -0.474
   |
 1 |   .    [-0.003] -0.220
   |
 0 |   .     +0.060  -0.294
        0       1       2
    (  5,  5)      (220,  5)
>>> G29 p3
SENDING:G29 P3
>>> G29 T
SENDING:G29 T
Bed Topography Report:
    (  5,220)      (220,220)
        0       1       2
 2 | +0.167  -0.154  -0.474
   |
 1 | +0.214 [-0.003] -0.220
   |
 0 | +0.414  +0.060  -0.294
        0       1       2
    (  5,  5)      (220,  5)
>>> G29 s1
SENDING:G29 S1
Done.
>>> G29 G29 L1 T
SENDING:G29 G29 L1 T
Done.
Bed Topography Report:
    (  5,220)      (220,220)
        0       1       2
 2 | +0.167  -0.154  -0.474
   |
 1 | +0.214 [-0.003] -0.220
   |
 0 | +0.414  +0.060  -0.294
        0       1       2
    (  5,  5)      (220,  5)
>>> G29 L1 T
SENDING:G29 L1 T
Done.
Bed Topography Report:
    (  5,220)      (220,220)
        0       1       2
 2 | +0.167  -0.154  -0.474
   |
 1 | +0.214 [-0.003] -0.220
   |
 0 | +0.414  +0.060  -0.294
        0       1       2
    (  5,  5)      (220,  5)
>>> G29 J
SENDING:G29 J
Tilting mesh (1/3)
echo:busy: processing
echo:busy: processing
echo:busy: processing
Tilting mesh (2/3)
echo:busy: processing
echo:busy: processing
echo:busy: processing
Tilting mesh (3/3)
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
>>> G29 T
SENDING:G29 T
Bed Topography Report:
    (  5,220)      (220,220)
        0       1       2
 2 | +1.366  +1.366  +1.366
   |
 1 | +1.366 [+1.366] +1.366
   |
 0 | +1.366  +1.366  +1.366
        0       1       2
    (  5,  5)      (220,  5)
>>> G29 L1 T
SENDING:G29 L1 T
Done.
Bed Topography Report:
    (  5,220)      (220,220)
        0       1       2
 2 | +0.167  -0.154  -0.474
   |
 1 | +0.214 [-0.003] -0.220
   |
 0 | +0.414  +0.060  -0.294
        0       1       2
    (  5,  5)      (220,  5)

tvixen, 2019-10-09:
And just to show it's not because of the inactive UBL. Here is the output with active UBL.

Unified Bed Leveling System v1.01 active
>>> G29 J
SENDING:G29 J
Tilting mesh (1/3)
echo:busy: processing
echo:busy: processing
echo:busy: processing
Tilting mesh (2/3)
echo:busy: processing
echo:busy: processing
echo:busy: processing
Tilting mesh (3/3)
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
>>> G29 T
SENDING:G29 T
Bed Topography Report:
    (  5,220)      (220,220)
        0       1       2
 2 | +1.351  +1.351  +1.351
   |
 1 | +1.351 [+1.351] +1.351
   |
 0 | +1.351  +1.351  +1.351
        0       1       2
    (  5,  5)      (220,  5)
>>> G29 l1 T
SENDING:G29 L1 T
Done.
Bed Topography Report:
    (  5,220)      (220,220)
        0       1       2
 2 | +0.167  -0.154  -0.474
   |
 1 | +0.214 [-0.003] -0.220
   |
 0 | +0.414  +0.060  -0.294
        0       1       2
    (  5,  5)      (220,  5)

dermigoe, 2019-10-10:
I can confirm this problem.
Here is my output of the terminal:

**G29 T**

Bed Topography Report:

    ( 35,185)                                                      (185,185)
        0       1       2       3       4       5       6       7       8
 8 | +0.074  +0.204  +0.164  -0.016  +0.062  +0.180  +0.038  -0.034  +0.137
   |
 7 | +0.069  +0.219  +0.174  +0.014  +0.111  +0.230  +0.100  +0.043  +0.242
   |
 6 | +0.116  +0.252  +0.184  -0.009  +0.105  +0.216  +0.022  +0.031  +0.187
   |
 5 | +0.032  +0.128  +0.122  -0.067  +0.056  +0.203  +0.008  +0.021  +0.184
   |
 4 | +0.083  +0.174  +0.128  -0.056 [+0.013] +0.154  -0.019  -0.060  +0.157
   |
 3 | +0.076  +0.178  +0.106  -0.054  +0.041  +0.131  -0.002  -0.072  +0.130
   |
 2 | -0.004  +0.134  +0.028  -0.132  -0.030  +0.080  -0.078  -0.103  +0.086
   |
 1 | -0.004  +0.198  +0.101  -0.070  +0.052  +0.178  -0.009  -0.075  +0.140
   |
 0 | +0.106  +0.240  +0.189  +0.004  +0.079  +0.212  +0.053  +0.008  +0.196
        0       1       2       3       4       5       6       7       8
    ( 35, 35)                                                      (185, 35)

**G29 J**
Tilting mesh (1/3)
Tilting mesh (2/3)
Tilting mesh (3/3)

X:75.00 Y:75.00 Z:3.43 E:0.00 Count X:6000 Y:6000 Z:1372

**G29 T**

Bed Topography Report:

    ( 35,185)                                                      (185,185)
        0       1       2       3       4       5       6       7       8
 8 | +0.952  +0.952  +0.952  +0.952  +0.952  +0.952  +0.952  +0.952  +0.952
   |
 7 | +0.952  +0.952  +0.952  +0.952  +0.952  +0.952  +0.952  +0.952  +0.952
   |
 6 | +0.952  +0.952  +0.952  +0.952  +0.952  +0.952  +0.952  +0.952  +0.952
   |
 5 | +0.952  +0.952  +0.952  +0.952  +0.952  +0.952  +0.952  +0.952  +0.952
   |
 4 | +0.952  +0.952  +0.952  +0.952  +0.952  +0.952  +0.952  +0.952  +0.952
   |
 3 | +0.952  +0.952  +0.952  +0.952  +0.952  +0.952  +0.952  +0.952  +0.952
   |
 2 | +0.952  +0.952  +0.952  +0.952  +0.952  +0.952  +0.952  +0.952  +0.952
   |
 1 |[+0.952] +0.952  +0.952  +0.952  +0.952  +0.952  +0.952  +0.952  +0.952
   |
 0 | +0.952  +0.952  +0.952  +0.952  +0.952  +0.952  +0.952  +0.952  +0.952
        0       1       2       3       4       5       6       7       8
    ( 35, 35)                                                      (185, 35)

stefi01, 2019-10-14:
when i use G29 J it will probe the left corner then fail without moving to the next point, tried with 1.19 and all works ok

tvixen, 2019-10-16:
> when i use G29 J it will probe the left corner then fail without moving to the next point, tried with 1.19 and all works ok

Yeps... also one of the included bugs.
@shitcreek is there a "fixed bugs" .log somewhere ?

thinkyhead, 2019-10-26:
> @shitcreek is there a "fixed bugs" .log somewhere ?

Closed issues. Merged commits.

closed 2019-10-26 by thinkyhead

Roxy-3D, 2019-10-30:
> any update about when this bug will be fixed ?

Yes!   Just as soon as somebody....  Anybody!!! has time to figure out the 'root cause' and tweak a couple lines of code to resolve the issue.

If you understand C....   Dig in!