MarlinFirmware/Marlin #15495: [BUG] Z = 1 after G28 on delta

DkOuvE, 2019-10-09:
I git pulled the latest 2.0 and after G28 on my delta, the Z height is set to 1mm instead of the delta height.

### Steps to Reproduce

1. Pull latest marlin
2. Send G28

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3706821/Configuration.zip)

tincatinca, 2019-10-09:
Strange because I came on here because I have the same problem except my Z = 5 been using ver 2 successfully for 2 weeks but now has developed this problem

DkOuvE, 2019-10-09:
Ok I've determined that the Z is correct when bed levelling is turned "Off". When it is turned back on again and auto homed, Z = 1 again.

tvixen, 2019-10-10:
Seems to be the same issue as #15494

DkOuvE, 2019-10-10:
Updated to dev 2.1 with the same issues.

Studna3D, 2019-10-10:
I have the same problem after bed leveling. When I start printing, runs G28 and G29, but after Z axis not moving, it's only Z=5 over the bed and prints only at this height.

407s, 2019-10-12:
If we use #define AUTO_BED_LEVELING_**LINEAR**
G28 give Z 10 and then G29 give Z 10->1mm

If we use #define AUTO_BED_LEVELING_**BILINEAR**
G28 give Z 10 and then G29 give Z 10

DkOuvE, 2019-10-17:
Plugged into MatterControl, I get X: -0.34 Y: -0.22 Z:1.00 with bed leveling on. With it off, I get X:0, Y:0, Z:230.94

DkOuvE, 2019-10-17:
Looks like there's a sync_plan_position that causes the values to change.

[LP_print_log.txt](https://github.com/MarlinFirmware/Marlin/files/3738819/LP_print_log.txt)

Evgeny-SPB, 2019-10-20:
I have the same problem on prusa like printer. 
The problem is in "void Planner::apply_leveling(xyz_pos_t &raw)" in planner.cpp and concerns only ABL_PLANAR levelling (in my case 3 point levelling).
Function "apply_rotation_xyz(bed_level_matrix, d.x, d.y, raw.z);" always returns Z = 1.0

DkOuvE, 2019-10-21:
#15610 seems to work. Now I need to get my TMC5160 to extrude so that I can see if bed levelling is applied.

boelle, 2019-10-26:
will close this one as it prob fixed by https://github.com/MarlinFirmware/Marlin/pull/15636

we can always reopen

closed 2019-10-26 by boelle