MarlinFirmware/Marlin #15496: Sensorless Homing/Probing not working on SKR Pro v1.1 + TMC2209 v1.1

hackebike, 2019-10-09:
I am setting up a custom printer using the **SKR Pro v1.1** with **TMC 2209 v1.1** from BIGTREETECH.

I complied Marlin from yesterday October 8th.
I have setup the configuration files but when testing the following happen: 

- X moving, suppose arrives at the start point (x=0) of X axis but does not stop
- same for Y and Z
- The endstops are not triggered but are open

I played with sensitivity but is not working. 
I uploaded the 3 config files, or can be viewed here https://github.com/hackebike/3dprinter

[Marlin-config.zip](https://github.com/MarlinFirmware/Marlin/files/3707131/Marlin-config.zip)

tvixen, 2019-10-09:
Try disabling #define SQUARE_WAVE_STEPPING

hackebike, 2019-10-09:
I tested without #define SQUARE_WAVE_STEPPING but not working.
I tested many different sensitivity from 0 to 200 increment of 10, no luck.
I also tested without #define HYBRID_THRESHOLD but nothing.

any other suggestions?

psim, 2019-10-09:
Almost same issues here:
endstops are not working even if I trigger them manually. 

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3707319/Marlin.zip)

tvixen, 2019-10-09:
Did you do a M502 M500 M501, every time for every step in the software ? (just to be sure)

hackebike, 2019-10-09:
did not tried those commands yet, but I think SKR Pro has some issues with EEPROM.
I used M914 to set sensitivity, 2209 goes from 0 (less) to 255 (more) sensitive.

Not sure what other tests to do.

tvixen, 2019-10-09:
Oki M914 should work too.

hackebike, 2019-10-09:
will give it a try and report, thx

tvixen, 2019-10-09:
Well if you got the skills, you could hook it up on a veroboard and try a single driver + a stepper, and a oscilloscope to trigger the stallguard acc. to https://www.trinamic.com/fileadmin/assets/Products/ICs_Documents/TMC2209_Datasheet_V103.pdf

But I think it will only confirm your suspicion, that the SKR Pro has issues.

hackebike, 2019-10-09:
Cannot use the EEprom as the SKR pro has issues, does this affect my issue?

tvixen, 2019-10-09:
No not really ....directly. But the whole 32bit project has some thing who require EEprom settings, so you will (inevitably) run into problems sooner or later.

hackebike, 2019-10-09:
will do more testing tomorrow, also trying on the skr v1.3 to see if the tmc2209 are the issue or something else.

randellhodges, 2019-10-10:
> Well if you got the skills, you could hook it up on a veroboard and try a single driver + a stepper, and a oscilloscope to trigger the stallguard acc. to https://www.trinamic.com/fileadmin/assets/Products/ICs_Documents/TMC2209_Datasheet_V103.pdf
> 
> But I think it will only confirm your suspicion, that the SKR Pro has issues.
> Actually I see no STM32 chip fully complying with Marlin.

Why do you think no STM32 chip will fully comply with Marlin? Isn't Marlin and the concept of the HAL to abstract that away and "make it work?"

Disclaimer: I have SKR 1.1 PRO sitting in the box, waiting.

tvixen, 2019-10-10:
You can see the progress of the chipset here:
https://github.com/MarlinFirmware/Marlin/issues/14345

hackebike, 2019-10-10:
@ randellhodges did you had a chance to test sensorless homing ?

randellhodges, 2019-10-10:
@hackebike No, I'm still using my SKR 1.3 along with the TMC2208 drivers. From what I've read, I've decided that when I do switch over to the TMC2209 drivers, I'm going to stick with endstop switches.

I'm going to try the TMC2209 in the SKR 1.3 tonight probably, then work on moving them into the SKR PRO 1.1 and try that out. I've done a lot of work on the printer and I'm trying to introduce these changes slowly.

hackebike, 2019-10-10:
ok great make sense.
I will try the 2209 on skr 1.3 too tomorrow so will report my findings. Will use the same config as I am using now but compiling for skr1.3.

hackebike, 2019-10-10:
I will continue with sensorless homing since my custom printer is huge and do not want to deal with mechanical switches and wiring all over the places!

It's just annoying that they release hardware but they do not release configurations for the supported drivers. They must have tested these scenario otherwise how they tested?

Yes I know about the clipping thing and that's one thing I want to avoid, messing with hardware that is suppose to work. I did for the 2130 with soldering but is just a mess!

My print will last couple of days so less (?) is better for me.

I will push BTT to provide a config files that works with laster Marlin and TMC 2209 !

hackebike, 2019-10-10:
btw do you know what is the difference between TMC2209, TMC2209_STANDALONE in the configuration.h when you select which driver to use?

randellhodges, 2019-10-10:
> also did you see this? [https://www.youtube.com/watch?v=6oKtRML0kUU](url)

Yup, but I also failed to mention, while I ordered the v1.1, they actually sent me v1.2, so I don't think mine will have that issue.

hackebike, 2019-10-10:
I have both 1.1 and 1.2, tested both for sensorless, no luck !
I am starting to think that they did not test it.

teemuatlut, 2019-10-10:
https://github.com/teemuatlut/Marlin/tree/tmc2209_stallguard

Let me know if anything changes.

hackebike, 2019-10-10:
ok, great, will test it tomorrow and report results. thx!

hackebike, 2019-10-10:
getting an error while compiling, check image and config files

[Marlin-config.zip](https://github.com/MarlinFirmware/Marlin/files/3714023/Marlin-config.zip)

[image]

LongLiveCHIEF, 2019-10-10:
the settings I use in my branch of Marlin are designed for the SKR boards, and I try to keep them updated. I know they instructions from Bigtreetech are incorrect (and I've got fixes and open issues for all this stuff if you check out their repo), but mine _should_ work for you. Check 'em out at [github.com/HackerHappyHour/Marlin skr config branch](https://github.com/HackerHappyHour/Marlin/tree/ender3-skr-v1.3-config)

I'll be doing another rebase today or tomorrow to update for firmware changes of 2.x that have occurred between July and now.

hackebike, 2019-10-10:
sounds good!
So at the moment I have couple of tests to do, one with some hybrid threshold set to 0 (some people said to try!), than will use @teemuatlut release and than will do your for sure!

Will report my findings and thx a lot for the support!

hackebike, 2019-10-12:
If stallguard is not sure the SKR Pro manual mention to remove the PIn on the TMC 2209 !!
What a great idea ..... !!!  https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/blob/master/manual/SKR%20PRO%20V1.1%20user%20manual%20v1.1.pdf

hackebike, 2019-10-14:
did more tests and nothing is working!
If anyone has some suggestions please let me know, for now will use the 2208 without sensorless homing!

TB1405, 2019-10-14:
There is a big in the actual marlin Version i have a skr pro with btt2209 and Version from a month ago and with this it works. With the actual it didnt work

hackebike, 2019-10-14:
really?
could you upload the Marlin code with configuration so that we can test it ?
it would be great! thx

hackebike, 2019-10-15:
I got a Marlin code with configs from BTT, but I had issues compiling it. So I just changed the config files to what I needed and the standard platform.ini file, it compiles but no luck on sensorless homing. Anyone want to give it a try?
This is the [BTT Marlin that suppose to work with TMC 2209](https://drive.google.com/file/d/1wfGp8NIYPT5HBQP5XU6p1m2uLF7oRz45/view?usp=drive_web) sensorless homing

hackebike, 2019-10-15:
@ phellwing so it does compile for you without modifying anything? are you using VS Code?

phellwing, 2019-10-15:
Download, import to visual studio, compile. Nothing else.
(Have to add a driver to E1, I have only one extruder)

hackebike, 2019-10-15:
ok great, which means it must be something with VS Code extension/configuration.
Did you test if the sensorless homing works with TMC 2209?

hackebike, 2019-10-15:
Re-installed the VS Code and it did compile, which is a good step, but not the final one! :)

[Here](https://drive.google.com/file/d/0B68JrbJTtiADNllOOVhIdmMtdDRQX2daVS1qa1dIemxlZjZv/view?usp=sharing) is the original BTT Marlin code that they sent me and sensorless homing is suppose to work. If you want to take a look. Not sure when this code was taken from Marlin github repo so have no idea what works and what does not!

hackebike, 2019-10-16:
I tested the BTT marlin code that was suppose to work, but nothing!

hackebike, 2019-10-16:
I tested BTT marlin on 2 different board, SKR Pro v1.1 and SKR v1.3 with 2 sets of TMC 2209 v1.1/v1.2 and sensorless homing did not work either of them.

hackebike, 2019-10-22:
I got it working both sensorless homing and probing with SKR pro v1.1 and TMC 2209 v1.2.
Not sure how, since I downloaded Marlin from few days ag, compiled with new settings in the configuration files and it worked with sensitivity X60 Y60 Z76. Here are the [config files.](https://github.com/hackebike/3dprinter/commits/master)

closed 2019-10-22 by hackebike

fiveofds, 2020-02-13:
I have an SKR PRO V1.1 with TMC2209s using Configurations from Bigtreetec I can get sensorless homing working IF it is configured as a cartesian printer but then configuring for my delta printer:

// Direction of endstops when homing; 1=MAX, -1=MIN
// :[-1,1]
#define X_HOME_DIR 1 //MPD
#define Y_HOME_DIR 1 //MPD
#define Z_HOME_DIR 1 //MPD

When this is switched from -1 to 1 sensorless homing stops working
Any advice?