MarlinFirmware/Marlin #15500: [BUG] Not properly passing through SD card over USB to raspberry pi (2.0.x)

Kokorone, 2019-10-09:
### Description

SD card will not mount and instead have Disk I/O errors with recent marlin 2.0.x firmware (10/5/2019 and newer tested, 9/19/2019 version confirmed working properly). Tested with manual mounting attempts and with USBMount.

### Steps to Reproduce

1. Set up a new marlin build and flash to board
2. Connect board over usb to raspberry pi, (shouldnt matter if running octo or not)
3. Mount the sd card to the pi, I used USBMount configured as outlined with Octoprint-FirmwareUpdater to auto mount.

**Expected behavior:** [What you expect to happen]
SD card properly mounts to the raspberry pi and is RW accessible

**Actual behavior:** [What actually happens]
SD card fails to properly pass though and never mounts, with an I/O error

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
[ConfigFiles.zip](https://github.com/MarlinFirmware/Marlin/files/3709778/ConfigFiles.zip)

* Provide pictures or links to videos that clearly demonstrate the issue.
[image]

* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

tvixen, 2019-10-10:
Which version of Octoprint is running on your Pi? (Stable or Release Candidate)
Did you try with a older version of Octoprint?

Kokorone, 2019-10-10:
Currently using stable octopi release with updated raspbian

tvixen, 2019-10-10:
Hmm, I still think the problem is on your Pi side. Does it auto update (look in manager)?
Can Windows still see your SD card as a drive, when switching USB to windows ?

Kokorone, 2019-10-11:
Updated to the dev version of usbmount as recommended here and it appears to be fixed

[https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/issues/109](url)

closed 2019-10-11 by Kokorone