MarlinFirmware/Marlin #15502: [BUG] Planner_bezier.cpp LONG_FILENAME_LENGTH was not declared in this scope

thisiskeithb, 2019-10-10:
### Description

Using the latest commit (b119c1401749037bc23352633f629268217ee5bf), I am getting the following error messages when trying to compile for an SKR 1.3:

```log
Compiling .pio/build/LPC1768/src/src/module/planner_bezier.cpp.o
Marlin/src/lcd/ultralcd.cpp:47:48: error: 'LONG_FILENAME_LENGTH' was not declared in this scope
     constexpr uint8_t MAX_MESSAGE_LENGTH = max(LONG_FILENAME_LENGTH, MAX_LANG_CHARSIZE * 2 * (LCD_WIDTH));
                                                ^~~~~~~~~~~~~~~~~~~~
Marlin/src/lcd/ultralcd.cpp:47:48: note: suggested alternative: 'TONE_QUEUE_LENGTH'
     constexpr uint8_t MAX_MESSAGE_LENGTH = max(LONG_FILENAME_LENGTH, MAX_LANG_CHARSIZE * 2 * (LCD_WIDTH));
                                                ^~~~~~~~~~~~~~~~~~~~
                                                TONE_QUEUE_LENGTH
Compiling .pio/build/LPC1768/src/src/module/printcounter.cpp.o
```

### Steps to Reproduce

1. Use [my config / this build](https://github.com/thisiskeithb/Marlin/tree/7026218defd9383016e70d2ba1c341a975e30dc1) and try to compile
2. Observe error message

**Expected behavior:** Successful compile.

**Actual behavior:** Failed compile.

tvixen, 2019-10-10:
Thiese two should fix the long filename support
#define LONG_FILENAME_HOST_SUPPORT
#define SCROLL_LONG_FILENAMES

But the question is why you have different display's setup on your ender 3? did you replace your CR10 origin display ?

thisiskeithb, 2019-10-10:
I have `CR10_STOCKDISPLAY` in my config, which is the stock Ender-3 LCD.

tvixen, 2019-10-10:
I see Scott is working on this one. So I'm sure the Issue will be solved shortly :+1:

thisiskeithb, 2019-10-11:
> Oki...sorry. I just saw severals LCD displays in your config file. But now when I check again it's normal CR10 stock display :) ??

I’m not sure where you’re looking other than what I’ve linked, but I only have one LCD active in my config.

> So the question is: You just downloaded the new version of 2.0x and compiled it (without changing anything) and it gives you these errors....right? Did you previously compile ok, for this board ? or is it a new setup to test ?

This was a rebased config that I’ve been using since the beginning of the year.

thisiskeithb, 2019-10-11:
I'm guessing this bug has something to do with changes made in https://github.com/MarlinFirmware/Marlin/commit/6a865a614620d71521bd3dc032adffe0bf82378c.

CRCinAU, 2019-10-15:
Just wondering if there is any progress on this? Seems to still be broken.

thisiskeithb, 2019-10-15:
> Just wondering if there is any progress on this? Seems to still be broken.

Nope. Still broken in my config after rebasing to the latest commit.

Morallying, 2019-10-15:
Hrm, for me it only occurred when I disabled SDSUPPORT. Since I am running on an Ender 3, I just added  `#define LONG_FILENAME_LENGTH 0` to Conditionals_LCD.h under  `#elif ENABLED(CR10_STOCKDISPLAY)` First checking to see if SDSUPPORT was disabled before doing so.
Compiles fine now, though I have yet to flash it....

CRCinAU, 2019-10-16:
Yeah, I have SDSUPPORT disabled too as the Ender 3 doesn't really have much space for better features!

thisiskeithb, 2019-10-16:
> Hrm, for me it only occurred when I disabled SDSUPPORT. 

Enabling `SDSUPPORT` allowed me to compile! I have a print going right now, but I'll test it out later tonight.

`SDSUPPORT` is disabled in the default config, so I wonder why the test builds aren't catching this bug.

chrisqwertz, 2019-10-16:
Just had the same problem with the latest bugfix2.0.

Conf & Conf_adv
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3733771/Marlin.zip)

thisiskeithb, 2019-10-16:
Fixed with https://github.com/MarlinFirmware/Marlin/pull/15582. Thanks @Rotule666!

closed 2019-10-16 by thisiskeithb

Rotule666, 2019-10-16:
Thanks for the props, I didn't do much but I will try to contribute more ;)