MarlinFirmware/Marlin #15512: [BUG] Compile error PID_LABEL and array bounds

viper93458, 2019-10-11:
### Description

Compile errors with current download for SKR Pro 1.1 and TMC2208
[image]

### Steps to Reproduce

1. Download current version from 8:30PM PST on 10/10/19
2. Configure platformio.ini for SKR Pro 1.1
3. Compile using my configs

**Expected behavior:** [What you expect to happen]
Successful compile

**Actual behavior:** [What actually happens]
Compile errors
[image]

#### Additional Information
[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/3715527/Configs.zip)

boelle, 2019-10-12:
any clue what is wrong? maybe we can cook up a PR to fix it

viper93458, 2019-10-12:
Sadly no. Not being a developer, I could only report what's happening. :(

Cova, 2019-10-13:
I'm not sure what is wrong either - but having updated to the current code from a few months ago, I'm now getting this error as well.  In my case on a SKR V1.3 with TMC2208 drivers.

Easy workaround for the time being is to disable PID_AUTOTUNE_MENU - that gets everything compiling and working properly, just have to use a serial terminal to run a PID autotune now instead of being able to do it from the LCD.

tubl07, 2019-10-13:
Also encountered this error using arduino ide 1.8.10 for mega2560 (RUMBA).  Disabled menu PID_AUTOTUNE_MENU to compile successfully.

viper93458, 2019-10-13:
I tried and still got the array bounds error although the PID error went away.

-William

viper93458, 2019-10-13:
Disabling BABYSTEP_DISPLAY_TOTAL  in Configuration_adv.h eliminates the array bounds error:

Previous:
#define BABYSTEP_DISPLAY_TOTAL

New:
//#define BABYSTEP_DISPLAY_TOTAL

-William

boelle, 2019-10-13:
> Also encountered this error using arduino ide 1.8.10 for mega2560 (RUMBA). Disabled menu PID_AUTOTUNE_MENU to compile successfully.

tried ide 1.9.0 beta?

viper93458, 2019-10-14:
Looks like the only issue left here is the babystep_display_total issue. Any thoughts on that piece?

-William

boelle, 2019-10-22:
i think the PR merged 9 days ago has fixed the problem

please test again

Cova, 2019-10-22:
I'm running with PID_AUTOTUNE_MENU enabled again so it seems fixed to me.

boelle, 2019-10-22:
oki will close this one then

we can always reopen

closed 2019-10-22 by boelle

viper93458, 2019-10-22:
Yes, I was able to compile with PID_AUTOTUNE_MENU enabled and babystep_display_total enabled. Thanks!

-William