MarlinFirmware/Marlin #15513: [BUG] Stutter after G29 (bugfix 2.0.x)

Kokorone, 2019-10-11:
### Description

Printer will have consistent stutter on all steppers after a G29 is done prior to a print. 

### Steps to Reproduce

1. Compile Marlin (I tested today's build 10/10/2019)
2. Flash to printer
3. Prepare gcode with a G28 and G29 in the start code
4. Attempt to print from either Octoprint or from SD card

**Expected behavior:** [What you expect to happen]
Printer will print as normal

**Actual behavior:** [What actually happens]
Printer will stutter constantly, causing blobs of extrusion and in general a horrible print in a slow choppy manner

#### Additional Information

Printer is a Creality CR-10 using an SKR 1.3 with bilinear leveling enabled. Flashing firmware pulled from 10/5/2019 compiled with identical flags prints flawlessly so it appears to be a very recent commit that started causing this problem. Print does not appear to stutter if abl is not used (removing G29). 

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3715923/Marlin.zip)

* Provide pictures or links to videos that clearly demonstrate the issue.
[image]

* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

GhostlyCrowd, 2019-10-11:
I'm palso in this same boat. Same print defects and the same pulsating stepper issue.

GhostlyCrowd, 2019-10-11:
As an update tho this issue, Using ClassicJerk thus disabling Junction Devation is a work around for this bug.

gretel, 2019-10-11:
made sure my jerk settings are non-zero (`M205`) but i got this plague, too.

looxonline, 2019-10-12:
Had the same issue but I just installed the 11.10.19 build and now it works. Roxy made some commits relating to junction deviation so they may have fixed it.

wffbos, 2019-10-12:
I still have the same issue. Cloned the bugfix branch 1hour ago. The problem seem to be less severe though. The stuttering movement is lesss, well, stuttering.
My steps to reproduce:
SKR v1.3
Tested with all TMC2208 uart  and all TMC2130 spi
G28
G29
X0 Y0 F10000
X150 Y150 F10000
Results in stuttering movement.

407s, 2019-10-12:
Yes. Confirmed. Was same situation.
CLASSIC_JERK activated -> no problem

boelle, 2019-10-22:
@Kokorone can you confirm what @GhostlyCrowd did works?

Kokorone, 2019-10-23:
Did a pull of today's build with manually set pulse delay width and hz as stated with GhostlyCrowd's fix. Also set up square wave stepping and it appears to be moving correctly. Still a bug though, since setting driver names in configuration.h should automatically set the correct values that I manually defined in configuration_adv.h.

GhostlyCrowd, 2019-10-24:
Well I'm back there definitely is something funny still happening. Im getting micro stuttering still that i didnt notice right away. And My co worker can hear a high pitched whine from my printer that i cant hear because its out of my audible range but he said has never been there before.

GhostlyCrowd, 2019-10-24:
Just went back to classic jerk and my co worker said the awful noise he could hear is gone.  The noise was just with the printer sitting idle after a G28 he could hear the noise

Kokorone, 2019-10-30:
Agreed with Ghostly, faster movements like travel are still having a few stutters here and there with JD enabled as of the latest build on 10/30

boelle, 2019-11-03:
@Kokorone i assume this is still an issue?

Kokorone, 2019-11-03:
Not to the same extent as the original bug filed but yes it’s still having a stutter issue. @boelle

boelle, 2019-11-11:
could it be that the MCU is overloaded and cant execute fast enough?

GhostlyCrowd, 2019-11-11:
> could it be that the MCU is overloaded and cant execute fast enough?

Possibly, but that would mean we're already outrunning our 32 bit platforms with basic features. as this occours on LPC1768 for me.

theofficialgman, 2019-11-13:
I seem to be experiencing something similar on 8 bit boards on the current version. When manual mesh bed leveling is active, the printer stops briefly after every segment interpolated between the start and end point of a move (SEGMENT_LEVELED_MOVES / LEVELED_SEGMENT_LENGTH in configuration.h). This occurs for all z values below the 5mm fade height which I have set. It seems that the buffer is not being populated or maybe the segments are being treated as individual lines where the printer has to come to a complete stop before starting the next line.

Kokorone, 2019-11-26:
I cant say if the steppers have a high pitch noise, since I cant hear it, but this issue appears to be fixed on recent marlin builds since I have reverted any stepper pulse changes and there is no stutter present with JD enabled. Closing issue

closed 2019-11-26 by Kokorone