MarlinFirmware/Marlin #15514: [BUG]  'JUNCTION_DEVIATION_MM' - not declared (module\configuration_store)

kubik256, 2019-10-11:
Hello,
I have strange error today while compilling bugfix 2.0.x:

```
Compiling .pio\build\STM32F103RC_bigtree\src\src\module\motion.cpp.o
Marlin\src\module\configuration_store.cpp: In static member function 'static void MarlinSettings::reset()':
Marlin\src\module\configuration_store.cpp:2239:43: error: 'JUNCTION_DEVIATION_MM' was not declared in this scope
     planner.junction_deviation_mm = float(JUNCTION_DEVIATION_MM);
                                           ^~~~~~~~~~~~~~~~~~~~~
Marlin\src\module\configuration_store.cpp:2239:43: note: suggested alternative: 'ENC_DEVIATION_t'
     planner.junction_deviation_mm = float(JUNCTION_DEVIATION_MM);
                                           ^~~~~~~~~~~~~~~~~~~~~
                                           ENC_DEVIATION_t
*** [.pio\build\STM32F103RC_bigtree\src\src\module\configuration_store.cpp.o] Error 1
================================================================================================= [FAILED] Took 168.51 seconds =================================================================================================

Environment              Status    Duration
-----------------------  --------  ------------
megaatmega2560           IGNORED
megaatmega1280           IGNORED
at90usb1286_cdc          IGNORED
at90usb1286_dfu          IGNORED
DUE                      IGNORED
DUE_USB                  IGNORED
DUE_debug                IGNORED
LPC1768                  IGNORED
LPC1769                  IGNORED
melzi                    IGNORED
melzi_optiboot           IGNORED
rambo                    IGNORED
sanguino_atmega644p      IGNORED
sanguino_atmega1284p     IGNORED
STM32F103RE              IGNORED
STM32F103RC_fysetc       IGNORED
STM32F103RC_bigtree      FAILED    00:02:48.508
STM32F103RC_bigtree_USB  IGNORED
STM32F4                  IGNORED
STM32F7                  IGNORED
ARMED                    IGNORED
STM32F103VE_longer       IGNORED
mks_robin                IGNORED
mks_robin_lite           IGNORED
mks_robin_mini           IGNORED
mks_robin_nano           IGNORED
jgaurora_a5s_a1          IGNORED
STM32F407VE_black        IGNORED
BIGTREE_SKR_PRO          IGNORED
BIGTREE_BTT002           IGNORED
teensy31                 IGNORED
teensy35                 IGNORED
STM32F103CB_malyan       IGNORED
esp32                    IGNORED
fysetc_f6_13             IGNORED
linux_native             IGNORED
SAMD51_grandcentral_m4   IGNORED
include_tree             IGNORED
============================================================================================ 1 failed, 0 succeeded in 00:02:48.508 ============================================================================================  
The terminal process terminated with exit code: 1
```

Same Configuration.h and Configuration_ADV.h has been successfully compiled at 4th October 2019 at 12:00 AM.

Thank you, BR ;)

[config_files.zip](https://github.com/MarlinFirmware/Marlin/files/3716304/config_files.zip)

kubik256, 2019-10-11:
Enabling `JUNCTION_DEVIATION` and define `JUNCTION_DEVIATION_MM` in Configuration.h also doesn't solve the problem:(
```
In file included from Marlin\src\HAL\HAL_STM32F1\../../inc/MarlinConfig.h:38:0,
                 from Marlin\src\HAL\HAL_STM32F1\HAL_SPI.cpp:35:
Marlin\src\HAL\HAL_STM32F1\../../inc/SanityCheck.h:408:4: error: #error "JUNCTION_DEVIATION is no longer required. (See CLASSIC_JERK). Please remove it from Configuration.h."
   #error "JUNCTION_DEVIATION is no longer required. (See CLASSIC_JERK). Please remove it from Configuration.h."
    ^~~~~
In file included from Marlin\src\HAL\HAL_STM32F1\../../inc/MarlinConfig.h:38:0,
                 from Marlin\src\HAL\HAL_STM32F1\HAL.cpp:31:
Marlin\src\HAL\HAL_STM32F1\../../inc/SanityCheck.h:408:4: error: #error "JUNCTION_DEVIATION is no longer required. (See CLASSIC_JERK). Please remove it from Configuration.h."
   #error "JUNCTION_DEVIATION is no longer required. (See CLASSIC_JERK). Please remove it from Configuration.h."
    ^~~~~
In file included from Marlin\src\HAL\HAL_STM32F1\../../inc/MarlinConfig.h:38:0,
                 from Marlin\src\HAL\HAL_STM32F1\Servo.cpp:26:
Marlin\src\HAL\HAL_STM32F1\../../inc/SanityCheck.h:408:4: error: #error "JUNCTION_DEVIATION is no longer required. (See CLASSIC_JERK). Please remove it from Configuration.h."
   #error "JUNCTION_DEVIATION is no longer required. (See CLASSIC_JERK). Please remove it from Configuration.h."
    ^~~~~
In file included from e:\marlin-bugfix-2.0.x\marlin\src\inc\marlinconfig.h:38:0,
                 from Marlin\src\HAL\HAL_STM32F1\dogm\u8g_com_stm32duino_fsmc.cpp:29:
e:\marlin-bugfix-2.0.x\marlin\src\inc\SanityCheck.h:408:4: error: #error "JUNCTION_DEVIATION is no longer required. (See CLASSIC_JERK). Please remove it from Configuration.h."
   #error "JUNCTION_DEVIATION is no longer required. (See CLASSIC_JERK). Please remove it from Configuration.h."
    ^~~~~
*** [.pio\build\STM32F103RC_bigtree\src\src\HAL\HAL_STM32F1\HAL.cpp.o] Error 1
*** [.pio\build\STM32F103RC_bigtree\src\src\HAL\HAL_STM32F1\HAL_SPI.cpp.o] Error 1
*** [.pio\build\STM32F103RC_bigtree\src\src\HAL\HAL_STM32F1\Servo.cpp.o] Error 1
*** [.pio\build\STM32F103RC_bigtree\src\src\HAL\HAL_STM32F1\dogm\u8g_com_stm32duino_fsmc.cpp.o] Error 1
================================================================================================== [FAILED] Took 9.07 seconds ==================================================================================================
```

GMagician, 2019-10-11:
I think you have to update your cnf file...

> error: #error "JUNCTION_DEVIATION is no longer required. (See CLASSIC_JERK). Please remove it from Configuration.h."

Now you have to define CLASSIC_JERK for old stile otherwise junction is enabled and mm has to be present of course

kubik256, 2019-10-11:
THANKS! Old configuration from example or stock one from manufacturer doesn't work anymore.
Sucessfully compiled with this:

```
//#define JUNCTION_DEVIATION
//#if ENABLED(JUNCTION_DEVIATION)
  #define JUNCTION_DEVIATION_MM 0.065  // (mm) Distance from real junction edge
//#endif
```

Whole JUNCTION_DEVIATION block and IF clausule must be commented out ;)

closed 2019-10-11 by kubik256

thisiskeithb, 2019-10-11:
> Whole JUNCTION_DEVIATION block and IF clausule must be commented out ;)

I'd recommend comparing your Configuration.h with the default to see what's changed. This is the new code block:

https://github.com/MarlinFirmware/Marlin/blob/1df90aaf0ef50efedbb0e238b44cf0eabdcc7abc/Marlin/Configuration.h#L779-L788