MarlinFirmware/Marlin #15516: [BUG] [bugfix-2.0.x] Linear-Advance going crazy when printing circles

Nebukadneza, 2019-10-11:
### Description

I’m running Marlin bugfix-2.0.x on commit `639f526f3f84de8fd914c9695d9bd0f84a046b73` on a CR-10 with Titan extruder, E3D V6 hotend and a SKR 1.3 Board with A4988 drivers on all axes.

My configuration does include Linear-Advance, S-Curve Acceleration and Junction Deviation instead of traditional Jerk. I tuned LA using the test-pattern generator. The rest of my config is at:
https://github.com/Nebukadneza/Marlin/blob/skr13cr10/Marlin/Configuration.h
https://github.com/Nebukadneza/Marlin/blob/skr13cr10/Marlin/Configuration_adv.h

The problem I’m seeing is that when printing circle-structures, such as simple test pieces like https://www.thingiverse.com/thing:856956 , linear-advance goes crazy in the circle-motion. What it does is to constantly jiggle the extruder forward and backward very quickly — and of course slightly more forward. When feeling the filament in front of the extruder, i’d say it’s »retract-0.5mm, forward 0.55mm, retract 0.5mm, forward 0.55mm« … in 4-6×second or so intervals. This leads to very bad quality in circles.

### Steps to Reproduce

1. Activate and tune Linear-Advance on bugfix-2.0.x
2. Use a simple model including circloid structures, such as https://www.thingiverse.com/thing:856956
3. Print, observing the extruder on outer or inner shells, where the movement is »round«

**Expected behavior:** Linear-Advance keeps a constant extrusion speed, since combined X-Y movement speed is (at least seemingly) constant in a circle.

**Actual behavior:** Linear-Advance seems to regulate on the individual circle-segments, making the Extruder jiggle backward- and forward real quick.

#### Additional Information
* [configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3717089/configuration.zip)

Lord-Quake, 2019-10-11:
I thought I'd do a test but upon loading the object in my slicer I get 48 non-manifold edges and 12 error orientation faces. It also looks like the object has a low poly count.
Here is my version to test: [Cylinder_Test.zip](https://github.com/MarlinFirmware/Marlin/files/3718702/Cylinder_Test.zip)
I'm curious if you have the same result.

Nebukadneza, 2019-10-11:
Oh, I’m sorry! That was a horrible copy’n’paste mistake — I linked the wrong test-model. I meant to link this one (also corrected in original post)
https://www.thingiverse.com/thing:2110628
Yours is simple and fine too, thanks for pointing me to the mistake!

GhostlyCrowd, 2019-10-11:
Can you try enabbling classic jerk. if this fixes it then its the same issue we have https://github.com/MarlinFirmware/Marlin/issues/15513

gretel, 2019-10-11:
will try with classic jerk enabled.

Lord-Quake, 2019-10-11:
I did the test using STRING_DISTRIBUTION_DATE "2019-10-08" and can confirm the problem.
With classic jerk disabled I get stutter movements while printing.
With classic jerk enabled the movements are smooth as expected.

Kokorone, 2019-10-11:
Looks to be an additional issue extension of #15513 then

Kokorone, 2019-10-11:
> I did the test using STRING_DISTRIBUTION_DATE "2019-10-08" and can confirm the problem.
> With classic jerk disabled I get stutter movements while printing.
> With classic jerk enabled the movements are smooth as expected.

Can you pull the marlin image from 10/5/2019 and try to compile with JD enabled? In my testing 10/5/2019 has no issues

Lord-Quake, 2019-10-11:
The last version I used was STRING_DISTRIBUTION_DATE "2019-10-04" and JD active. No problems occurred.
Could you please post the link for 10/5/2019 as I'm not so familiar with GitHub.

gretel, 2019-10-11:
@Kokorone can you get the revision id please?

Kokorone, 2019-10-11:
https://github.com/MarlinFirmware/Marlin/tree/fc6a0937b86f501b39263d7f4dfc7a4fa70b2636
There are probably newer versions prior to 10/10/2019 that are also working but I've only tested with 10/5/2019 and 10/10/2019.

gretel, 2019-10-11:
enabling `CLASSIC_JERK` (disabling `JUNCTION_DEVIATION` makes my prints look good again.

looxonline, 2019-10-12:
Compile the latest commit and it should work fine again. Had the same issue and compiled earlier and it works.

Lord-Quake, 2019-10-12:
I just tried 10/10/2019 and it looks good with JD active.

Nebukadneza, 2019-10-12:
I didn’t have time to test yesterday, but will do now! Thanks for your replies

One thing that confuses me:
> Compile the latest commit and it should work fine again. Had the same issue and compiled earlier and it works.
I don’t see any commits related to JD or LA movement or core-calculation in the days between 05.10.2019 and 10.10.2019 (I’m still confused by date formats :sweat_smile: :sweat_drops: ) … do you have any inkling where it could have happened? (I only see the enable-JD-as-default one, which seems to do a lot around config/disable/enable-JD, but not the logic …)

Anyhow, I’ll try all 3: disable in current state, rebase my config on latest `bugfix-2.0.x` and test newest commit of day 05.10.19 …. I’ll report back!

Nebukadneza, 2019-10-12:
So, my results:
* Current `bugfix-2.0.x`  (`59378db7de0dec95d9c9fba6bcefc1c523ad6ef2`) with JD and LA produces this: http://nextcloud.ghostdub.de/s/gqbEbTnPSepJDi5 (bad stuttering)
* `fc6a0937b86f501b39263d7f4dfc7a4fa70b2636` with JD and LA also stutters, but it seems at a slightly lower frequency than current
* My previous state, `639f526f3f84de8fd914c9695d9bd0f84a046b73` also exhibits the problem
* When disabling JD, all 3 states work fine and extrude smoothly

What’s left is maybe to repeat the test without `S_Curve` acceleration, because that might also play into it (looking at #14728 ?).

looxonline, 2019-10-12:
Roxy made a bunch of commits yesterday setting the correct defaults for JD for each machine. After that mine came right. Don't have s-curve enabled though.

gretel, 2019-10-12:
did try without `s_curve` - made no difference for me.

Nebukadneza, 2019-10-12:
Mh, I’ve checked the big JD commit, and re-set my JD value from 0.028 (which came out of the equation linked in the blogpost in `Configuration.h`) to the 0.08 from the CR-10 examples. This led to a change — the glitching is much more high-frequency now, and seems to have a smaller amplitude — but it still goes batshit crazy in circles.
When disabling either LA or going to classic jerk, it extrudes smoothly. At least on current `bugfix-2.0.x`, s-curve does not influence this.

gretel, 2019-10-12:
also tried without `ADAPTIVE_STEP_SMOOTHING` which did not change anything quality-wise.. just the sounds.

GhostlyCrowd, 2019-10-17:
Fresh Pull today. Used Roxy 3D's equations (which are linked to a blog in the config comments) to set my JD settings. Also manually Set my Stepper puls delay width and hz in Configuration_adv.h rather then letting it auto config from Configuration.h settings, and I'm printing perfectly fine now.

No more pulsing steppers.

Nebukadneza, 2019-10-20:
Hi,

sorry for the late reply — and thanks for the tip! Indeed, i did pull again to `f2ce4a25a3600749ab043c737d34933cd27d930f` although i didn’t see any changes that I’d relate to our motion problems here … enabled `LIN_ADV`, `S_CURVE_ACCELERATION` and `ADAPTIVE_STEP_SMOOTHING`, disabled `CLASSIC_JERK` and went to print a cylinder …
… and it was horrible again, as expected.

Then i did change the `MINIMUM_STEPPER_POST_DIR_DELAY`, `MINIMUM_STEPPER_PRE_DIR_DELAY`, `MINIMUM_STEPPER_PULSE` and `MAXIMUM_STEPPER_RATE` to the values recommended for A4988 in the comments above. Print again, and … mhhh, it’s better, there’s no strange humming noise without visible movement from the extruder, but it’s still going back and forth visibly in 1/8th or so whole circle increments.

Then, I went »fugg dis«, and set the values somewhere in the middle region of all stepper driver models … knowing i’m printing fairly slowly anyways.
```
   #define MINIMUM_STEPPER_POST_DIR_DELAY 650
   #define MINIMUM_STEPPER_PRE_DIR_DELAY 650
   #define MINIMUM_STEPPER_PULSE 4
   #define MAXIMUM_STEPPER_RATE 200000
```
(i didn’t get around to isolating which one does the magic)

This … gave me smooth prints! I can now use LA, JD, SCurve and Adaptive Smoothing all together! And I can print circles without LA going crazy or without weird noise …

… the only question that for me as a Marlin noob remains: Is this ok~ish or intended behavior, or should it be considered a bug?

In any case, thanks for your hints!

boelle, 2019-10-22:
will close this one as it seems "fixed"

we can always reopen

closed 2019-10-22 by boelle

Nebukadneza, 2019-10-22:
Fine by me, thanks a lot!

thierryzoller, 2020-04-09:
I have the exact same problem as the Original Reporter. Can reproduce when printing circles.

thinkyhead, 2020-04-27:
@thierryzoller — File a new issue.

qwewer0, 2020-04-27:
@thierryzoller Could you try the latest bugfix?

thierryzoller, 2020-04-27:
I planned on doing so, merged the latest bugfix-2.0. Will try again, document and open an issue in case i can reproduce. I have my head scratching my bed on extremeties with G29 currently.

thierryzoller, 2020-04-27:
My printer head - that is. :D