MarlinFirmware/Marlin #15517: [BUG] SKR1.3 cannot be flashed with EXT2 attached MKS mini12864

karabas2011, 2019-10-11:
### Description

SKR1.3 + LCD mks mini 12864
MARLIN 2.0 from SKR github

SKR hangs during flashing.
If I disconnect ext2  it is flashed successfully

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]
Success flashing

**Actual behavior:** [What actually happens]

hangs. Does not boot.  Even need to format micrtosd card.
If I disconnect LCD it is flashed successfully. Then I connect LCD and use printer as usually.
I thing it is related to shared SPI of secondary sdcard on lcd

thisiskeithb, 2019-10-11:
> MARLIN 2.0 from SKR github

You’ll need to post in their issue queue or try downloading a copy of `bugfix-2.0.x` from this repo and see if your issue persists.

boelle, 2019-10-24:
closing this one as we cant support SKR fork

closed 2019-10-24 by boelle