MarlinFirmware/Marlin #15520: [Marlin 2.0] Error:Field hotendPID mismatch

RCGe0rge, 2019-10-11:
I have some problems with EEPROM

there is what pc displays me when I try M50X codes
 
>>>M501
SENDING:M501
echo:EEPROM version mismatch (EEPROM=ERR Marlin=V64)
echo:Hardcoded Default Settings Loaded
>>>M503
SENDING:M503
  G21    ; Units in mm (mm)
  M149 C ; Units in Celsius
  M200 D1.75
  M200 D0
  M92 X80.00 Y80.00 Z1627.00 E99.00
  M203 X200.00 Y200.00 Z5.00 E25.00
  M201 X3000.00 Y3000.00 Z100.00 E10000.00
  M204 P3000.00 R3000.00 T3000.00
  M205 B20000.00 S0.00 T0.00 X10.00 Y10.00 Z0.30 E5.00
  M206 X0.00 Y0.00 Z0.00
  M420 S0 Z0.00
  M145 S0 H210 B60 F0
  M145 S1 H240 B110 F0
  M145 S2 H240 B75 F0
  M145 S3 H240 B0 F0
  M145 S4 H280 B120 F0
  M145 S5 H280 B120 F0
  M145 S6 H200 B40 F0
  M145 S7 H200 B60 F0
  M145 S8 H0 B0 F0
  M145 S9 H0 B0 F0
  M145 S10 H0 B0 F0
  M145 S11 H0 B0 F0
  M301 P8.71 I0.43 D44.31
  M304 P198.62 I21.57 D457.31
  M851 Z0.00
  M906 X1000 Y1000 Z2000
  M906 T0 E500
  M913 X130 Y130 Z5
  M913 T0 E50
  M569 S1 X Y Z
  M569 S1 T0 E
>>>M500
SENDING:M500
Error:Field hotendPID mismatch.
[ERROR] Error:Field hotendPID mismatch.

>>>M504
SENDING:M504
echo:EEPROM version mismatch (EEPROM=ERR Marlin=V64)
>>>M502
SENDING:M502
echo:Hardcoded Default Settings Loaded
>>>M502
SENDING:M502
echo:Hardcoded Default Settings Loaded
>>>M501
SENDING:M501
echo:EEPROM version mismatch (EEPROM=ERR Marlin=V64)
echo:Hardcoded Default Settings Loaded

>>>M500
SENDING:M500
Error:Field hotendPID mismatch.
[ERROR] Error:Field hotendPID mismatch.

I use BIQU SKR1.3

I have uncommanded these commands
from configuration.h
`#define EEPROM_SETTINGS`
`#define EEPROM_CHITCHAT`

and from configuration_store.cpp
`#define DEBUG_EEPROM_READWRITE`

In my opinion, I tried maybe all.

can anybody help me with this issue

thisiskeithb, 2019-10-11:
If you send `M502` then `M500` immediately after, what does `M503` report?

RCGe0rge, 2019-10-11:
`M500` dont work
here is what my pc displays

Connecting...
Printer is now online.
>>>M502
SENDING:M502
echo:Hardcoded Default Settings Loaded
>>>M500
SENDING:M500
Error:Field hotendPID mismatch.
[ERROR] Error:Field hotendPID mismatch.

>>>M503
SENDING:M503
  G21    ; Units in mm (mm)
  M149 C ; Units in Celsius
  M200 D1.75
  M200 D0
  M92 X80.00 Y80.00 Z1627.00 E99.00
  M203 X200.00 Y200.00 Z5.00 E25.00
  M201 X3000.00 Y3000.00 Z100.00 E10000.00
  M204 P3000.00 R3000.00 T3000.00
  M205 B20000.00 S0.00 T0.00 X10.00 Y10.00 Z0.30 E5.00
  M206 X0.00 Y0.00 Z0.00
  M420 S0 Z0.00
  M145 S0 H210 B60 F0
  M145 S1 H240 B110 F0
  M145 S2 H240 B75 F0
  M145 S3 H240 B0 F0
  M145 S4 H280 B120 F0
  M145 S5 H280 B120 F0
  M145 S6 H200 B40 F0
  M145 S7 H200 B60 F0
  M145 S8 H0 B0 F0
  M145 S9 H0 B0 F0
  M145 S10 H0 B0 F0
  M145 S11 H0 B0 F0
  M301 P8.71 I0.43 D44.31
  M304 P198.62 I21.57 D457.31
  M851 Z0.00
  M906 X1000 Y1000 Z2000
  M906 T0 E500
  M913 X130 Y130 Z5
  M913 T0 E50
  M569 S1 X Y Z
  M569 S1 T0 E

thisiskeithb, 2019-10-11:
Can you zip both your `Configuration.h` & `Configuration_adv.h` files and attach them?

tpruvot, 2019-10-11:
Marlin=V64, means its not current code... we are at V70... has months so

RCGe0rge, 2019-10-12:
yes
but now I am not at home
I am coming home approximately at 7 pm

thisiskeithb, 2019-10-12:
Pull down a fresh copy of `bugfix-2.0.x` to make sure you’re using the most up to date version of Marlin.

RCGe0rge, 2019-10-12:
here you have my configuration.h and configuration_adv.h

[marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3720920/marlin.zip)

RCGe0rge, 2019-10-12:
and here is readme from my marlin
i think its the newest

[README.zip](https://github.com/MarlinFirmware/Marlin/files/3720922/README.zip)

boelle, 2019-10-22:
@RCGe0rge as @tpruvot says your copy is some time behind

download from here, and you can use a tool like winmerge to compare your old settings to the new ones and copy over old config

RCGe0rge, 2019-10-22:
Ok
I tried update my marlin but there was few issues

1 i had old tmc library but it wasnt hard to update it

2 after i updated my config and compiling marlin my printer after few seconds displayed me tmc error 8

So i found on github this issue 
It qas because i use tmc5160 and in library tmc there are error about timing or something like that
They recommended unplug one clk pin
Now all works but motor on tmc5160 is whistling

What you recommend for motor whistling

boelle, 2019-10-22:
what motor voltage? 12 or 24?

RCGe0rge, 2019-10-22:
24V

boelle, 2019-10-22:
then its strange.... 24V is the recommended for TMC

RCGe0rge, 2019-10-26:
Hmm
I dont think it is a noise in hardware

Maybe it could be bad settings of current

I have stealthchop on this driver (TMC5160) and i have 2 motors connected in parallel 

Im thinking about 2 drivers (TMC2208) for each motor separately

RCGe0rge, 2019-10-26:
And it whistling just few seconds 
After maybe 3 seconds a frequency begins to slowly decrease to silence

I think it is because i removed clk pin from driver
I had to because the display was showing error and that was the solution

thisiskeithb, 2019-10-26:
> I think it is because i removed clk pin from driver
> I had to because the display was showing error and that was the solution

This is standard depending on who made your driver and revision. I run genuine first-gen (v1.3?) Watterott TMC5160s on my Ender-3 (SKR 1.3) and also had to cut the CLK pin and tie it to GND for SPI to work properly.

The motors hissed/buzzed/whined when the current was cranked way too high for the stock Creality steppers. Once those were tuned properly, the motors are silent until they kick into SpreadCycle mode.

RCGe0rge, 2019-10-26:
Ok
I will try to reduce the current

RCGe0rge, 2019-10-27:
Hmm
Interesting

I updated pid values in marlin and send M502 and M500 to the printer and i changed nothing with motors and now they are silence

RCGe0rge, 2019-10-27:
Now i haven't got any problem with my printer so i close this. Thanks guys for your time

closed 2019-10-27 by RCGe0rge