MarlinFirmware/Marlin #15522: [BUG] (Compiling Marlin bugfix 2.0.x  error : conflicting declaration TMC5160Stepper)

CharlesSee, 2019-10-11:
When Compiling Marlin 2.0.x I get the below compile error.
I'm trying to compile for a MKS Gen L board with CNC Shield with 1 mixing Hotend with 5 Extruders using SPI cabling and TMC5160 drivers.
I have no idea as to what this error means.
----

Compiling .pio\build\megaatmega2560\src\src\sd\SdFile.cpp.o

Marlin\src\module\stepper\trinamic.cpp:49:84: error: conflicting declaration 'TMCMarlin<TMC5160Stepper, 'E', '2', (AxisEnum)3u> stepperE2'
   #define __TMC_SPI_DEFINE(IC, ST, L, AI) TMCMarlin<IC##Stepper, L, AI> stepper##ST(ST##_CS_PIN, ST##_RSENSE)
                                                                                    ^
Marlin\src\module\stepper\trinamic.cpp:55:37: note: in expansion of macro '__TMC_SPI_DEFINE'

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3718226/Marlin.zip)
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3718241/Marlin.zip)
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3718243/Marlin.zip)
 #define _TMC_SPI_DEFINE(IC, ST, AI) __TMC_SPI_DEFINE(IC, ST, TMC_##ST##_LABEL, AI)
                                     ^
Marlin\src\module\stepper\trinamic.cpp:56:32: note: in expansion of macro '_TMC_SPI_DEFINE'
 #define TMC_SPI_DEFINE(ST, AI) _TMC_SPI_DEFINE(ST##_DRIVER_TYPE, ST, AI##_AXIS)
                                ^
Marlin\src\module\stepper\trinamic.cpp:65:32: note: in expansion of macro 'TMC_SPI_DEFINE'
   #define TMC_SPI_DEFINE_E(AI) TMC_SPI_DEFINE(E##AI, E)
                                ^

teemuatlut, 2019-10-11:
I wasn't able to reproduce the errors with your config files (+whatever was necessary to pass the sanitychecks).
Try updating.

CharlesSee, 2019-10-15:
This no longer occurs in the latest bugfix 2.0.x.

closed 2019-10-15 by CharlesSee