MarlinFirmware/Marlin #15526: [BUG] [bugfix-2.0.x] Error compile with CUSTOM_USER_MENU_TITLE

kpter, 2019-10-11:
### Description

Error when compile with uncommented `#define CUSTOM_USER_MENU_TITLE` in Configuration_adv.h
When commented - no error.

#### Additional Information

Compiling .pio\build\megaatmega2560\src\src\lcd\menu\menu_main.cpp.o
Compiling .pio\build\megaatmega2560\src\src\lcd\menu\menu_media.cpp.o
In file included from Marlin\src\lcd\menu\menu_main.cpp:31:0:
Marlin\src\lcd\menu\menu_main.cpp: In function 'void menu_main()':
Marlin\src\lcd\menu\../../inc/../../Configuration_adv.h:2502:34: error: expected unqualified-id before string constant
   #define CUSTOM_USER_MENU_TITLE "- MY -"
                                  ^
Marlin\src\lcd\menu\menu.h:341:28: note: in definition of macro '_MENU_ITEM_P'
       PGM_P const plabel = PLABEL;                                \
                            ^
Marlin\src\lcd\menu\menu.h:391:48: note: in expansion of macro 'MENU_ITEM_P'
 #define SUBMENU_P(PLABEL, DEST)                MENU_ITEM_P(submenu, PLABEL, DEST)
                                                ^
Marlin\src\lcd\menu\menu.h:392:50: note: in expansion of macro 'SUBMENU_P'
 #define SUBMENU(LABEL, DEST)                     SUBMENU_P(GET_TEXT(LABEL), DEST)
                                                  ^
Marlin\src\lcd\menu\menu.h:392:60: note: in expansion of macro 'GET_TEXT'
 #define SUBMENU(LABEL, DEST)                     SUBMENU_P(GET_TEXT(LABEL), DEST)
                                                            ^
Marlin\src\lcd\menu\menu_main.cpp:177:5: note: in expansion of macro 'SUBMENU'
     SUBMENU(MSG_USER_MENU, menu_user);
     ^
Marlin\src\lcd\menu\menu_main.cpp:175:29: note: in expansion of macro 'CUSTOM_USER_MENU_TITLE'
       #define MSG_USER_MENU CUSTOM_USER_MENU_TITLE
                             ^
Marlin\src\lcd\menu\menu_main.cpp:177:13: note: in expansion of macro 'MSG_USER_MENU'
     SUBMENU(MSG_USER_MENU, menu_user);
             ^
*** [.pio\build\megaatmega2560\src\src\lcd\menu\menu_main.cpp.o] Error 1
============================= [FAILED] Took 139.82 seconds =============================

Bergerac56, 2019-10-12:
I confirm this issue. 59378db7de0dec95d9c9fba6bcefc1c523ad6ef2 does not compile with #define CUSTOM_USER_MENU_TITLE

stefi01, 2019-10-14:
same for me, will not compile with custom menu enabled

Bergerac56, 2019-10-14:
In fact, it compiles with only #define CUSTOM_USER_MENU_TITLE disabled. The option will correctly appear on the LCD menu.

Bergerac56, 2019-10-17:
This is now solved by 40de135d4dd1462f0c13cc3bf56f3abdc72af69a

I suppose this issue can be closed