MarlinFirmware/Marlin #15528: [BUG] Problem with disabling Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN on a STM32 board

tilk, 2019-10-11:
### Description

I have a BIGTREETECH SKR E3 DIP 1.0 board, using Marlin 2.0 beta, and I'm trying to get BLTOUCH working by connecting it to the dedicated sensor connector. However, commenting out Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN in the configuration leads to disabling of all endstops (but the BLtouch works).

I traced the problem to the `_HAS_STOP` and `IS_Z2_OR_PROBE` macros. Here is the definition:

```C++
#define IS_Z2_OR_PROBE(A,M) ( \
     (Z_MULTI_ENDSTOPS && Z2_USE_ENDSTOP == _##A##M##_) \
  || (USES_Z_MIN_PROBE_ENDSTOP && Z_MIN_PROBE_PIN == A##_##M##_PIN ) )
#define _HAS_STOP(A,M) (PIN_EXISTS(A##_##M) && !IS_X2_ENDSTOP(A,M) && !IS_Y2_ENDSTOP(A,M) && !IS_Z2_OR_PROBE(A,M))
```

It turns out that, on platformio for STM32, pin names are not `#define`s, they are `enum`s, and as such not visible to the C preprocessor. And by the rules of C preprocessor, any undefined identifier in `#if` means zero, so the equality check in `IS_Z2_OR_PROBE` always succeeds, leading to the result described.

### Steps to Reproduce

1. Comment out Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
2. Compile and upload firmware to the printer
3. Send M119

**Expected behavior:** Status of all endstops

**Actual behavior:** Status of BLTOUCH only

#### Additional Information

The problem can be worked around by adding `#define`s for pin numbers in board pin header. But that's ugly.

tpruvot, 2019-10-12:
pins are now defined in Marlin/src/HAL/HAL_STM32F1/fastio.h for this reason...

tilk, 2019-10-12:
Looks like the issue is solved here, it just needs to be merged downstream.

closed 2019-10-12 by tilk