MarlinFirmware/Marlin #15529: TOUCH MI with NOZZLE_TO_PROBE_OFFSET

boubouexpress, 2019-10-12:
Hello

with the latest version of Marlin 2
I have a difference of 10 mm on NOZZLE_TO_PROBE_OFFSET of Z

I go from -3.80 to -13.80 which is not possible to exceed -10 in marlin

marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3720621/marlin.zip)

merci

kramer04, 2019-10-14:
issue too with NOZZLE_TO_PROBE_OFFSET 
it seems not work because nozzle is out on the right X

boelle, 2019-10-22:
maybe attach config so we know a bit about setup

boelle, 2019-10-30:
closing this one due to missing configs

we can reopen when we have those

kramer04, 2019-11-01:
i've attached the cong.zip . Don't you see it ?

boelle, 2019-11-03:
@kramer04 you cant attach files from email, you have to go to github for that

but we need config from @boubouexpress

closed 2019-11-03 by boelle