MarlinFirmware/Marlin #15532: MKS Robin tft V2.0 don't work

fizga, 2019-10-12:
Hello, I have a problem with Robin TFT V2.0 (3.2'') connected to the motherboard MKS Robin 2.4 I use Marlin 2.0!  compilation is going well, I put the file Robin.bin on the SD card but when I turn on the screen shows the progress bar, it arrives at 100% then displays COMPLETE then it stays like that.  there is just a blue LED on the flashing motherboard.

boelle, 2019-10-12:
where are your config files, the issue template says to attach them

tpruvot, 2019-10-13:
cp config/examples/Mks/Robin/Configuration* Marlin/

i guess... :p

fizga, 2019-10-13:
Sorry, I'm traveling this weekend, I'll give all the files tomorrow

fizga, 2019-10-14:
here are the files

[config.zip](https://github.com/MarlinFirmware/Marlin/files/3723166/config.zip)

fizga, 2019-10-14:
for more details, the blinking blue led is the one connected to PB2.

tpruvot, 2019-10-14:
yes, the heartbeat/watchdog led... Is there any other user of this mks robin able to flash and use a serial connection ? Still not shipped, after a week...

boelle, 2019-10-14:
you just love china stuff :-D
ni hao

tpruvot, 2019-10-14:
well, delivery time is another issue then, but that help a bit when they ship it without delay... first

boelle, 2019-10-14:
shipping is one thing
customer service is a city in the center of siberia
quality is sometimes a joke

tpruvot, 2019-10-14:
[image]

Louphoque...

fizga, 2019-10-15:
I tried again to upload by serial link but I still have an error.
yet when I look at serial port on platforio I see USB-SERIAL CH340
for the serial monitor the bauderate must be 250000?

fizga, 2019-10-15:
[image]
here what i have in serial monitor
and here when i try to upload 
[image]

boelle, 2019-10-24:
are you sure this is not due to a bad made board? ie hardware failure?

fizga, 2019-10-25:
I'm sure that the hardware works, when I load the original firmware it works well but I am limited in options

https://github.com/makerbase-mks/MKS-Robin 

After installing marlin 2.0 I still have screen that does not work!  So I tried to communicate in serial with s3d and it works, I have a return of the temperatures

github051, 2019-10-25:
HI FIZA,
Can you share your entire marlin 2.0 folder ?

tpruvot, 2019-10-25:
ok, so its only a tft communication or init issue... i received it, but still have to wire some cables to make some backup with the stlink before... if its even possible...

tpruvot, 2019-10-25:
so indeed, i can reproduce the issue... that stay on "Complete", heatbeat led is working, touchscreen also doing beeps when you click on the bottom of the screen, but nothing more displayed, look like some missing tft reset or such

boelle, 2019-10-25:
have changed labels to reflect this

tpruvot, 2019-10-25:
yep, uncommenting PF6 make a white screen, we can see the small reset trick, but initial data transfert may be problematic... will need more time to be able to debug that, trace some lcd id value over serial...

tpruvot, 2019-10-25:
so, for now, i can say the 3.2" TFT V2.0 is a ST7789V (0x8552 TFT Identifier)... Tried various changes on [original repo](https://github.com/makerbase-mks/MKS-Robin/blob/master/MKS%20Robin/firmware/Marlin2.0-MKS-Robin/Marlin/src/HAL/HAL_STM32F1/u8g_com_stm32duino_fsmc.cpp) (like reducing the init delay to 50ms, and changing the DMA2 channel CH4&CH5 so, to match the longer3D board, without issue... on their repo)

Touchscreen seems well calibrated by default... also... no issue about it, unrelated to the problem i think... so.

Its kinda weird there is not even a pixel changed, even on reinit...

tpruvot, 2019-10-25:
ok, i think i found the issue, -DSTM32_XL_DENSITY is not automatically set by the framework, unlike -DSTM32_HIGH_DENSITY

tpruvot, 2019-10-25:
fixed in PR #15675

github051, 2019-10-26:
HI all,

Thanks for every one in this loop. Finally i'm able to complie the fw in single shot and the best part os its working nice including SD card printing.

Thanks to all once again.

[image]

boelle, 2019-10-26:
oki will close this one as fixed, 

we can always reopen

closed 2019-10-26 by boelle

fizga, 2019-10-26:
Thank you all nice job