MarlinFirmware/Marlin #15534: [BUG] Junction Deviation not saving / set at 0 when machine turned on

DkOuvE, 2019-10-12:
Running latest Marlin 2.0 bugfix and Junction Deviation is not saving after a power cycle. All other values are saved. It resets to 0.000 when the minimum settable in the LCD menu is 0.010. Using an SKR.

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3720792/Configuration.zip)

Roxy-3D, 2019-10-12:
> It resets to 0.000 when the minimum settable in the LCD menu is 0.010.

If this is the case...   We need to change that.   I know there are some machines that need a value more like .003 to accurately reflect the X-Jerk, Y-Jerk and Default Acceleration values.

Lord-Quake, 2019-10-13:
@Roxy-3D Thanks for the update.
You set the new default values to 0.0025f, 0.5f respectively.
Wouldn't it make more sense to set the lowest value to 0.001 since setting the values via display will only use 3 digits after the comma?
I'm also suggesting 0.001 on the assumption that the value must be >0.

Roxy-3D, 2019-10-13:
Yes...   I think you are correct.

BastR, 2019-10-17:
@Roxy-3D  thanks for your update, but  unfortunately it still doesn't work, I tested with your 2 commits and in my case, I still have the problem.

Unlike @DkOuvE , I have an SKR PRO and not an SKR 1.3. I use emulated eeprom and when I restart the printer junction deviation is 0.
I tried doing M500 and then M501 each time I restart and it doesn't work either, moreover when I make a M501 the deviation junction value doesn't appear in the console, the problem may be here.
My workaround is to navigate in the LCD to the jerk menu and put my junction deviation manually every time I restart the printer.

boelle, 2019-10-22:
is this still an issue? there was an update that allowed a wider range of values that might have fixed it

trisomeyr, 2019-10-23:
Works for me with the latest Build. JD is now also shown in Terminal using M503.

BastR, 2019-10-23:
Good news, I will try with the last build and post feedback too

boelle, 2019-10-23:
will close this one as fixed

we can always reopen

closed 2019-10-23 by boelle