MarlinFirmware/Marlin #15537: temperature display error

hn0550, 2019-10-12:
What's the problem?
I found that my extruder heating temperature display error is large, heating to 180 degrees, the motherboard stopped heating immediately disappeared about 250 degrees. I am worried because I configured incorrectly and I used the default firmware still the same. But I used 2 thermistors on the extruder, one of them on the hot bed. Once the heating starts, the hot bed displays a normal temperature, while the extruder appears very cold. The hot bed temperature is 180 degrees. my thermistors is 100K

boelle, 2019-10-12:
attach 2 config files just as the issue template told you to

hn0550, 2019-10-13:
> attach 2 config files just as the issue template told you to
I'm sorry
The configuration file is as follows
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3721934/Configuration.zip)

boelle, 2019-10-22:
reading this again i was thinking that maybe the thermistor is bad?

hn0550, 2019-10-22:
> reading this again i was thinking that maybe the thermistor is bad?
---------------------------------------
Thermistors are good. I use two of the same thermistors for monitoring. One extruder, another hot bed. The actual monitoring results are shown in the attached pictures. I'm using Chinese
[image]

hn0550, 2019-10-26:
> 只是好奇，您从哪里下载了marlin？
-------------------------------------------------
https://github.com/MarlinFirmware/Marlin/issues/15537#issuecomment-545830398
I found that the temperature monitoring lags behind when it is heated all the time, but there is no problem with the constant temperature in the machine printing process.

boelle, 2019-10-26:
give direct link to where you get marlin from

hn0550, 2019-10-26:
> 直接链接到您从哪里获得马林鱼的地方
---------------------------------------
https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x
It's right here

boelle, 2019-10-26:
yes that is the correct place

sometimes we see people get marlin from other places

hn0550, 2019-10-26:
I don't know what the problem is now, the hardware confirmation is good, I use ramps1.4+2560, and the result is the same with due+FDv2.  The temperature monitoring is wrong when heating, but the printing temperature is not a problem

closed 2019-10-27 by hn0550