MarlinFirmware/Marlin #15538: [BUG] (error compiling with TMC2209 and SKR Pro v1.1)

psim, 2019-10-12:
### Description

Error compiling latest build. Did not have before. When compiling now you need to modify also the pins_BIGTREE_SKR_PRO_V1.1.h file in the Hardware serial part, otherwise you get an error:

#if HAS_TMC220x
  /**
   * TMC2208/TMC2209 stepper drivers
   *
   * Hardware serial communication ports.
   * If undefined software serial is used according to the pins below
   */
  //#define X_HARDWARE_SERIAL  Serial
  //#define X2_HARDWARE_SERIAL Serial1
  //#define Y_HARDWARE_SERIAL  Serial1
  //#define Y2_HARDWARE_SERIAL Serial1
  //#define Z_HARDWARE_SERIAL  Serial1
  //#define Z2_HARDWARE_SERIAL Serial1
  //#define E0_HARDWARE_SERIAL Serial1
  //#define E1_HARDWARE_SERIAL Serial1
  //#define E2_HARDWARE_SERIAL Serial1
  //#define E3_HARDWARE_SERIAL Serial1
  //#define E4_HARDWARE_SERIAL Serial1

Obviously if you modify this, printer do not work at all.

### Steps to Reproduce

**Expected behavior:** [No need to modify the pins file]

**Actual behavior:** [Compilation errors]
Compiling .pio\build\BIGTREE_SKR_PRO\src\src\sd\usb_flashdrive\lib-uhs2\parsetools.cpp.o
Marlin\src\module\stepper\trinamic.cpp: In function 'void tmc_serial_begin()':
Marlin\src\module\stepper\trinamic.cpp:296:18: error: 'class TMCMarlin<TMC2209Stepper, 'X', '0', (AxisEnum)0>' has no member named 'beginSerial'; did you mean 'HWSerial'?
         stepperX.beginSerial(115200);
                  ^~~~~~~~~~~
                  HWSerial
Marlin\src\module\stepper\trinamic.cpp:310:18: error: 'class TMCMarlin<TMC2209Stepper, 'Y', '0', (AxisEnum)1>' has no member named 'beginSerial'; did you mean 'HWSerial'? 
         stepperY.beginSerial(115200);
                  ^~~~~~~~~~~
                  HWSerial
Marlin\src\module\stepper\trinamic.cpp:324:18: error: 'class TMCMarlin<TMC2209Stepper, 'Z', '0', (AxisEnum)2>' has no member named 'beginSerial'; did you mean 'HWSerial'? 
         stepperZ.beginSerial(115200);
                  ^~~~~~~~~~~
                  HWSerial
Marlin\src\module\stepper\trinamic.cpp:345:19: error: 'class TMCMarlin<TMC2209Stepper, 'E', '0', (AxisEnum)3>' has no member named 'beginSerial'; did you mean 'HWSerial'? 
         stepperE0.beginSerial(115200);
                   ^~~~~~~~~~~
                   HWSerial
In file included from Marlin\src\module\stepper\trinamic.h:35:0,
                 from Marlin\src\module\stepper\trinamic.cpp:32:
Marlin\src\module\stepper\../../feature/tmc_util.h: In instantiation of 'TMCMarlin<TMC2209Stepper, AXIS_LETTER, DRIVER_ID, AXIS_ID>::TMCMarlin(uint16_t, uint16_t, float, uint8_t, bool) [with char AXIS_LETTER = 'X'; char DRIVER_ID = '0'; AxisEnum AXIS_ID = (AxisEnum)0; uint16_t = short unsigned int; uint8_t = unsigned char]':
Marlin\src\module\stepper\trinamic.cpp:203:7:   required from here
Marlin\src\module\stepper\../../feature/tmc_util.h:213:38: error: no matching function for call to 'TMC2209Stepper::TMC2209Stepper(const uint16_t&, const uint16_t&, const 
float&, const uint8_t&)'
       TMC2209Stepper(RX, TX, RS, addr)
                                      ^
In file included from Marlin\src\module\stepper\trinamic.h:29:0,
                 from Marlin\src\module\stepper\trinamic.cpp:32:
.pio\libdeps\BIGTREE_SKR_PRO\TMCStepper_ID5513\src/TMCStepper.h:1013:3: note: candidate: TMC2209Stepper::TMC2209Stepper(Stream*, float, uint8_t)
   TMC2209Stepper(Stream * SerialPort, float RS, uint8_t addr) :
   ^~~~~~~~~~~~~~
.pio\libdeps\BIGTREE_SKR_PRO\TMCStepper_ID5513\src/TMCStepper.h:1013:3: note:   candidate expects 3 arguments, 4 provided
.pio\libdeps\BIGTREE_SKR_PRO\TMCStepper_ID5513\src/TMCStepper.h:1011:7: note: candidate: constexpr TMC2209Stepper::TMC2209Stepper(const TMC2209Stepper&)
 class TMC2209Stepper : public TMC2208Stepper {
       ^~~~~~~~~~~~~~
.pio\libdeps\BIGTREE_SKR_PRO\TMCStepper_ID5513\src/TMCStepper.h:1011:7: note:   candidate expects 1 argument, 4 provided
.pio\libdeps\BIGTREE_SKR_PRO\TMCStepper_ID5513\src/TMCStepper.h:1011:7: note: candidate: constexpr TMC2209Stepper::TMC2209Stepper(TMC2209Stepper&&)
.pio\libdeps\BIGTREE_SKR_PRO\TMCStepper_ID5513\src/TMCStepper.h:1011:7: note:   candidate expects 1 argument, 4 provided
compilation terminated due to -fmax-errors=5.
*** [.pio\build\BIGTREE_SKR_PRO\src\src\module\stepper\trinamic.cpp.o] Error 1
======================================================================= [FAILED] Took 38.29 seconds =======================================================================

Environment              Status    Duration
-----------------------  --------  ------------
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3720890/Marlin.zip)
BIGTREE_SKR_PRO          FAILED    00:00:38.292

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

boelle, 2019-10-12:
please use pastebin for long errors, then insert link to pastebin in your post

psim, 2019-10-12:
done

darknode, 2019-10-12:
Just use ```TMCStepper=https://github.com/bigtreetech/TMCStepper``` instead of original trinamic in ```[common]``` and ```[env:BIGTREE_SKR_PRO]``` sections of platform.ini. Also you'll need to comment error in trinamic.h:
```
#if TMCSTEPPER_VERSION < 0x000500
//  #error "Update TMCStepper library to 0.5.0 or newer."
#endif
```

boelle, 2019-10-12:
you dont need the bigtree tech version

look last reply here: https://github.com/MarlinFirmware/Marlin/pull/15530

darknode, 2019-10-12:
@boelle It just don't compile.

Steps to reproduce:
* Get a fresh clone of repo, switch to branch bugfix-2.0..., do full cleanup(to make sure PIO will download fresh versions of packages and etc), you can use cleanup script like that [build_cleanup.sh.zip](https://github.com/MarlinFirmware/Marlin/files/3721153/build_cleanup.sh.zip) or do it manually.
* Apply SKR Pro + TMC2209 configuration:
** From a patch file
[skrpro_tmc2209.patch.zip](https://github.com/MarlinFirmware/Marlin/files/3721156/skrpro_tmc2209.patch.zip) with ```git apply skrpro_tmc2209.patch```
** OR manually like in attached files [configs.zip](https://github.com/MarlinFirmware/Marlin/files/3721157/configs.zip)

What you'll get is:
```
Compiling .pio/build/BIGTREE_SKR_PRO/src/src/sd/SdFatUtil.cpp.o
Marlin/src/module/stepper/trinamic.cpp: In function 'void tmc_serial_begin()':
Marlin/src/module/stepper/trinamic.cpp:296:18: error: 'class TMCMarlin<TMC2209Stepper, 'X', '0', (AxisEnum)0>' has no member named 'beginSerial'; did you mean 'HWSerial'?
         stepperX.beginSerial(115200)
```

Why? TMC library has a [commit](https://github.com/teemuatlut/TMCStepper/commit/2d14f340a0966a16ac66ecfa435434aac3dd0f7d) which assumed to fix that issue, but it still not release. So you need to switch from version to just master branch of ```TMCStepper``` lib. That switch in turn will get you next error, about absent "SoftwareSerial" (as you will see @extesy has asked about that issue), to fix it you need to get new version of ```framework-arduinoststm```(which is not part of 5.6.0 version of PIO platform for STM32 yet) like that:
```
platform_packages =
  framework-arduinoststm32@~3.10700.0
```
After that you will get issue with timers HAL and at that point I don't have any fixes at the moment :(

I'd like to get rid of BTT TMCStepper version too, but it looks like there is no way to do that at the moment.

boelle, 2019-10-12:
@extesy what do you think is wrong here?

extesy, 2019-10-12:
Here are my configs, I've just verified that it compiles on the latest Marlin code: [configs.zip](https://github.com/MarlinFirmware/Marlin/files/3721202/configs.zip)

You can also check if your build is using same versions as mine:
```
Dependency Graph
|-- <U8glib-HAL> 0.4.1   
|   |-- <Wire> 1.0       
|   |-- <SPI> 1.0        
|-- <LiquidCrystal> 1.3.4
|   |-- <Wire> 1.0       
|-- <TMCStepper> 0.5.1   
|   |-- <SPI> 1.0        
|-- <Adafruit NeoPixel> 1.2.5
|-- <LiquidTWI2>
|   |-- <Wire> 1.0
|-- <Arduino-L6470> 0.7.0
|-- <SPI> 1.0
|-- <Servo> 1.1.2
|-- <EEPROM> 2.0.1
|-- <IWatchdog> 1.0.0
|-- <Wire> 1.0
```

darknode, 2019-10-12:
@extesy It look like drivers are not 2209:
[image]

extesy, 2019-10-12:
No, but #15530 is also doing it for all drivers, not just 2209s.

darknode, 2019-10-12:
@extesy Sorry, just don't get it. #15530 has one changed line:
[image]
which is forces to use master branch of BTT, in which they bumped version to get rid of version check error:
[image]
It work fine, but it's still BTT fork of TMCStepper :)
You've said that "It works for me without using bigtreetech's fork.", but there are no evidences that ***original*** TMCStepper lib works for Skr Pro + 2209.

boelle, 2019-10-12:
change https://github.com/bigtreetech/TMCStepper#master 

to

TMCStepper@>=0.5.0,<1.0.0

you dont need  https://github.com/bigtreetech/TMCStepper#master

darknode, 2019-10-12:
@boelle It will not work as I described [here](https://github.com/MarlinFirmware/Marlin/issues/15538#issuecomment-541363235)

extesy, 2019-10-13:
@darknode I meant that #15530 will affect me because it will use a fork of TMCStepper library that will not be maintained well or updated often unlike the original TMCStepper library.

extesy, 2019-10-13:
Try uncommenting hardware serial in `pins_BIGTREE_SKR_PRO_V1.1.h` file:
```
  #define X_HARDWARE_SERIAL  Serial
  //#define X2_HARDWARE_SERIAL Serial1
  #define Y_HARDWARE_SERIAL  Serial1
  //#define Y2_HARDWARE_SERIAL Serial1
  #define Z_HARDWARE_SERIAL  Serial1
  //#define Z2_HARDWARE_SERIAL Serial1
  #define E0_HARDWARE_SERIAL Serial1
  #define E1_HARDWARE_SERIAL Serial1
```

viper93458, 2019-10-13:
While that does allow a compile it unfortunately doesn't work. At least not for me.

-William

psim, 2019-10-13:
Same here. I tried to explain when I opened this issue. If you modify pins_BIGTREE_SKR_PRO_V1.1.h file it compiles, but doesn't work.

TB1405, 2019-10-13:
same problem here. With the version a month ago and the btt libary it works but now it isnt possible to compile without error.

hackebike, 2019-10-14:
If you are trying to use the TMC 2209 for sensorless homing, it's not working, filed a bug [15496](https://github.com/MarlinFirmware/Marlin/issues/15496)

xeddog, 2019-10-14:
I have the same error for my SKR Pro V1.1 and TMC2208's (UART) as well.  Using "TMCStepper@>=0.5.0,<1.0.0" in Platformio I get the errors,  Using the BTT library compiles.

[TMC2208 Build errors.txt](https://github.com/MarlinFirmware/Marlin/files/3726642/TMC2208.Build.errors.txt)

extesy, 2019-10-14:
@teemuatlut Do you have any opinion on which component (Marlin, TMCStepper, stm32) has a problem that causes this failure?

viper93458, 2019-10-14:
ststm32 needs to be updated but when you force platform IO to use the new version, you get other marlin compile errors even though it resolves the missing software serial implementation.

-William

hackebike, 2019-10-15:
Even if you make it compile, which I have it working point the tmc library to BTT depo, for all the tests I have done, the SKR PRo v1.1 with BTT TMC 2209 sensorless homing does not work ! And if you want to use the TMC  2209 in SPI mode you need to cut a pin !

TB1405, 2019-10-15:
> Even if you make it compile, which I have it working point the tmc library to BTT depo, for all the tests I have done, the SKR PRo v1.1 with BTT TMC 2209 sensorless homing does not work ! And if you want to use the TMC 2209 in SPI mode you need to cut a pin !

2209 and SPI ?

hackebike, 2019-10-15:
yes you can but you need to cut a pin, it's in the BTT manual

sjasonsmith, 2019-10-15:
> yes you can but you need to cut a pin, it's in the BTT manual

The "cut a pin" part of the manual is to disable StallGuard. The TMC2209 does not support SPI, at least according to Trinamic.

Telling you to cut pins is pretty ridiculous, though, regardless of the reason.

hackebike, 2019-10-15:
@sjasonsmith yes you are correct, got confuse with 2208 which I am testing now!
Got a marlin config from BTT for 2209 and sensorless homing, will test it and report.

hackebike, 2019-10-16:
all my tests with SKR Pro 1.1/SKR v1.3 and BTT TMC2209 v1.1/1.2 for sensorless homing failed.
So I tested on 2 different boards and 2 different 2209, setting different sensitivity, but nothing.

hackebike, 2019-10-24:
update: I got sensorless homing and probing [working](https://www.youtube.com/watch?v=UJwzSTcQslo).

curvi, 2019-10-28:
Very nice, maybe you could show how you did this? I cannot get things to run. Neither with TMCStepper from BTT, nor with 0.5.1. I've also tried numerous fixes from multiple related topics, such as [this](https://github.com/MarlinFirmware/Marlin/issues/15680) the [timer4 compile flag](https://github.com/MarlinFirmware/Marlin/pull/15655) and a ton of combinations of those.
Just downloading the newest Marlin 2.0 I can compile without issues, but as soon as I use a TMC2209 in the config -> I get errors like:
```
Marlin\src\module\stepper\trinamic.cpp: In function 'void tmc_serial_begin()':
Marlin\src\module\stepper\trinamic.cpp:296:18: error: 'class TMCMarlin<TMC2209Stepper, 'X', '0', (AxisE
num)0>' has no member named 'beginSerial'; did you mean 'HWSerial'?
         stepperX.beginSerial(115200);
                  ^~~~~~~~~~~
                  HWSerial
Marlin\src\module\stepper\trinamic.cpp:310:18: error: 'class TMCMarlin<TMC2209Stepper, 'Y', '0', (AxisE
num)1>' has no member named 'beginSerial'; did you mean 'HWSerial'?
```

hackebike, 2019-10-28:
these are my [config files](https://github.com/hackebike/3dprinter), take a look and make the modification you need for your printer.

the first thing that you should do is pointing the TMC library to: https://github.com/bigtreetech/TMCStepper

boelle, 2019-11-03:
@psim still an issue?

curvi, 2019-11-03:
@boelle if I may answer your question: pulling the latest release as of now and compiling with SKR PRO 1.1 compiles fine, but as soon as you include TMC2209 it doesn't compile with Serial problems. Therefore, I would say at the moment the problem persists.

sjasonsmith, 2019-11-08:
@psim & @curvi, can you try again. This should be fixed.

hackebike, 2019-11-08:
if I may, I compiled Marlin 2 days ago for the SKR Pro without any changes to platform.ini and it works fine.

boelle, 2019-11-11:
@psim (and only @psim ) is the issue gone?

boelle, 2019-11-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-11-21 by boelle