MarlinFirmware/Marlin #15539: [BUG]  No connection to MMU2s, only after flashing

Sythx, 2019-10-12:
### Description

German:

Keine Verbindung von SKR 1.3 zum MMU2s mit Marlin 2.0 (Heute runtergeladen).
MMU2s reagiert auf Reset aber mehr nicht.
Erst nach dem ich das MMU2s geflasht habe über PrusaSlicer habe ich eine Verbindung und kann per Display es bedienen.
Nach einen Rest oder Neustart geht wieder nichts.
Erst nach dem erneuten Flashen des MMU2s.

English:

No connection from SKR 1.3 to MMU2s with Marlin 2.0 (downloaded today).
MMU2s responds to reset but nothing more.
Only after I have flashed the MMU2s on PrusaSlicer I have a connection and can operate it on the display.
After a rest or a restart is nothing.
Only after re-flashing the MMU2s.

### Steps to Reproduce

German:
1. 3D Drucker Starten.
2. LED's des MMU2s leuchten auf.
3. Filament übers Display Laden.
4. Keine Verbindung zum MMU2s
5. Flashen des MMU2s
6. Verbindung zum MMU2s hergestellt.
7. Filament übers Display laden möglich.
8. Nach Neustart oder Rest wieder Flashen nötig.

English:
1. Start 3D printer.
2. LEDs of the MMU2 light up.
3. Load filament over the display.
4. No connection to the MMU2s
5. Flashing the MMU2s
6. Connection made to MMU2s.
7. Loading filament via the display possible.
8. After restart or rest, flash again.

#### Additional Information
[Config_h+adv.zip](https://github.com/MarlinFirmware/Marlin/files/3720899/Config_h%2Badv.zip)
[Marlin-2.0.x_12.10.19.zip](https://github.com/MarlinFirmware/Marlin/files/3720900/Marlin-2.0.x_12.10.19.zip)

Sythx, 2019-10-12:
German:
Neue Informationen:
Wenn ich das MMU2s vom strom trenne und neustarte, verbindet sich alles.

English:
New information:
When I disconnect the MMU2s from the power and restart, everything connects.

BastR, 2019-10-14:
It could be a bad serial board declaration? Where did you connect the MMU to the SKR1.3?

I've got a SKR PRO 1.1 with MMU2 working great with Marlin 2.0.X_11.10.19, I've connected MMU2 pins to the TFT port because there all needed pins on this port.

Maybe try to replace :

#define MMU2_SERIAL Serial

boelle, 2019-10-22:
@Sythx did you try what @BastR suggested?

Sythx, 2019-10-22:
Hey, When changing from Serial to MSerial1 this error message comes

[Terminal.txt](https://github.com/MarlinFirmware/Marlin/files/3756863/Terminal.txt)

BastR, 2019-10-23:
> Hey, When changing from Serial to MSerial1 this error message comes
> 
> [Terminal.txt](https://github.com/MarlinFirmware/Marlin/files/3756863/Terminal.txt)

Try this:

#if ENABLED(PRUSA_MMU2)

  // Serial port used for communication with MMU2.
  // For AVR enable the UART port used for the MMU. (e.g., internalSerial)
  // For 32-bit boards check your HAL for available serial ports. (e.g., Serial2)
  //#define INTERNAL_SERIAL_PORT 2
  #define MMU2_SERIAL Serial1

Sythx, 2019-10-25:
With Serial1 it compiled without error, but no connection to the MMU2.
With Serial0 he makes this error message.

[Terminal v2.txt](https://github.com/MarlinFirmware/Marlin/files/3772908/Terminal.v2.txt)

boelle, 2019-11-03:
@BastR does the error from @Sythx make sense?

BastR, 2019-11-03:
@boelle I don't know, It could by a wiring problem or bad configuration in marlin or marlin itself, but I'm running a MMU2 with a SKR PRO and it works well... @Sythx you have to try various things:

I'm starting from scratch:

### -You need to connect your MMU to the TFT port:
[image]
**make sure to put the MMU RX wire on the TX pin on the board and the MMU TX wire on the RX pin on the board**
 
### You should have (in the order):

-+5V --> BLUE
-GND--> NOTHING
-TX0 --> WHITE
-RX0 --> GREEN
-RST --> BROWN (this pin I'm not sure, try it)

### Now you need to configure marlin like:

[image]

[image]

You must find the right serial port to communicate with the MMU2 according to those available in the HAL.h file (right part of the screenshot). In the exemple above I have **Serial1** because in the HAL.h of my CPU I have this port enable. On the screenshot you can see all available ports for your hardware.

I can't do nothing more I don't have this board and I don't know anyone who has it with an MMU2, sorry :(

tedukes, 2019-11-09:
Just got a used MMU2S off eBay and going to try to set it up. Would really like to know if Sythx got his working and what he did. TIA

tedukes, 2019-11-09:
Does the runout sensor on the MMU2S shutdown the printer on a filament change or runout? TIA

Sjoerrdd, 2019-11-13:
No @tedukes, it's supposed to run the "change fillament" command (M600) :(

tedukes, 2019-11-13:
Thanks, maybe I should used 'pause' instead of shutdown.

tedukes, 2019-11-17:
@BastR The wiring you posted above, is that for the SKR 1.3 on the TFT port? Also, I tried Serial1 but I have no LCD unless I reset the SKR board. I changed it to just Serial, got it working, but the printer boots twice on startup and I have lost some function in Octoprint using the terminal. I can send commands but i do not get/see a response. Other than that, I was able to calibrate the MMU2S and start printing in Single mode.

Any help with the above issues would be greatly appreciated!
[Marlin_MMU2S_Configs.zip](https://github.com/MarlinFirmware/Marlin/files/3855881/Marlin_MMU2S_Configs.zip)

boelle, 2019-11-24:
@Sythx have you tried to ask for help any of these places?

RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
[Tom's 3D Forums](https://discuss.toms3d.org/)
Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
[Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
Marlin Discord server. Join link: https://discord.gg/n5NJ59y

boelle, 2019-12-03:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-12-03 by boelle