MarlinFirmware/Marlin #15540: non-constant condition for static assertion marlin

disappear123df, 2019-10-12:
Arduino: 1.8.8 (Linux), Board: "Anet V1.0"

In file included from sketch/endstops.cpp:35:0:
sketch/endstop_interrupts.h: In function 'void setup_endstop_interrupts()':
endstop_interrupts.h:121:7: error: non-constant condition for static assertion
       static_assert(digitalPinToPCICR(X_MIN_PIN) != NULL, "X_MIN_PIN is not interrupt-capable");
       ^~~~~~~~~~~~~
In file included from /home/wally/.arduino15/packages/arduino/hardware/avr/1.8.1/cores/arduino/Arduino.h:257:0,
                 from sketch/HAL.h:36,
                 from sketch/MarlinConfig.h:39,
                 from sketch/Marlin.h:35,
                 from sketch/endstops.cpp:27:
/home/wally/Documents/arduino-1.8.8/hardware/anet/avr/variants/sanguino/pins_arduino.h:117:66: error: reinterpret_cast from integer to pointer
 #define digitalPinToPCICR(p)    ( (((p) >= 0) && ((p) <= 31)) ? (&PCICR) : ((uint8_t *)0) )
                                                                 ~^~~~~~~
sketch/endstop_interrupts.h:121:21: note: in expansion of macro 'digitalPinToPCICR'
       static_assert(digitalPinToPCICR(X_MIN_PIN) != NULL, "X_MIN_PIN is not interrupt-capable");
                     ^~~~~~~~~~~~~~~~~
In file included from sketch/endstops.cpp:35:0:
endstop_interrupts.h:141:7: error: non-constant condition for static assertion
       static_assert(digitalPinToPCICR(Y_MIN_PIN) != NULL, "Y_MIN_PIN is not interrupt-capable");
       ^~~~~~~~~~~~~
In file included from /home/wally/.arduino15/packages/arduino/hardware/avr/1.8.1/cores/arduino/Arduino.h:257:0,
                 from sketch/HAL.h:36,
                 from sketch/MarlinConfig.h:39,
                 from sketch/Marlin.h:35,
                 from sketch/endstops.cpp:27:
/home/wally/Documents/arduino-1.8.8/hardware/anet/avr/variants/sanguino/pins_arduino.h:117:66: error: reinterpret_cast from integer to pointer
 #define digitalPinToPCICR(p)    ( (((p) >= 0) && ((p) <= 31)) ? (&PCICR) : ((uint8_t *)0) )
                                                                 ~^~~~~~~
sketch/endstop_interrupts.h:141:21: note: in expansion of macro 'digitalPinToPCICR'
       static_assert(digitalPinToPCICR(Y_MIN_PIN) != NULL, "Y_MIN_PIN is not interrupt-capable");
                     ^~~~~~~~~~~~~~~~~
In file included from sketch/endstops.cpp:35:0:
endstop_interrupts.h:161:7: error: non-constant condition for static assertion
       static_assert(digitalPinToPCICR(Z_MIN_PIN) != NULL, "Z_MIN_PIN is not interrupt-capable");
       ^~~~~~~~~~~~~
In file included from /home/wally/.arduino15/packages/arduino/hardware/avr/1.8.1/cores/arduino/Arduino.h:257:0,
                 from sketch/HAL.h:36,
                 from sketch/MarlinConfig.h:39,
                 from sketch/Marlin.h:35,
                 from sketch/endstops.cpp:27:
/home/wally/Documents/arduino-1.8.8/hardware/anet/avr/variants/sanguino/pins_arduino.h:117:66: error: reinterpret_cast from integer to pointer
 #define digitalPinToPCICR(p)    ( (((p) >= 0) && ((p) <= 31)) ? (&PCICR) : ((uint8_t *)0) )
                                                                 ~^~~~~~~
sketch/endstop_interrupts.h:161:21: note: in expansion of macro 'digitalPinToPCICR'
       static_assert(digitalPinToPCICR(Z_MIN_PIN) != NULL, "Z_MIN_PIN is not interrupt-capable");
                     ^~~~~~~~~~~~~~~~~
exit status 1
non-constant condition for static assertion

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

boelle, 2019-10-12:
tried 1.9.0 beta of ide?
where is the config files ? issue template says to attach them as zip
what version of marlin?

GMagician, 2019-10-12:
> "X_MIN_PIN is not interrupt-capable

This is the error...maybe wrong pins used (or enabled endstop interrupts on a board that doesn't support it?)

effeerre75, 2019-10-15:
Was my case, now compiling fine. 
Same error but on Y_MAX_PIN.
Disabled ENDSTOP_INTERRUPTS_FEATURE.
Arduino+RAMPS 1.4

boelle, 2019-10-15:
will close as its mostly a user error, ie wrong use pins used

we can always reopen

closed 2019-10-15 by boelle

Gabrielerusso, 2019-10-20:
I've the same issue with a configuration i've used for years and pin14 is interrupt capable (PCINT10), with ide 1.8.10 and 1.8.11.

> will close as its mostly a user error, ie wrong use pins used
> 
> we can always reopen

got the same error, it's an IDE problem not recognizing the pointer of the following macro:
```
  #define digitalPinToPCICR(p)    ( WITHIN(p, 10, 15) || \
                                    WITHIN(p, 50, 53) || \
                                    WITHIN(p, 62, 69) ? &PCICR : (uint8_t*)0 )
```

throwing this error that causes the "non-constant condition for static assertion" error

`endstop_interrupts.h:57:57: error: reinterpret_cast from integer to pointer

                                     WITHIN(p, 62, 69) ? &PCICR : (uint8_t*)0 )`

with windows-store version 1.8.21.0 working perfectly.

disappear123df, 2019-10-24:
how do i deactivate ENDSTOP_INTERRUPTS or change the pin configuration!

sublime392, 2019-10-31:
I'm not sure why this is closed. I can't compile after version 1.8.8, and now I can't use that version because it doesn't work on Catalina. 

This is not user error @boelle.

boelle, 2019-10-31:
> ide 1.8.10 and 1.8.11.

recommended is 1.9.0 beta or vscode+platform.io

boelle, 2019-10-31:
> is not the proper

what is then?

sublime392, 2019-10-31:
> > is not the proper
> 
> what is then?

What I mean to say is, the ENDSTOP_INTERRUPTS are supposed to be enabled. I have tried the 1.9.0 version and the same error exists there.

ansemjo, 2019-11-02:
I hit this same error before when trying to compile with the Arduino IDE. After updating to the latest commit I got a different error but still couldn't compile with the `ENDSTOP_INTERRUPTS_FEATURE` enabled.

Reproducible with PlatformIO and the latest example configuration for the Anet A8, see #15770.

Interestingly, it works fine on the `bugfix-1.1.x` branch.

dzamosenchuk, 2019-11-03:
I've tried to compile for Anet A2+ printer and have same error. I've tried Arduino IDE 1.8.10 and beta 1.9.

GMagician, 2019-11-03:
@dzamosenchuk please check my PR to see if it fixes this issue

disappear123df, 2019-11-03:
Comment out ENDSTOP_INTERRUPTS in the Configuration.h on in it should do the trick

ansemjo, 2019-11-03:
I checked @GMagician's updated PR and it compiled fine with Arduino IDE and `arduino-cli` while `ENDTOP_INTERRUPTS_FEATURE` was enabled. Thank you!

I used this script to check:

```
#!/usr/bin/env bash

# enter temporary directory
cd "$(mktemp -d)" && pwd

# clone marlin source, checkout pull request 15771
git clone https://github.com/MarlinFirmware/Marlin.git ./
git fetch origin pull/15771/head:pr15771
git checkout pr15771

# copy Anet A8 example config
cp config/examples/Anet/A8/Configuration* Marlin/

# compile with arduino-cli (ensure that LiquidCrystal library and Sanguino board definitions are installed)
arduino-cli compile --fqbn Sanguino\:avr\:sanguino\:cpu=atmega1284p Marlin/Marlin.ino
```