MarlinFirmware/Marlin #15542: [FR] 3 HEATERS_PARALLEL it's possible?

el-quique, 2019-10-12:
Hi. I wanted to know if it is possible to use 3 heaters at the same time with the function heaters parallel. What part of the code would you have to look for?
Additional Information
Board Rumba (3 heater: HE0 pin 2 - HE1 pin 3 - HE2 pin 6)
3D Printer (delta) Geeetech Rostock 301 with new hotend
https://reprap.org/wiki/RUMBA

robbycandra, 2019-10-13:
Easy way: use 1 thermistor, then connect all heater in pararel.

el-quique, 2019-10-13:
> Easy way: use 1 thermistor, then connect all heater in pararel.

Hi. The function Heater Paralles only 2 HE (HE0 and HE1) and 1 termistor. In my case 3 HE and 1 termistor, 2 wire for each HE so as not to overextract the terminals.

el-quique, 2019-10-22:
Another ideas?

el-quique, 2019-10-22:
It is for a very special case. I just have to try it. THX

closed 2020-05-12 by el-quique