MarlinFirmware/Marlin #15543: [BUG] Junction deviation breaking arc moves on CNC

040medien, 2019-10-12:
### Description

I'm running Marlin 2.0 as of 1df90aa on my CNC. I'm using EXTRUDERS=0, junction deviation and S-curve acceleration on a SKR v1.3. My steppers are TMC5160 and I'm using square wave stepping.

When doing smaller arc moves, the machine gets jerky and seems to lose steps (position of (0,0,0) different from the one where it started). When switching back to classic jerk, it works fine.

### Steps to Reproduce

1. Install a recent Marlin 2.0 from github (e.g. 1df90aa)
2. Use the attached configs to install it on a SKR v1.3 running 
3. Run the attached gcode containing arc moves

**Expected behavior:**
Smooth arc moves

**Actual behavior:** 
Janky, stuttering arc moves
When enabling classic jerk the arc moves are smooth
When disabling S-Curve-acceleration but keeping JD on, the arc moves are still stuttering

#### Additional Information
test gcode
[arc test gcode.txt](https://github.com/MarlinFirmware/Marlin/files/3721198/arc.test.gcode.txt)
configs:
[configs.zip](https://github.com/MarlinFirmware/Marlin/files/3721200/configs.zip)

Video showing junction deviation:
https://photos.google.com/share/AF1QipN7PVDIAZeS0s5HcXHNGIhE-5gdW7MJgzZfaHjZONPKP3LFFpiOxLo2XaJ0_-JVVA?key=LS1oRVR6X01hODBaeDd6NzM2WDA3SkNEQ0Rac3Z3

Video showing classic jerk
https://photos.app.goo.gl/NRx2sWdQjcGeb3Gm9

robbycandra, 2019-10-13:
Have you try to edit the JD? If not try to use JD = 0.08.

040medien, 2019-10-13:
Same result with JD = 0.08

boelle, 2019-10-13:
taken from discord server:

commit 2cf35267c4a9e85e7fecfc58aa7cab3edcabeffc
Author: Roxy-3D <Roxy-3D@users.noreply.github.com>
Date:   Thu Oct 10 19:44:15 2019 -0500

    Set Junction Deviation value from Jerk and Default Acceleration (#15509)

    Formula: `0.4 * sq(min(xjerk, yjerk)) / (default accel)`

does it help you?

040medien, 2019-10-15:
@boelle This did indeed help. Turns out the value in my case is much lower at 0.002. This seems to work fine.