MarlinFirmware/Marlin #15552: [BUG] Unable to compile "sketch/src/feature/spindle_laser.cpp:42:1: error: missing '(' in expression"

mjscosta, 2019-10-13:
### Description

I'm working on a RADDS variant board, redefining some of the RADDS pins for laser support.
After redefining the pins and enabling the LASER_FEATURE in the configuration I get his compilation error.

```
Arduino: 1.8.10 (Linux), Board: "Arduino Due (Programming Port)"

In file included from /home/user/.arduino15/packages/arduino/hardware/sam/1.6.12/cores/arduino/Arduino.h:44:0,
                 from sketch/src/HAL/HAL_DUE/usb/usb_task.c:50:
/home/user/.arduino15/packages/arduino/hardware/sam/1.6.12/cores/arduino/wiring_constants.h:33:0: warning: "true" redefined [enabled by default]
 #define true 0x1
 ^
In file included from sketch/src/HAL/HAL_DUE/usb/compiler.h:85:0,
                 from sketch/src/HAL/HAL_DUE/usb/conf_usb.h:54,
                 from sketch/src/HAL/HAL_DUE/usb/usb_task.c:48:
/home/user/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/lib/gcc/arm-none-eabi/4.8.3/include/stdbool.h:34:0: note: this is the location of the previous definition
 #define true 1
 ^
In file included from /home/user/.arduino15/packages/arduino/hardware/sam/1.6.12/cores/arduino/Arduino.h:44:0,
                 from sketch/src/HAL/HAL_DUE/usb/usb_task.c:50:
/home/user/.arduino15/packages/arduino/hardware/sam/1.6.12/cores/arduino/wiring_constants.h:34:0: warning: "false" redefined [enabled by default]
 #define false 0x0
 ^
In file included from sketch/src/HAL/HAL_DUE/usb/compiler.h:85:0,
                 from sketch/src/HAL/HAL_DUE/usb/conf_usb.h:54,
                 from sketch/src/HAL/HAL_DUE/usb/usb_task.c:48:
/home/user/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/lib/gcc/arm-none-eabi/4.8.3/include/stdbool.h:35:0: note: this is the location of the previous definition
 #define false 0
 ^
sketch/src/feature/spindle_laser.cpp:42:1: error: operator "defined" requires an identifier
   #if ENABLED(SPINDLE_LASER_PWM) && PIN_EXISTS(SPINDLE_LASER_PWM)
 ^
sketch/src/feature/spindle_laser.cpp:42:1: error: user-defined literal in preprocessor expression
sketch/src/feature/spindle_laser.cpp:42:1: error: missing '(' in expression
Multiple libraries were found for "Wire.h"
 Used: /home/user/.arduino15/packages/arduino/hardware/sam/1.6.12/libraries/Wire
exit status 1
Error compiling for board Arduino Due (Programming Port).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

```

### Steps to Reproduce

1. Clone the following branch `git clone -b bugfix-2.0.x-laser --single-branch git@github.com:mjscosta/Marlin.git
`
2. Open with Arduino IDE v >⁼ 1.8.10, select board "Arduino Due"
3. Hit "Verify/Compile"

**Expected behavior:** Expect the code to compile without errors.

**Actual behavior:** It does not compile

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

DerAndere1, 2019-10-15:
branches other than bugfix-2.0 or the tagged releases from the download section of marlin.org are usually meant for development of new features and not for usage and may be work in progress. Do you have an issue if you clone branch `bugfix-2.0.x` instead of `bugfix-2.0.x-laser`?

mjscosta, 2019-10-15:
Hi @DerAndere1, I've branched bugfix-2.0.x into bugfix-2.0.x-laser in my fork.
I was introducing a new board variant, it is based on the current bugfix-2.0.x.

To reproduce the issue in the current bugfix-2.0.x without the RADDS board variant for laser that I've added I think I could try the RAMPS_FD_V1 board, that is the only board that has laser configuration pins, but then RAMPS_FD_V1 is meant for Arduino Mega, and I want to test it for Arduino Due on RADDS.

AnHardt, 2019-10-16:
> RAMPS_FD_V1 is meant for Arduino Mega

Nonsens!
[Ramps FD](https://reprap.org/wiki/RAMPS-FD). Translate **FD** as **F**or **D**ue

mjscosta, 2019-10-16:
@AnHardt thanks for the clarification.

Then indeed, the issue exists in the bugfix-2.0.x branch and can be reproduced using RAMPS_FD_V1. 
I'll create a config and add it to the ticket reproducing the issue.

Cheers

boelle, 2019-10-22:
so we are waiting for the config? or is the issue solved?

mjscosta, 2019-10-22:
Hi @boelle , thanks for checking, a lot of things going on, but I managed to squeeze a set of config files for the bugfix-2.0.x.

boelle, 2019-10-30:
@shitcreek any idea why this will not compile?

boelle, 2019-11-03:
@mjscosta many updates have been added to 2.0.x in last 12 days, is this still an issue?

shitcreek, 2019-11-06:
No issue compiling with the latest build and set to your motherboard, RAMPS_FD_V2, and `LASER_FEATURE` enabled, aside from a pin conflict with the fan.

Since you are testing out the laser, you may want to try out this reworked laser code:
https://github.com/MarlinFirmware/Marlin/pull/15335

mjscosta, 2019-11-07:
@shitcreek Thanks for checking!

I'll test as soon as possible. 
Just checking, is there any log on the laser functionality/features I could check?
A friend of mine, has been testing the laser functionality of Marlin 2.0 on the old code, he had to make a switch to the fan and make some hacks due to delays (he was performing some engraving/raster of images) and the laser functionality was behaving worst then fan option.

I know that Marlin Kimbra has integrated some functionality for the laser, to address some issues when performing raster jobs.

shitcreek, 2019-11-08:
No log other than our discourse in that PR. I'm getting great raster results using the inline feature and PWM frequency set at 2500 on an 8bit board.

mjscosta, 2019-11-08:
Hi @shitcreek nice to know, thanks your feedback, Ill let you know ASAP.

boelle, 2019-11-11:
@mjscosta any news?

shitcreek, 2019-11-12:
@mjscosta I'm closing this thread since the original post isn't an issue / or is fixed. Please report your feedback regarding the laser stuff in the aforementioned PR. Thanks.

closed 2019-11-12 by shitcreek