MarlinFirmware/Marlin #15558: [BUG] Probe failed with G32 & G29. When probing almost done, it stopped

VietLongMaker, 2019-10-14:
Hi, i'm making a delta bot , type of probe is allen key. When levelling bed or auto calibrate, probe moved and touched bed lightly. Unfortunately, when job almost done, nozzle moved to center of heatbed and  stopped. I tried to decrease probing speed, debug, but i cannot fix it.. 
Here is my config file, video when levelling bed and screenshot when probe stopped

[image]
[configuration Link](https://drive.google.com/open?id=1aS8nhVw6LpaOJ6j5a9gf2URMpNvOnqjB)
[VIDEO when bed levelling](https://drive.google.com/open?id=112oFkyVJc93sqkALF2JxHPZhijjR8eqa)

I debug with M111 S32 and here is result:
23:53:07.280 : destination=(-39.50, -37.63, 50.40) : prepare_uninterpolated_move_to_destination
23:53:07.285 : current_position=(-39.50, -37.63, 50.40) : z raise move
23:53:07.288 : current_position=(-39.50, -37.63, 50.40) : xy move
23:53:07.869 : <<< do_blocking_move_to
23:53:07.874 : >>> do_blocking_move_to(-64.00, 56.00, 23.00)
23:53:07.874 : >>> do_blocking_move_to(-64.00, 56.00, 3.00)
23:53:07.878 : >>> do_blocking_move_to(-64.00, 56.00, 43.00)
23:53:07.878 : >>> do_blocking_move_to(0.00, 0.00, 43.00)
23:53:07.882 : destination=(-39.50, -37.63, 50.40) : set_destination_from_current
23:53:09.377 : current_position=(0.00, 0.00, 50.40) : xy move
23:53:09.381 : destination=(0.00, 0.00, 43.00) : prepare_uninterpolated_move_to_destination
23:53:09.385 : current_position=(0.00, 0.00, 43.00) : z lower move
23:53:09.836 : <<< do_blocking_move_to
23:53:09.840 : Error:Z-Probe failed
23:53:09.858 : Error:Printer stopped due to errors. Fix the error and use M999 to restart. (Temperature is reset. Set it after restarting)
23:53:10.197 : current_position=(0.00, 0.00, 43.00) : > probing complete
23:53:10.201 : current_position=(0.00, 0.00, 43.00) : clean_up_after_endstop_or_probe_move
23:53:10.201 : <<< G29
23:53:10.204 : X:0.00 Y:0.00 Z:43.00 E:0.00 Count X:23035 Y:23035 Z:23035
23:53:12.519 : Resend: 5
23:53:12.525 : Error:Line Number is not Last Line Number+1, Last Line: 4
23:53:12.525 : Resend: 5

My english wasn't good. I'm sorry. Please help me.
Thank you very much.
 Viet Long.

boubouexpress, 2019-10-14:
Hello
check with my configuration
I had the same problems
but I'm in AUTO_BED_LEVELING_UBL
I changed the size of the bed and #define MESH_INSET was 45
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3725819/Configuration.zip)
 the bed

VietLongMaker, 2019-10-15:
> Hello
> check with my configuration
> I had the same problems
> but I'm in AUTO_BED_LEVELING_UBL
> I changed the size of the bed and #define MESH_INSET was 45
> [Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3725819/Configuration.zip)

Oh, thank you. I used FIX_MOUNTED_PROBE instead of ALLEN_KEY  and it seem be improved. Although I can't understand the reason of that. Thank you again. I think its time to close this issue

closed 2019-10-15 by VietLongMaker