MarlinFirmware/Marlin #15561: G29 staying ALOT inside of the bed bounding

bywebberen, 2019-10-15:
Hej Guys.

I have compiled the 1.1.9 marlin to my corexy printer, and i have the probe sat at 30mm(-30) to the left of the nozzle. My bed size is x175 y165 (z242)mm when i do the G28 and G29 with the bounding box  no larger than the bed size. 
It homes(G28) like it should. but it doesn't go left position of 0 or 5 mm but keeps about the (-30mm) to the right on the ABL ??

Videolink with the ABL in action: (notice that it doesn't want to go all the way to the left.
on my previous compiled OLDER version I could just tell it probe Left -25 and that took care of it, but the 1.1.9 wont let me do this) 
[https://www.dropbox.com/s/cjs738hztn23ehy/XiaoYing_Video_1571131502471.mp4?dl=0](url)

CONFIG.H
`/**
 * Marlin 3D Printer Firmware
 * Copyright (C) 2016 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (C) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <http://www.gnu.org/licenses/>.
 *
 */

/**
 * Configuration.h
 *
 * Basic settings such as:
 *
 * - Type of electronics
 * - Type of temperature sensor
 * - Printer geometry
 * - Endstop configuration
 * - LCD controller
 * - Extra features
 *
 * Advanced settings can be found in Configuration_adv.h
 *
 */
#ifndef CONFIGURATION_H
#define CONFIGURATION_H
#define CONFIGURATION_H_VERSION 010109

//===========================================================================
//============================= Getting Started =============================
//===========================================================================

/**
 * Here are some standard links for getting your machine calibrated:
 *
 * http://reprap.org/wiki/Calibration
 * http://youtu.be/wAL9d7FgInk
 * http://calculator.josefprusa.cz
 * http://reprap.org/wiki/Triffid_Hunter%27s_Calibration_Guide
 * http://www.thingiverse.com/thing:5573
 * https://sites.google.com/site/repraplogphase/calibration-of-your-reprap
 * http://www.thingiverse.com/thing:298812
 */

//===========================================================================
//============================= DELTA Printer ===============================
//===========================================================================
[... 3509 lines cut ...]
  #define MAX7219_DEBUG_PLANNER_TAIL  5  // Show the planner queue tail position on this and the next LED matrix row

  #define MAX7219_DEBUG_PLANNER_QUEUE 0  // Show the current planner queue depth on this and the next LED matrix row
                                         // If you experience stuttering, reboots, etc. this option can reveal how
                                         // tweaks made to the configuration are affecting the printer in real-time.
#endif

/**
 * NanoDLP Sync support
 *
 * Add support for Synchronized Z moves when using with NanoDLP. G0/G1 axis moves will output "Z_move_comp"
 * string to enable synchronization with DLP projector exposure. This change will allow to use
 * [[WaitForDoneMessage]] instead of populating your gcode with M400 commands
 */
//#define NANODLP_Z_SYNC
#if ENABLED(NANODLP_Z_SYNC)
  //#define NANODLP_ALL_AXIS  // Enables "Z_move_comp" output on any axis move.
                              // Default behaviour is limited to Z axis only.
#endif

// Enable Marlin dev mode which adds some special commands
//#define MARLIN_DEV_MODE

#endif // CONFIGURATION_ADV_H
`

boelle, 2019-10-15:
have you tried with marlin 2.0.x ?
1.1.x is not updated anymore

InsanityAutomation, 2019-10-15:
And actually I just revised a ton of the calculation here, so it absolutely warrants revisiting in 2.0 right now!

bywebberen, 2019-10-16:
I just couldn't get atom and VS to work on my pc some dll error, so I went with this, but you might be right. And when I used this on one of my other printers, with the sensor behind the nozzle, I had the same issue just in that direction (back)

InsanityAutomation, 2019-10-16:
Looking at the video, the behavior seems to be correctly applying the probe offset vs the maximum movement limits. Those positions are defined with 

#define LEFT_PROBE_BED_POSITION 5
  #define RIGHT_PROBE_BED_POSITION 140
  #define FRONT_PROBE_BED_POSITION 5
  #define BACK_PROBE_BED_POSITION 160

and appears to be the right position (140) that is stopped short of where you expect. I get the feeling the X_MAX position is short and the hotend can physically go a bit farther.

bywebberen, 2019-10-16:
I could see what you mean, bur when I jog the machine it goes in the
correct direction??

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virusfri.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

bywebberen, 2019-10-16:
I perhaps found a place in conf.h where it was commented out in a funny way.. Ill test it now

bywebberen, 2019-10-16:
Arhg... I might have found my issue.. the nozzle is actually 30mm to the right of both the origin zero and the sensor, so of course it want to make the measurements here.. Now I just need to tell it that home 0 is actually home 30

Does it make sense ??

bywebberen, 2019-10-16:
Yeah ! WE GOT LIFT OFF! ! !

It was in fact because of the wrong input HOME_POSITION on the X axis !

bywebberen, 2019-10-16:
Hmm... Now the issue is on the MAX right instead... if I try to make it go further than 140 in marlin it prompts me that it is outside of the bed area, which it is clearly not...

bywebberen, 2019-10-16:
I will post my conf.h - now it works . . NOT SURE WHY... 
`/**
 * Marlin 3D Printer Firmware
 * Copyright (C) 2016 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (C) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <http://www.gnu.org/licenses/>.
 *
 */

/**
 * Configuration.h
 *
 * Basic settings such as:
 *
 * - Type of electronics
 * - Type of temperature sensor
 * - Printer geometry
 * - Endstop configuration
 * - LCD controller
 * - Extra features
 *
 * Advanced settings can be found in Configuration_adv.h
 *
 */
#ifndef CONFIGURATION_H
#define CONFIGURATION_H
#define CONFIGURATION_H_VERSION 010109

//===========================================================================
//============================= Getting Started =============================
//===========================================================================

/**
 * Here are some standard links for getting your machine calibrated:
 *
 * http://reprap.org/wiki/Calibration
 * http://youtu.be/wAL9d7FgInk
 * http://calculator.josefprusa.cz
 * http://reprap.org/wiki/Triffid_Hunter%27s_Calibration_Guide
 * http://www.thingiverse.com/thing:5573
 * https://sites.google.com/site/repraplogphase/calibration-of-your-reprap
 * http://www.thingiverse.com/thing:298812
 */

//===========================================================================
//============================= DELTA Printer ===============================
//===========================================================================
// For a Delta printer start with one of the configuration files in the
// example_configurations/delta directory and customize for your machine.
//

//===========================================================================
//============================= SCARA Printer ===============================
//===========================================================================
// For a SCARA printer start with the configuration files in
// example_configurations/SCARA and customize for your machine.
[... 1831 lines cut ...]

/**
 * R/C SERVO support
 * Sponsored by TrinityLabs, Reworked by codexmas
 */

/**
 * Number of servos
 *
 * For some servo-related options NUM_SERVOS will be set automatically.
 * Set this manually if there are extra servos needing manual control.
 * Leave undefined or set to 0 to entirely disable the servo subsystem.
 */
//#define NUM_SERVOS 3 // Servo index starts with 0 for M280 command

// Delay (in milliseconds) before the next move will start, to give the servo time to reach its target angle.
// 300ms is a good value but you can try less delay.
// If the servo can't reach the requested position, increase it.
#define SERVO_DELAY { 300 }

// Only power servos during movement, otherwise leave off to prevent jitter
//#define DEACTIVATE_SERVOS_AFTER_MOVE

#endif // CONFIGURATION_H
`

closed 2019-10-16 by bywebberen