MarlinFirmware/Marlin #15562: [BUG] (short description)

Ziyayildizz3, 2019-10-15:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

Evgeny-SPB, 2019-10-19:
Hello! I have the same problem on bugfix-2.0.x branch.
I have dual Z carriage (Z_DUAL_STEPPER_DRIVERS) setup on Prusa like printer with LPC1768 processor. Bed leveling is not working.

I did some research: Everything is working as [expected] before commit "Add custom types for position" (https://github.com/MarlinFirmware/Marlin/commit/50e4545255605eb506c20eb107270038b0fe7bdb)

After this commit Z axis stops moving after G29. Printer completes G29 and ignores all following commands to move Z axis. No mistakes or warnings is produced.

Ziyayildizz3, 2019-10-20:
It doesn't work when I send my mesh bed settings. I've recorded all levels of 25 points. When I throw it out of the repetier, the values I entered without leveling must be passed. Am not getting ...

Ziyayildizz3, 2019-10-20:
Sample:
Mesh Bed Leveling:
M420 S0 Z0.00
G29 S3 X1 Y1 Z0.30000
G29 S3 X2 Y1 Z0.25000
G29 S3 X3 Y1 Z0.17500
G29 S3 X4 Y1 Z0.07500
G29 S3 X5 Y1 Z-0.00000
G29 S3 X1 Y2 Z0.25000
G29 S3 X2 Y2 Z0.20000
G29 S3 X3 Y2 Z0.17500
G29 S3 X4 Y2 Z0.10000
G29 S3 X5 Y2 Z-0.02500
G29 S3 X1 Y3 Z0.15000
G29 S3 X2 Y3 Z0.17500
G29 S3 X3 Y3 Z0.15000
G29 S3 X4 Y3 Z0.07500
G29 S3 X5 Y3 Z-0.00000
G29 S3 X1 Y4 Z0.02500
G29 S3 X2 Y4 Z0.07500
G29 S3 X3 Y4 Z0.05000
G29 S3 X4 Y4 Z0.05000
G29 S3 X5 Y4 Z-0.00000
G29 S3 X1 Y5 Z0.05000
G29 S3 X2 Y5 Z0.05000
G29 S3 X3 Y5 Z0.05000
G29 S3 X4 Y5 Z0.02500
G29 S3 X5 Y5 Z-0.00000
M500

boelle, 2019-10-26:
@Ziyayildizz3 do you still have issues?

boelle, 2019-11-03:
this is most likely been fixed with pr #15654 

closing

if its not the case we can reopen

closed 2019-11-03 by boelle