MarlinFirmware/Marlin #15564: [BUG] TMC2130 LayerShift on CoreXY Thesis

GeorgZiegler, 2019-10-15:
Thesis: Layer Shift issues with Trinamics TMC2160 did occur more often on CoreXY machines

On Cartesian machines the X <-> Y Axes are completely decoupled,
meaning one axe can move without interfering the mechanical load
of the other axe. So the stepper driver has 'only' to deal with things like:
on this axes mass to be accelarated and frictions.

In contradiction the CoreXY kinematics :
With the equations of motion (https://corexy.com/theory.html)
∆X=0.5*(∆A+∆B)  ∆Y=0.5*(∆A-∆B)  
∆A=∆X+∆Y       ∆B=∆X-∆Y    
we see 
a. for producing a 45° movement, only 1 motor has to run
b. for x-direction movement only, both steppers has to run
   in same rotation direction simultanously.
c. for y-direction movement only, both steppers has to run
   in opposite rotation direction simultanously.

A thought experiment, to point out the 'coupling' :
According to a, if we do a 45° move e.g. only the 'A' stepper
has to step while the other one 'B' has to blocked.
Now, what if 'B' stepper driver sees
'Ah, no stepping pulse', i can reduce my holding torque by reducing the current ?

Due to a non ideal symmetric mechanical system ( e.g. by different belt tensions ) 
a stepping in one driver, produces mechanical forces on the other one,
which this one doesn't know about.
So it makes it more difficult to handle this.

Possible to be compared / evaluted by other users:
In my CoreXY i have a Layer Shift !!always!! in +X direction.
                  __________
So like  ___|           ___|     in front view
            |___________|

           ______________  
           |    |                |
           |    |                |     in top view
           |__ |__________|

and e.g. not in a 45° direction in top view

So with equation of motion this could only happen
if we had step losses while a (-∆A + -∆B) on !both! drivers
simultanously! 
Did other users see the same effect?
But i couldn't interpret this enough, at least it did point out
the interfering coupling effect for me.

boelle, 2019-10-24:
@teemuatlut @shitcreek you think he is on to something here?

teemuatlut, 2019-10-25:
It sounds very plausible and reasonable but as we already offer the option to change the hold current I don't think there's a need to make any changes. It's still nice to know and might certainly help someone when troubleshooting.

boelle, 2019-10-30:
will close this one as there is no need to change 
but the good info can still be found with the search tool

closed 2019-10-30 by boelle