MarlinFirmware/Marlin #15567: [BUG] EXTRUDERS == 0 makes machine unstable if some features such as S_CURVE_ACCELERATION and X/Y_DUAL_STEPPER_DRIVERS are also enabled

anttix, 2019-10-15:
### Description

Setting `EXTRUDERS = 0` (e.g. to power a CNC machine) with a combination of `S_CURVE_ACCELERATION` or `X/Y_DUAL_STEPPER_DRIVERS` will make firmware unstable and slow.

### Steps to Reproduce

1. Apply the following patch to stock config.

```
diff --git a/Marlin/Configuration.h b/Marlin/Configuration.h
index 4568965..73f2fc0 100644
--- a/Marlin/Configuration.h
+++ b/Marlin/Configuration.h
@@ -143,7 +143,7 @@
 
 // This defines the number of extruders
 // :[1, 2, 3, 4, 5, 6]
-#define EXTRUDERS 1
+#define EXTRUDERS 0
 
 // Generally expected filament diameter (1.75, 2.85, 3.0, ...). Used for Volumetric, Filament Width Sensor, etc.
 #define DEFAULT_NOMINAL_FILAMENT_DIA 3.0
@@ -796,7 +796,7 @@
  *
  * See https://github.com/synthetos/TinyG/wiki/Jerk-Controlled-Motion-Explained
  */
-//#define S_CURVE_ACCELERATION
+#define S_CURVE_ACCELERATION
 
 //===========================================================================
 //============================= Z Probe Options =============================
```

or

apply this patch

```
diff --git a/Marlin/Configuration.h b/Marlin/Configuration.h
index 4568965..2d34d58 100644
--- a/Marlin/Configuration.h
+++ b/Marlin/Configuration.h
@@ -143,7 +143,7 @@
 
 // This defines the number of extruders
 // :[1, 2, 3, 4, 5, 6]
-#define EXTRUDERS 1
+#define EXTRUDERS 0
 
 // Generally expected filament diameter (1.75, 2.85, 3.0, ...). Used for Volumetric, Filament Width Sensor, etc.
 #define DEFAULT_NOMINAL_FILAMENT_DIA 3.0
@@ -605,8 +605,8 @@
 #define USE_XMIN_PLUG
 #define USE_YMIN_PLUG
 #define USE_ZMIN_PLUG
-//#define USE_XMAX_PLUG
-//#define USE_YMAX_PLUG
+#define USE_XMAX_PLUG
+#define USE_YMAX_PLUG
 //#define USE_ZMAX_PLUG
 
 // Enable pullup for all endstops to prevent a floating state
diff --git a/Marlin/Configuration_adv.h b/Marlin/Configuration_adv.h
index 7b7ae9c..3ac7eee 100644
--- a/Marlin/Configuration_adv.h
+++ b/Marlin/Configuration_adv.h
@@ -420,20 +420,20 @@
  * in X2. Dual endstop offsets can be set at runtime with 'M666 X<offset> Y<offset> Z<offset>'.
  */
 
-//#define X_DUAL_STEPPER_DRIVERS
+#define X_DUAL_STEPPER_DRIVERS
 #if ENABLED(X_DUAL_STEPPER_DRIVERS)
   #define INVERT_X2_VS_X_DIR true   // Set 'true' if X motors should rotate in opposite directions
-  //#define X_DUAL_ENDSTOPS
+  #define X_DUAL_ENDSTOPS
   #if ENABLED(X_DUAL_ENDSTOPS)
     #define X2_USE_ENDSTOP _XMAX_
     #define X_DUAL_ENDSTOPS_ADJUSTMENT  0
   #endif
 #endif
 
-//#define Y_DUAL_STEPPER_DRIVERS
+#define Y_DUAL_STEPPER_DRIVERS
 #if ENABLED(Y_DUAL_STEPPER_DRIVERS)
   #define INVERT_Y2_VS_Y_DIR true   // Set 'true' if Y motors should rotate in opposite directions
-  //#define Y_DUAL_ENDSTOPS
+  #define Y_DUAL_ENDSTOPS
   #if ENABLED(Y_DUAL_ENDSTOPS)
     #define Y2_USE_ENDSTOP _YMAX_
     #define Y_DUAL_ENDSTOPS_ADJUSTMENT  0
```

2. Run a complex gcode file that has simultaneous xyz moves e.g. [lion.gcode](https://github.com/MarlinFirmware/Marlin/files/3730452/lion.gcode.gz)

**Expected behavior:** The gcode file completes normally with nominal time. Physics limited runtime of the above file is around 30min, on an Atmega board the file should take about 40min due to CPU not being powerful enough.

**Actual behavior:** The file takes over 4h to complete or never completes at all, the firmware reports being busy for extended periods of time, there are random resets on LPC1768 based  boards.

#### Additional Information

Oddly enough enabling dual steppers without separate end-stops works fine.

Tested with two different boards: 
* Atmega 2560 with RAMPS 1.4 
* Bigtreetech SKR 1.3

shitcreek, 2019-10-19:
Could you clarify on this: "Oddly enough enabling dual steppers without separate end-stops works fine." ?

anttix, 2019-10-21:
@shitcreek if I disabled ```X_DUAL_ENDSTOPS``` and ```Y_DUAL_ENDSTOPS``` then the planner didn't freeze any more. I didn't do any extensive testing in this mode though.

thinkyhead, 2019-10-25:
It's not yet supported to set `EXTRUDERS` to 0. We already have issues open on the topic, so this can be closed.

closed 2019-10-25 by thinkyhead

anttix, 2019-11-25:
Currently open issue: https://github.com/MarlinFirmware/Marlin/issues/10212