MarlinFirmware/Marlin #15574: [BUG] Enabling  CUSTOM_USER_MENU_TITLE will break the build

anttix, 2019-10-16:
### Description

Setting CUSTOM_USER_MENU_TITLE will break the build

### Steps to Reproduce

1. Enable LCD, custom user menu and a custom title by applying the patch below to stock config
```
diff --git a/Marlin/Configuration.h b/Marlin/Configuration.h
index 155605f..72fef6c 100644
--- a/Marlin/Configuration.h
+++ b/Marlin/Configuration.h
@@ -1727,7 +1727,7 @@
 //
 // Note: Usually sold with a white PCB.
 //
-//#define REPRAP_DISCOUNT_SMART_CONTROLLER
+#define REPRAP_DISCOUNT_SMART_CONTROLLER
 
 //
 // Original RADDS LCD Display+Encoder+SDCardReader
diff --git a/Marlin/Configuration_adv.h b/Marlin/Configuration_adv.h
index be8cc57..959c825 100644
--- a/Marlin/Configuration_adv.h
+++ b/Marlin/Configuration_adv.h
@@ -2498,9 +2498,9 @@
 /**
  * User-defined menu items that execute custom GCode
  */
-//#define CUSTOM_USER_MENUS
+#define CUSTOM_USER_MENUS
 #if ENABLED(CUSTOM_USER_MENUS)
-  //#define CUSTOM_USER_MENU_TITLE "Custom Commands"
+  #define CUSTOM_USER_MENU_TITLE "Custom Commands"
   #define USER_SCRIPT_DONE "M117 User Script Done"
   #define USER_SCRIPT_AUDIBLE_FEEDBACK
   //#define USER_SCRIPT_RETURN  // Return to status screen after a script
```
2. Run `platformio run -e megaatmega2560`

**Expected behavior:** Successful build

**Actual behavior:** Compiler error

```
In file included from Marlin/src/lcd/menu/menu_main.cpp:31:0:
Marlin/src/lcd/menu/menu_main.cpp: In function 'void menu_main()':
Marlin/src/lcd/menu/../../inc/../../Configuration_adv.h:2503:34: error: expected unqualified-id before string constant
#define CUSTOM_USER_MENU_TITLE "Custom Commands"
^
Marlin/src/lcd/menu/menu.h:341:28: note: in definition of macro '_MENU_ITEM_P'
PGM_P const plabel = PLABEL;                                \
^
Marlin/src/lcd/menu/menu.h:391:48: note: in expansion of macro 'MENU_ITEM_P'
#define SUBMENU_P(PLABEL, DEST)                MENU_ITEM_P(submenu, PLABEL, DEST)
^
Marlin/src/lcd/menu/menu.h:392:50: note: in expansion of macro 'SUBMENU_P'
#define SUBMENU(LABEL, DEST)                     SUBMENU_P(GET_TEXT(LABEL), DEST)
^
Marlin/src/lcd/menu/menu.h:392:60: note: in expansion of macro 'GET_TEXT'
#define SUBMENU(LABEL, DEST)                     SUBMENU_P(GET_TEXT(LABEL), DEST)
^
Marlin/src/lcd/menu/menu_main.cpp:177:5: note: in expansion of macro 'SUBMENU'
SUBMENU(MSG_USER_MENU, menu_user);
^
Marlin/src/lcd/menu/menu_main.cpp:175:29: note: in expansion of macro 'CUSTOM_USER_MENU_TITLE'
#define MSG_USER_MENU CUSTOM_USER_MENU_TITLE
^
Marlin/src/lcd/menu/menu_main.cpp:177:13: note: in expansion of macro 'MSG_USER_MENU'
SUBMENU(MSG_USER_MENU, menu_user);
^
*** [.pioenvs/megaatmega2560/src/src/lcd/menu/menu_main.cpp.o] Error 1

```

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

InsanityAutomation, 2019-10-16:
I just saw this on the discord bot and seems we both opened PR's for this. The undef is no longer needed as the source is not a #define variable anymore and the convention is PSTR for the menu text typically outside of the translatables.

anttix, 2019-10-16:
@InsanityAutomation thanx. I kept the #undef to surface other possible uses of that constant but you're right, this is unnecessary. I've posted an updated revision w/o undef.

Bergerac56, 2019-10-17:
#15526

Bergerac56, 2019-10-17:
#15526

GMagician, 2019-10-18:
I think this has been solved and so may be closed

PR #15575 (2019-10-16): Fix CUSTOM_USER_MENU_TITLE compile error (#15574)
### Requirements

Setting CUSTOM_USER_MENU_TITLE in `Configuration_adv.h` should compile and work.

### Description

### Benefits

Fixes #15574

### Testing done

Tested on Bigtreetech SKR 1.3 board with RepRap discount FG display.
closed 2019-10-19 by anttix