MarlinFirmware/Marlin #15577: [BUG] Z Probe Past Bed

DkOuvE, 2019-10-16:
I have Z_SAFE_HOMING enabled, but when I home all axis, the printer doesn't move to the probe point and I get the "Z Probe Past Bed" message on screen. Z does not home. I use an inductive sensor with an offset of -0.6. 

The Z axis still does not home after moving the printer coordinates to where the probe point is either.

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3733457/Configuration.zip)

shitcreek, 2019-10-16:
Your 'Z_SAFE_HOMING_X_POINT' is set at 10 and `NOZZLE_TO_PROBE_OFFSET` for X is 18, which would put nozzle off the bed at -8 while your `X_MIN_POS` is -3.

DkOuvE, 2019-10-16:
That's the reason why I tried to manually move to the centre of the bed and home there. It still refused to home. Usually, the printer automatically moves to a safe X and Y coord before I pulled the latest Marlin 2.0

boelle, 2019-10-22:
is this still an issue?

DkOuvE, 2019-10-23:
Git pulled the latest marlin and I'm still getting "Z Probe Past Bed" and the XY isn't automatically moving to a valid location.

boelle, 2019-10-23:
that is when it has to go to far down to activate the probe

does the switch touch the bed? you can use M119 to check endstop status and use finger to activate the probe

DkOuvE, 2019-10-23:
After further testing, it seem that the bug is caused by manually setting a safe z home point. When I uncommented the manual probe points, the printer homes as expected in the centre of the build platform.

boelle, 2019-10-24:
i will close this one as it looks to be a config issue rather than a bug

we can always reopen

closed 2019-10-24 by boelle